mavlink/mavlink #1373: Gimbal V2 Protocol Discussion II: Gimbal device information vs Gimbal manager information

olliw42, 2020-05-06:
The information in the two messages GIMBAL_DEVICE_INFORMATION and GIMBAL_MANAGER_INFORMATION is quite confusing. 

One the one hand some data in GIMBAL_DEVICE_INFORMATION are duplicated in GIMBAL_MANAGER_INFORMATION but on the other hand some are not.

The effect is that a GCS for instance wants to request them both in order to have the complete information, since some data in GIMBAL_DEVICE_INFORMATION is not also in GIMBAL_MANAGER_INFORMATION, but due to the duplication it receives some data twice.

I have difficulties to understand the rational here.

I would see three canonical approaches:

A. GIMBAL_MANAGER_INFORMATION is a superset of GIMBAL_DEVICE_INFORMATION

In that approach all data carried in GIMBAL_DEVICE_INFORMATION would be mirrored in GIMBAL_MANAGER_INFORMATION. A GCS would thus only request GIMBAL_MANAGER_INFORMATION, but not GIMBAL_DEVICE_INFORMATION. The gimbal manager on the other hand would have had to first request  GIMBAL_DEVICE_INFORMATION before it could serve a GIMBAL_MANAGER_INFORMATION request from the GCS.

B. GIMBAL_MANAGER_INFORMATION and GIMBAL_DEVICE_INFORMATION are disjoint

In that approach all data is either carried in GIMBAL_DEVICE_INFORMATION or carried in GIMBAL_MANAGER_INFORMATION, but no data field would be duplicated. A GCS would thus request both in order to have the full information. The gimbal manager would still have to request  GIMBAL_DEVICE_INFORMATION in order to get this data, some of which it may need for doing its job, however, having received it would not be a precondition for serving a GIMBAL_MANAGER_INFORMATION request from the GCS.

C. The data in GIMBAL_DEVICE_INFORMATION which is duplicated in COMPONENT_INFORMATION is removed

In that approach all data in GIMBAL_DEVICE_INFORMATION which is also found in COMPONENT_INFORMATION would be removed from GIMBAL_DEVICE_INFORMATION. Currently this would make GIMBAL_MANAGER_INFORMATION a superset of GIMBAL_DEVICE_INFORMATION as in A, but one also could opt for disjoint messages like in B. In that approach the GCS would have in addition to request COMPONENT_INFORMATION, which however it may want to do anyhow.

May suggestion:
Please explain the rational in the current design, or if it happens that there is none consider choosing A, B, or C.

:)

julianoes, 2020-06-02:
> The effect is that a GCS for instance wants to request them both in order to have the complete information

The GCS should not request GIMBAL_DEVICE_INFORMATION! Only the gimbal manager should do so and it can then provide whatever GIMBAL_MANAGER_INFORMATION it finds appropriate.

Therefore, the rational is very much A. I guess that's not clear from the docs and we should clarify it.

olliw42, 2020-06-10:
>The GCS should not request GIMBAL_DEVICE_INFORMATION! Only the gimbal manager should do so 

perfect. This makes sense and also streamlines/simplifies things.

the GIMBAL_MANAGER_INFORMATION might need some changes to fully realize this however. Even though the issue is settled I therefore don't yet close to not forget.

olliw42, 2020-12-05:
discussion has kind of settled, and the issue went stale anyway, so I close it. :)

closed 2020-12-05 by olliw42