mavlink/mavlink #1376: Gimbal V2 Protocol Discussion II: Gimbal manager set attitude command vs set attitude message

olliw42, 2020-05-08:
I find the two gimbal manager control messages DO_GIMBAL_MANAGER_ATTITUDE and GIMBAL_MANAGER_SET_ATTITUDE, and especially their interplay and inter-dependence, somewhat not so intuitive. I will comment on the interplay in an extra issue, but want to here suggest the following:

It is somewhat strange that both refer to ATTITUDE, but DO_GIMBAL_MANAGER_ATTITUDE in fact related to tilt and pan angles and not the attitude quaternion., as all the others do.

Now, I very well understand the motivation for that, since for setting a mission one really doesn't want to fool around with a quaternion but wants more human friendly variables. And this generally also bodes very well with the idea of the gimbal manager to abstract the gimbal operation, as it's also explicitly mentioned in the documentation.

However, IMHO, a very similar argument could be made also for GIMBAL_MANAGER_SET_ATTITUDE. In many if not most typical applications one would prefer to deal directly with tilt and pan angles and not with a quaternion. As example I mention a GCS doing stick control (which is also mentioned in the docs).

In addition, I feel that appropriately descriptive command/message names also can help a lot.

**My suggestion:**

* I would rename the existing command DO_GIMBAL_MANAGER_ATTITUDE into DO_GIMBAL_MANAGER_TILTPAN (or DO_GIMBAL_MANAGER_PITCHYAW)

* I would add a new message GIMBAL_MANAGER_SET_TILTPAN (or GIMBAL_MANAGER_SET_PITCHYAW), which in function is analogous to the existing GIMBAL_MANAGER_SET_ATTITUDE except for more human friendly tilt, pan fields (and no roll related fields)

* Possibly, largely for the sake of symmetry (the mathematics version of it I love so much), one also could add a new command DO_GIMBAL_MANAGER_ATTITUDE, which uses a quaternion (at the cost of no rates due to the 7 param limit). But I would agree that it might not be the most often used command, and it would not the biggest loss to world if it would not be added.

:)

Cheers,
Olli

julianoes, 2020-06-02:
> However, IMHO, a very similar argument could be made also for GIMBAL_MANAGER_SET_ATTITUDE. In many if not most typical applications one would prefer to deal directly with tilt and pan angles and not [...]

I actually completely agree and I see how it could make it more straightforward and simpler to describe. @LorenzMeier had a very strong opinion on this arguing for quaternion everywhere in MAVLink, so I was guided by that.

> I would rename the existing command DO_GIMBAL_MANAGER_ATTITUDE into DO_GIMBAL_MANAGER_TILTPAN (or DO_GIMBAL_MANAGER_PITCHYAW)

I like `DO_GIMBAL_MANAGER_TILTPAN`.

> I would add a new message GIMBAL_MANAGER_SET_TILTPAN

I'm against this because it makes deconfliction harder. You have to explain to only send one at a time, etc.

> Possibly, largely for the sake of symmetry (the mathematics version of it I love so much), one also could add a new command DO_GIMBAL_MANAGER_ATTITUDE, which uses a quaternion

Here I have a strong opinion, no! We did that for similar ideological reasons with [MAV_CMD_DO_MOUNT_CONTROL_QUAT](https://mavlink.io/en/messages/common.html#MAV_CMD_DO_MOUNT_CONTROL_QUAT) and then it was just there and no one ever implemented it anywhere as far as I know!

olliw42, 2020-06-09:
I think we could agree on 
* renaming the existing MAV_CMD_DO_GIMBAL_MANAGER_ATTITUDE to MAV_CMD_DO_GIMBAL_MANAGER_TILTPAN
* adding a new message GIMBAL_MANAGER_SET_TILTPAN which only carries tilt and pan angles and angular velocities (and of course the other stuff like flags, ids, etc)
* keep the existing GIMBAL_MANAGER_SET_ATTITUDE as it is (i.e. with quaternion and rate vector)

julianoes, 2020-06-09:
Yes, I think that makes sense, thanks!

olliw42, 2020-07-01:
done in PR #1397 
so closing

closed 2020-07-01 by olliw42