mavlink/mavlink #1378: Gimbal V2 Protocol Discussion II: Displaying angles

olliw42, 2020-05-12:
This issue may at first come across like nonsense, but I think there is real practical point.

One of the strong features of the gimbal protocol v2 is that the low-level control and status messages transport the information about the gimbal attitude as a quaternion, which avoids the issues with the (Euler) angles. 

However, angles still play a role, but there is no specification for how to handle them.

Angles appear e.g. in MAV_CMD_DO_GIMBAL_MANAGER_ATTITUDE. Since roll is zero, this however doesn't present a big issue, and conversion to and from a quaternion is straightforward. The common NED convention works well (as it's equal here to the Euler angles related to the typical pitch-roll-yaw gimbals)(in the notation I'm using here NED corresponds to roll-pitch-yaw).

However, angles also appear when one wants to e.g. **display** the gimbal attitude, but here roll is not zero a priori. This has the effect that if one converts the quaternion delivered by GIMBAL_DEVICE_ATTITUDE_STATUS into angles for display using NED, the roll and yaw angles go nuts when one is approaching pitch = +-90°, even though the gimbal does behave perfectly fine in this condition.

I note that tilting a gimbal down by 90° is a very common situation, so that this is a real practical issue.

I can't see a simple and yet fully satisfying solution (and as far as I can see, there can't be any), so some compromise must be made. Since the different compromises typically lead to different angles even for one and the same gimbal  attitude, the preferred compromise should ideally be part of the specification, in order to minimize confusion and inconsistencies (not everyone should cook its own soup).

One option would be to generally use the pitch-roll-yaw Euler system for displaying. This works well for the large majority of gimbals, and in addition has the advantage that the pitch and yaw angles are meaningful in the sense that they can be used in MAV_CMD_DO_GIMBAL_MANAGER_ATTITUDE.

The other options which came to my mind all effectively result in angles which are not directly related to or consistent with "naive" expectation. 

Anyway, whatever the final result, the purpose of this issue is to point to this practical problem in order to trigger a solution finding.

:)

julianoes, 2020-06-02:
I have no idea what to say. I suggest if you have a better idea, make a pull request and we change it, if not, we leave it as is.

olliw42, 2020-06-09:
I am using the roll-pitch-yaw Eulers instead of the pitch-roll-yaw in the conversions for quaternion and vice versa. It's the only option I could come up with so far. Which doesn't mean it is the only one or even the best one ;).

dlwalter, 2020-11-10:
> display the gimbal attitude, but here roll is not zero a priori. This has the effect that if one converts the quaternion delivered by GIMBAL_DEVICE_ATTITUDE_STATUS into angles for display using NED, [...]

I believe this is why we usually just express the gimbal attitude as tilt/pan or inclination/azimuth.  Trying to break down the quaternion attitude into an Euler sequence will have troubles at the singularity for that Euler sequence (+/-90deg for roll, pitch, yaw and straight forward for roll, pitch, roll) and becomes a big inverse kinematics problem.  So generalize the mechanical design out of it.  If we know the tilt and pan (or inclination/azimuth) the user/GCS can determine where it is pointing.

julianoes, 2020-11-11:
@dlwalter I would argue that's why we have a quaternion, so you can get whatever angles out of it with the right transformation, right?

olliw42, 2020-11-11:
right :)

dlwalter, 2020-11-11:
@julianoes yeah, exactly. 

I just understood this thread as concerning "displaying" angles like on a HUD on camera overlay.  But the quaternion has all the necessary information.  

I would recommend we just provide examples in the documentation on how to convert that into a more human-readable format such as azimuth/inclination or an Euler sequence for whatever mechanical setup is implemented.

mattstraehl, 2020-11-12:
I just came across the discussion here. As part of some work on Auterion Mission Control to consume the Gimbal V2 Protocol I have spent some time trying to wrap my head around Euler angles and quaternions, and I believe that the quaternion representation is the wrong choice for the V2 protocol. The reason is that in gimbal lock situations it's not possible to recover the **original** yaw, pitch and roll angles of the physical gimbal from the quaternion. I'm summarising my findings here.

The **yaw**, **pitch** and **roll** angles are the **ZYX Euler angles** (or [Tait-Bryan angles](https://en.wikipedia.org/wiki/Euler_angles#Rotation_matrix) to be precise) of a gimbal. If a gimbal's pitch angle is at minus 90 degrees, one degree of freedom is lost since the other two axes are aligned. As a result, [any combination of yaw and roll summing up to the same value, result in the same rotation](https://en.wikipedia.org/wiki/Gimbal_lock#Loss_of_a_degree_of_freedom_with_Euler_angles).

It's true that we can compute some Euler angles from a quaternion corresponding to the same rotation. But if a quaternion describes a rotation that would be a gimbal lock if expressed in Euler angles, then Euler angles cannot uniquely describe that rotation. In other words, on a physical gimbal there are different yaw, pitch, roll configurations resulting in the same gimbal orientation. These different configurations translate into the same quaternion, because the quaternion just describes the rotation, but not how this rotation is achieved step by step in terms of yaw, pitch and roll changes.

I think there is a confusion that comes from the fact that quaternions are thought to be more powerful than Euler angles as they don't suffer from gimbal lock. But since consumers of V2 will actually want to know the gimbal's Euler angles, gimbal lock is exactly the issue with the quaternion representation.

The bellow simple example proves my claim and shows how two valid gimbal configurations, both in a gimbal lock (due to `pitch=-PI/2`) and `roll + yaw` summing up to the same value, result in the same quaternion because they are actually the same rotation:

```cpp
float q1[4];
float q2[4];
mavlink_euler_to_quaternion(0.5/*roll*/, -M_PI_2/*pitch*/, 0.0/*yaw*/, q1);
mavlink_euler_to_quaternion(0.0/*roll*/, -M_PI_2/*pitch*/, 0.5/*yaw*/, q2);
printf("q1: %.6f, %.6f, %.6f, %.6f\n", q1[0], q1[1], q1[2], q1[3]);
printf("q2: %.6f, %.6f, %.6f, %.6f\n", q2[0], q2[1], q2[2], q2[3]);
```

The output is:
```
q1: 0.685125, 0.174941, -0.685125, 0.174941
q2: 0.685125, 0.174941, -0.685125, 0.174941
```

Since the two quaternions are identical, it is impossible to recover the original roll and yaw angles from it. In fact any Euler angles with `pitch=-PI/2` and roll and yaw summing up to 0.5 could be a solution.

The V2 protocol is still marked as work in progress, so I propose that we either replace the quaternion with the yaw, pitch and roll angles as in V1, or add them along with the quaternion.

dlwalter, 2020-11-12:
>  so I propose that we either replace the quaternion with the yaw, pitch and roll angles...

This makes a big assumption on the mechanical design of the gimbal.  Not all gimbals follow a yaw-pitch-roll servo sequence.  (An example is the U.S. Military Litening targeting pod which uses roll-pitch-roll).

The Quaternion will always tell you where the gimbal is trying to look and I think this is important for the operator on the ground.  You are correct in that the Quaternion doesn't tell you how the gimbal is mechanically achieving that orientation because you need to do the inverse kinematics and deal with those singularities (+/-90deg pitch for YPR, and 0/-180deg yaw for a RPR sequence).  

Maybe a generalized 'servo_angle1-3' would be appropriate to communicate the actual orientation of the gimbal stages?

olliw42, 2020-11-12:
> The reason is that in gimbal lock situations it's not possible to recover the original yaw, pitch and roll angles of the physical gimbal from the quaternion.

it's already in this sentence there the logical flaw appears. There IS NO SUCH THING AS TRUE OR CORRECT OR 'T H E' YAW, PITCH, ROLL ANGLES!!! Ergo, whatever yaw, pitch, roll angles the gimbal considers 'original' is totally irrelevant.

I think you want to read that statement carefully.

There are 12 sets of Euler angles which all are mathematically equally good or bad, and since that's mathematically proven there is no point in questioning this.

:)

> You are correct in that the Quaternion doesn't tell you how the gimbal is mechanically achieving that orientation 

from the yaw, pitch, roll angles you can't determine either how the gimbal is mechanically achieving that orientation. So, that's a tie.

Ergo, if there is a point one could consider adding is a field which describes the gimbals mechanical construction/Euler set. I can't see any practically useful need for it, except and only for displaying or for giving people a reason to cling to Euler angles.

LorenzMeier, 2020-11-12:
I just had an offline conversation with Matthias and its obvious that there was some confusion that is now straightened out.

dlwalter, 2020-11-12:
> Ergo, if there is a point one could consider adding is a field which describes the gimbals mechanical construction/Euler set. I can't see any practically useful need for it, except and only for displaying or for giving people a reason to cling to Euler angles.

@olliw42 I agree.

hamishwillee, 2021-01-27:
Julian, when you get back this is assigned to you to close or progress

closed 2021-01-28 by julianoes