mavlink/mavlink #1380: Gimbal V2 Protocol Discussion II: Multiple Clients and Deconfliction

olliw42, 2020-05-14:
The point raised here is touching the heart of the concept. 

A key part of the concept is - in my understanding - the ability to handle multiple parties, which may want to control the gimbal. Examples of such parties, which I'll call clients, are GCS, missions, RC inputs on the autopilot, companion computers, and so on. It was in fact the whole idea of introducing the gimbal manager to have a means to prevent the cacophony of multiple clients accessing the gimbal simultaneously. 

However, currently, only exactly two clients, namely missions and a GCS, can be effectively handled by the protocol, and in addition the gimbal manager needs to be on the autopilot (or where the missions are). From a fundamental perspective this raises the question for what that whole exercise is then good for in the first place, since for that particular situation easier solutions are well possible. To not be misunderstood, I'm not saying that the protocol as it currently is would not work fine for the situation it is obviously made for, not at all, it does it well. I'm saying however that the situation which it is made for is so particular and restricted in scope and constrained that the protocol hardly can be considered satisfying or useful. For instance, RC targeting, which would be just another client, is not possible within the protocol, and the docs in fact explicitly say that (it suggests an either-or mechanism, e.g set by a parameter, but the protocol has no generic means to report that). However, RC targeting is probably the very first thing the majority of folks which stick a gimbal to a craft will do and want. Even the old protocol which the new wants to supersede was better. For instance, a companion computer (yet another client) which wants to do e.g. quick shots. If it would use the SET_ATTITUDE message it would conflict with a GCS and if it would use the CMD_DO_SET_ATTITUDE command it would conflict with missions, and there is no means in the protocol to resolve that. However, a companion computer wanting to control the gimbal is really a most obvious application. So: The protocol can't handle even the most basic and obvious applications. I have a dozen more of this kind on my sheet, but I don't see value in listing them all, so I don't. Instead I directly progress to my suggestion(s).

It actually turns out that it just needs some slight adaptions (but maybe bigger shifts in conceptual thinking) to correct the situation. All it needs is to remove the artificially introduced restrictions and correlations.

The suggestion is based on these observations:
* there is no reason to require the gimbal manager to be located on the autopilot, or mission client
* there is no reason to distinguish one of the clients and to make it special in any way
* there is no reason to give each client its private message/command
* there is no reason to limit to two clients
* overriding the primary client is factually nothing else than to handing over the control to a client
* there is no reason to define explicit deconfliction rules as part of the specification, except of saying that it's the gimbal manager's job to deconflict. In the current spec they are needed to compensate for the deficiencies in the protocol. All it needs however is to provide the information on the protocol level so that the gimbal manager can deconflict, the explicit rules can then be implementation dependent.

Thus, the basic idea is to simply treat all clients alike, and to memorize the client so that multiple clients can be handled.

Each client can be in one of three states:
* no interest in the gimbal currently 
* wants to nudge
* wants to exclusively control the gimbal

Therefore, each client's intention can be expressed by two bits in the flags, and - to follow the precedence set by the current protocol - let's call them NUDGE and OVERRIDE. There would be such a pair of flags for each client. We can start with defining relatively common clients, e.g. MISSION, GCS, COMPANION, RC, and in order to have some flexibility one could add two more custom-defined, CLIENT1, CLIENT2.

So, all we need is this set of flags:

MISSION_NUDGE
MISSION_OVERRIDE
GCS_NUDGE
GCS_OVERRIDE
COMPANION_NUDGE
COMPANION_OVERRIDE
RC_NUDGE
RC_OVERRIDE
CLIENT1_NUDGE
CLIENT1_OVERRIDE
CLIENT2_NUDGE
CLIENT2_OVERRIDE

It now can be complained that it is inconvenient to always have to set the MISSION_OVERRIDE flag when one is setting up a mission, and I think that this was probably the original motivation for making the MISSION client the primary one in the current protocol. So, for that one client we don't define the flag as high-active, but as low-active, and - to follow again the precedence set by the current protocol - call it NONE (it might be very worthwhile though to find a better name).

So, we finally have:

MISSION_NUDGE
MISSION_NONE
GCS_NUDGE
GCS_OVERRIDE
COMPANION_NUDGE
COMPANION_OVERRIDE
RC_NUDGE
RC_OVERRIDE
CLIENT1_NUDGE
CLIENT1_OVERRIDE
CLIENT2_NUDGE
CLIENT2_OVERRIDE

Noteworthily, this is highly compatible with any current implementation: NONE maps to MISSION_NONE, NUDGE maps top GCS_NUDGE, and OVERRIDE maps to GCS_OVERRIDE. The huge shift in concept should however also be noted. All clients are fundamentally equal, which gives them all fundamentally the same possibilities - if the gimbal manager allows it. The implementer of the gimbal manager may well decide to make MISSION the default and primary client as in the current protocol, however this is now a decision by the implementer and not a restriction enforced by the protocol. Similarly, the implementer may well follow the deconfliction rules as they are currently in the spec, but again, this is now a decision by the implementer and not a restriction enforced by the protocol. The implementer of the gimbal manager may well want and decide to do things somewhat differently.

Obviously also, only one of the override flags can be set at any time, the gimbal manager always can ensure that. In contrast, multiple clients can be in a nudging state (it's up to the gimbal manager implementation which nudges it wants to allow).

There is one slight catch in that the 16 bit space reserved for the gimbal manager might be too tight to be future proof. This could be expanded in several ways, e.g. to put the gimbal device flags in their extra field, to have two fields flags1, flags2, or to use uint64, but I'm not sure which one would be best.

There is a second slight catch in that it could happen that it is not possible currently to sort out which client has send a message, and that a client_id would be needed (1=MISSION, 2= GCS, ...), I couldn't identify such a situation yet, but I might miss it.

Yes, it would be as simple as that to put the protocol's head on its feet!

:)

Even though only indirectly related, I'd like also to repeat this suggestion (which I think I've made in the old gimbal thread):

In the spec it's currently said that a gimbal manager should emit it's status at a "low rate (e.g. 5 Hz)". If the spec really means 5 Hz all is fine, but I'm not sure 5 Hz can really be considered a low rate and I thus suspect that the 5 Hz is a misprint, and that it e.g. should have been 0.5 Hz, but a several seconds rate in any case. I would find 5 Hz also quite excessive, actually.

A high rate is only needed in order to reduce any delay from a request to change a gimbal manager flag (via the SET messages/command) to communicating the new flag the gimbal manager has come up with. However, for most of the time such changes would not occur, and a pretty low reporting rate would be fine. In order to reduce traffic, I thus would suggest that the gimbal manager sends out its status at a high rate for only a short period when the flags have changed, but at a low rate otherwise. For instance, sending four status messages with an interval of 1/4 sec, after each change, but otherwise sending in 3 secs intervals. This I found quite easy to implement, and it looks very reasonable to me.

Cheers, Olli

julianoes, 2020-06-02:
> However, currently, only exactly two clients, namely missions and a GCS, can be effectively handled by the protocol, 

No, I don't see why more can't be handled.

> and in addition the gimbal manager needs to be on the autopilot (or where the missions are).

No, the autopilot can send the mission item commands to the gimbal manager.

> For instance, RC targeting, which would be just another client, is not possible within the protocol, and the docs in fact explicitly say that (it suggests an either-or mechanism, e.g set by a [...]

RC targeting is on the same level as the `GIMBAL_MANAGER_SET_ATTITUDE` streamed message. The autopilot needs to be configured to use exactly one (or auto switch if that's required).

> For instance, a companion computer (yet another client) which wants to do e.g. quick shots. If it would use the SET_ATTITUDE message it would conflict with a GCS

This is an interesting one. This would have to use override in order to override manual control, however, that means manual control can't override anymore. Therefore, my gut feeling is that the SET_ATTITUDE message needs a priority field at which point the configurator or implementer of the whole system can set the priority of the messages sent by the companion computer to be higher than what comes from the GCS to enable the smart shot.

> If it would use the SET_ATTITUDE message it would conflict with a GCS and if it would use the CMD_DO_SET_ATTITUDE command it would conflict with missions, and there is no means in the protocol to resolve that.

The way to resolve it is that something set with a command has to be "unset" again once complete. E.g. after a mission is done `DO_SET_ATTITUDE` with flag `NONE` unsets it. If that's not done the gimbal is stuck to what was before. It will take some time until these bugs are fixed but I think every deconfliction method would require some active handover to give back control, like that.

> - there is no reason to require the gimbal manager to be located on the autopilot, or mission client
> - there is no reason to distinguish one of the clients and to make it special in any way
> - there [...]

Yes, agreed to all above, and it's mostly how I'm already thinking about it. You're only change as I see it is that you want to also deconflict based on origin (GCS vs mission) and not just message type. If I remember correctly, that's always what I had in mind too but you were against it. But I might be wrong about that, anyway.

My vision was that commands are higher priority until they are finished. And if one player is "bad" and doesn't give back control, then that's a bug and needs to be fixed.

> The implementer of the gimbal manager may well decide to make MISSION the default and primary client as in the current protocol, however this is now a decision by the implementer and not a [...]

To me this approach sounds more complex and hence more difficult to explain to implements. Also, it means behaviour might differ depending on what gimbal manager you're talking to which is not what GCS implementer or GCS user would want.

Also the flags are somewhat limited and I can already see how more and more "components" are added to the flags as people have more bespoke setups. At this point you might as well look at the component ID as the source instead of this flag.

julianoes, 2020-06-02:
> In the spec it's currently said that a gimbal manager should emit it's status at a "low rate (e.g. 5 Hz)". If the spec really means 5 Hz all is fine, but I'm not sure 5 Hz can really be considered a [...]

Alright, low might not be fitting here. To me "high" would mean > 20 Hz, hence 5 Hz is low but sure let's just remove the "low" and say 5 Hz. 

> However, for most of the time such changes would not occur, and a pretty low reporting rate would be fine. In order to reduce traffic, I thus would suggest that the gimbal manager sends out its [...]

If you write that clearly and concisely in the spec, then I'm fine with that. If I have to keep explaining to people afterwards, I'll change it back to X Hz.

olliw42, 2020-06-06:
really many thx for looking at this, and answering.

I was not really expecting you would agree on the first shot. However I must admit, and I'm sorry to say this, that I find the answer disappointing. It's the sort of answer where someone doesn't really want to think through what has been said and just reject to avoid additional work. I also have the impression that you have not actually tried to put RC targeting and companion into actual implementation or put the gimbal manager on different components, because if you would have I can't get around to think that you would have seen the same limitations of the current scheme.

You actually contradict yourself.

You are claiming "No, I don't see why more can't be handled." but at the same time you can't tell me how to actually do RC targeting nor how to handle a companion. As regards RC targeting you say "The autopilot needs to be configured to use exactly one (or auto switch if that's required)" which however is nothing else than a breach of the manager concept since gimbal control is obviously managed by the autopilot configuration but NOT by the gimbal manager. As regards the companion all I hear is "This is an interesting one." plus some speculations.

If the current concept would be as easy as you say, and especially easier as what I suggested, I would have expected you could have told me now in few lines what I have to do. But you obviously couldn't. I take your words to actually confirm what I am saying.

In contrast, if we would sit next to each other, and not to have to communicate via this somewhat impractical text form, I would have explained you in 20 secs how my scheme works and you would have seen yourself in 5 secs how it expands to companion and RC targeting.

As regards the list I've given you say "it's mostly how I'm already thinking about it", but everything I read you saying is not along it as it's either not providing recipes for desired tasks, requires unspecified and dubious add-ons to the concept or just works around the gimbal manager. 

It is simply not possible to represent the multiple states of more than 2 clients with only 3 flags. I think that's as clear and logical as mathematics possibly can be.

Now, thinking about how to go forward, I have a strong feeling that if one would continue doing this per text form it would hardly converge in any reasonable time - I think we don't want another long gimbal thread like the old one, at least don't feel like it. So, I conclude it would best to get into contact e.g. with a zoom meeting (yeah, I can do this now LOL) if that would be a possibility.

Not sure if this is a helpful response, but let's see. 

:)

julianoes, 2020-06-06:
> It's the sort of answer where someone doesn't really want to think through what has been said and just reject to avoid additional work

I might have gotten tired as this was the last one I read through. I'll have another read through next week. Thx.

olliw42, 2020-06-07:
this is a very honest statement, and you have my highest respect. :+1: This was somewhat my impression, and I have great sympathy since I certainly agree and also know that altogether it's lots to read and digest. I would very much appreciate if you could give this one here a second read.

I'm not sure you've already tried to implement e.g. RC targeting support, or possibility of a companion as client. I believe I can truthfully say that I do have some practical experience. I do have a gimbal manager in my gimbal implementation as well as a gimbal manager in my autopilot code (of course only one is enabled via configuration, it's for testing and proof of concept), and do have a system consisting of a transmitter as gcs, an autopilot, a companion, and a gimbal.

cheers Olli

PS: even though it would massively add to the read, I'd like to mention that I started writing up in more plain words the gimbal manager concept. It's of course fluid, given that it's in the early stages of WIP, but it may shine light on my perspective (http://www.olliw.eu/2020/mavlink-gimbal-protocol-v2/). I certainly do not expect you to read this outpouring, but I thought I should mention.

julianoes, 2020-06-08:
Ok, I did brush over some of your arguments and did not really make myself clear. I can see that now. And I didn't mean it in a way that it is good enough (or too complex to change it) but rather that I agreed on the current shortcomings but was not convinced by your suggestions.

On a second read I now understand why you are saying that it only works if the gimbal manager is situated on the autopilot. The reason is that you can have RC gimbal commands conflicting with `GIMBAL_MANAGER_SET_ATTITUDE` coming directly over MAVLink. Bugger.

So how to deconflict all that?

As I understand your suggestion is to deconflict based on the flags set by the clients. The flags are a combination of origin (MISSION/GCS/COMPANION/RC/etc.) and intent (NONE/NUDGE/OVERRIDE).
The gimbal manager should then do the deconfliction based on whatever implemented rules which can be implementation specific.

Here are my concerns with this approach:
1. We encode the origin into the flags even though we already have the origin of a message, at least for most of the cases.
2. Due to the fact that origin and intent are mixed, the flags can explode in numbers, for instance a new intent is to be added. And you hinted at that with the flag type being potentially too small.
3. I would like the deconfliction rules to be at least somewhat guided by suggestions, otherwise it becomes the wild west and as a user/configurator you never know what is going to happen, especially if some (or many) of the gimbal managers might not be open-source.

> This is an interesting one. This would have to use override in order to override manual control, however, that means manual control can't override anymore. Therefore, my gut feeling is that the [...]

Basically, my suggestion was to do the deconliction not based on origin but priority. This means that the configuration is shared across all components sending commands and not defined inside the gimbal manager.
If you want sensible default priorities of components, we could even think of having some suggested values like:

Priority | Value
--- | ---
MIN | 0
GCS_MAVLINK | 90
RC | 100
COMPANION | 150
AUTOPILOT_MISSION | 200
MAX | 255

This would leave room for individual/bespoke setups as priorities of additional components can be chosen freely but still be compatible with any gimbal manager implemented (correctly).

I would see the priority added to all messages and commands (although I'm not sure how to add it to DO_GIMBAL_MANAGER_ATTITUDE yet as all 7 params are used).

---

> I'm not sure you've already tried to implement e.g. RC targeting support, or possibility of a companion as client. I believe I can truthfully say that I do have some practical experience.

We are currently working on many of these pieces: PX4, MAVSDK, QGC, companion computer gimbal manager, so things are likely to surface. The questions so far were about deconfliction/priorities (of course :smile:) and confusion around quaternions (or rather why?), as brought up by you in [this thread](https://github.com/mavlink/mavlink/issues/1376#issuecomment-637599606).

Btw. a zoom call is a good idea but I did have to think through this properly anyway first and put it in words. Let me know when is a good time for you.

olliw42, 2020-06-08:
many thx for getting back to this. And I think we actually are not very far from each other. 

I agree with your concerns as regards the flags, and I would not be against considering other schemes to achieve the very same. In fact the flags as they are have some more pain point which I haven't mentioned. I went with flags since they had been there now and I took it as indication of an decision to go with flags.

One point I however want to stress is that, in my opinion, going by priority cannot work and must fail for deconfliction. One must have the info about the origin. Priorities will of course be relevant elements for actually doing deconfliction, but this is then about the actual particular implementation, and not about the concept and messages itself. In addition, while it is correct that the origin is implicitly determinable in some cases, this is not so in all cases, which directly implies that if no mechanism is available to systematically keep the info about the origin these cases cannot be treated. An example would be RC targeting. Missions could be an example too.

I fully agree that it would be very helpful to have working suggestions for the deconfliction and even example implementations. However, this should be just this, examples and suggestions, but not predetermine or constrain the concept and protocol.

If you would find time for a zoom, this I guess would be the best (fastest) way forward. possible times:
today: now - 14:30, after 18:30
tomorrow: afternoon
wednesday: afternoon
(I'm in home office, so very flexible actually)

julianoes, 2020-06-08:
Ok, I've sent you an email.

olliw42, 2020-06-08:
@julianoes 

ausgerechnet jetzt scheint gmx zu haken

Join Zoom Meeting
https://uni-freiburg.zoom.us/j/91489424819

Meeting ID: 914 8942 4819
One tap mobile
+493056795800,,91489424819# Germany

Dial by your location
        +49 30 5679 5800 Germany
Meeting ID: 914 8942 4819
Find your local number: https://uni-freiburg.zoom.us/u/avffdB7d3

olliw42, 2020-06-09:
many thx for the discussion yesterday, I think it was highly useful

I realized that I have pushed my current gimbal manager implementation for the autopilot to my ArduPilot branch, so I'd like to share the link, just in case: https://github.com/olliw42/ardupilot/blob/BetaPlane4.0.4-v045-m20200526/libraries/AP_Mount/BP_Mount_STorM32_MAVLink.cpp. There is lots of cruft in the code because also other scenarios are supported and it is WIP anyway (e.g., its incomplete for missions), but I think you will have no issue with finding the relevant parts. The heart is _update_gimbal_manager_flags(). 

I realized that in one point I might not have been totally clear. I was emphasizing that it is important to know the source (client) of an override/nudge/none request, but think have missed to point out that the above flag suggestion is not a complete, self-contained solution. We touched the ambiguity in case of RC targeting, and concluded we can happily live with this, but it's also so to some extend for other clients. In my current gimbal manager implementations I thus use heuristics based on the source sys and comp IDs in a message, which possibly could be robustified/extended by also taking into account the mav type info from the heartbeat. This works fine for me, but is of course just an heuristics. In the above code the heuristics is in _determine_client(). Besides for deconfliction, the actually client is also needed in order to rule out that e.g. the GCS client (mistakenly or by purpose) sets e.g. the COMPANION override flag in its SET messages. So, the gimbal manager needs to sort out that only a GCS client sets a GCS flag and a COMPANION client a COMPANION flag, and so on. If the gimbal manager would not run on the autopilot, the heuristics would also include determining if the client is the autopilot, if the autopilot would be the client for MISSION.

that's of course just my current implementation, and might neither be acceptable nor a good example, but it might be an example.

julianoes, 2020-06-09:
> Besides for deconfliction, the actually client is also needed in order to rule out that e.g. the GCS client (mistakenly or by purpose) sets e.g. the COMPANION override flag in its SET messages.

Yes, I did get that part.

I'm tempted to argue that we have to assume that all components play correctly and we don't have to have additional checking. I see two drawbacks of applying these heuristics:
- It makes it harder to understand when setting up a system why some commands work and others don't. So it's tricky to debug unless you have the source of the gimbal manager and can add some printfs to see what is going on.
- It would have to be fully documented so that gimbal managers can implement it (correctly). At this point I usually argue for "keep it simple". 

---

I was wondering if the flags should be named after source (e.g. RC, COMPANION) or intent (e.g. PILOT, SMARTSHOT) and if this could make the deconfliction clearer. Also, the MISSION one is already an intent, I would argue.

julianoes, 2020-06-09:
@olliw42 more questions as I'm thinking more about the suggested flags:

1. The basic mode to "control" is to set OVERRIDE, right? In this case the naming would have to change because OVERRIDE signals an exceptional control, not normal operation. 
2. The OVERRIDE flag was basically a flag for anyone to raise the priority with the only difference that the priority goes from 0 to 1 instead of 0 to 255. By adding INTENT_OVERRIDE we are somewhat creating more levels of priority but also moving the priority evaluation / arbitration to the gimbal manager. And in order to avoid components from trying to game the system, the gimbal manager can do additional checks (heuristics) to check if component source and intend are valid. Do you agree with that?

olliw42, 2020-06-09:
> I'm tempted to argue that we have to assume that all components play correctly and we don't have to have additional checking. I see two drawbacks of applying these heuristics

> It makes it harder to understand when setting up a system why some commands work and others don't. 

two comments on the heuristics. 

First, I'd like to point out that the need for the heuristics is the direct consequence of not having properly represented the clients/source in the protocol! It's like building a bridge, but putting it on too weak pillars. Then only half as many cars may drive over it, which then needs a 2nd bridge to accommodate the traffic, plus additional roads, which will make the citizen to revolt, and so on ... . The heuristics is a direct consequence of not having done the job ideally elsewhere. 

Part of that is also that we actually have not even properly defined what clients/sources there are! We just talk about GCS, RC, MISSION, COMPANION, but each is a bit shady. So, maybe we should spend some more thought on what makes a client/source a client/source (this point will be relevant also in a point below). And maybe we should spend some more thought on what could be better alternatives to represent clients/source and the states. 

Second, it's actually funny, since I initially didn't had that heuristics and association to clients in the code, and in my frequent (re)configurations it let me run constantly into situations where I was wondering "what the heck!". So, it was exactly for that argument, to make setup simpler, which made me better understand my setup mistakes. So, the same argument lead me to exactly the opposite.

Now, this story has a much deeper fundamental point, for which there is no purely arguments-based resolution, but is a (subjective) balance of various arguments: Both the cooperative approach (= assume that all components play correctly) and the restrictive approach (= gimbal manager rules) do not prevent mistakes and situations where a user thinks "what the heck!", but they shift it to different types. The restrictive approach indeed makes it harder to understand when setting up a system why some commands don't work, as you point out, but the cooperative approach makes it harder to understand when setting up a system why some configurations (e.g. combinations of configuration parameters on the various components) don't work. The latter is what I ran into. 

As I said, I don't see that there would be a unique argument which would determine which one is the better, or simple to that matter. So it's a classical design decision which needs to be drawn.

My conclusion from the past experience with the (very lose) cooperative approach is a preference of the restrictive approach.

> So it's tricky to debug unless you have the source of the gimbal manager and can add some printfs to see what is going on.

>It would have to be fully documented so that gimbal managers can implement it 

This makes me come back to realizing that we haven't really well defined what clients/sources are. I would think if we would have that, a clear description of what GCS and MISSION and COMPANION in the context of the above flags are, I would think that it could and probably should become part of the specification to exactly say, XXXX is a GCS and is related to the GCS flags, and YYY is a MISSION and related to MISSION, etc.

That is, if we would have a specification for standard clients, the heuristics would then be unique (and not a heuristic in fact anymore), any gimbal manager would have to obey, and there would be no need for falling back to a cooperative approach in this part.

Note that we could have some "non-specificed" clients and associated flags (aka user flags).

> I was wondering if the flags should be named after source (e.g. RC, COMPANION) or intent (e.g. PILOT, SMARTSHOT) and if this could make the deconfliction clearer. Also, the MISSION one is already an intent, I would argue.

we touched this in the discussion, and I don't have a strong opinion here. 

I would however think, that if we would come up with specifications for the clients/sources, and the "heuristics" of how to determine them, that then it would appear more natural to also call the associated flags by the names of these clients/sources.

olliw42, 2020-06-09:
> 1. The basic mode to "control" is to set OVERRIDE, right? In this case the naming would have to change because OVERRIDE signals an exceptional control, not normal operation.

I think so too. I obviously took OVERRIDE since it was there, but better naming could be helpful.

One thing I would change for sure is to not have the negative-active behavior for MISSION.

> 2. The OVERRIDE flag was basically a flag for anyone to raise the priority with the only difference that the priority goes from 0 to 1 instead of 0 to 255. By adding INTENT_OVERRIDE we are somewhat [...]

I'm not sure. I need to more carefully read and think about what you are saying here :)

julianoes, 2020-06-09:
> The heuristics is a direct consequence of not having done the job ideally elsewhere.

So how can we make the job better "elsewhere". What about making the source clear by adding the field `GIMBAL_MANAGER_SET_ATTITUDE.actual_compid_source`?

olliw42, 2020-06-09:
> So how can we make the job better "elsewhere". 

I'll think about it too, but have little to say now :)

olliw42, 2020-06-09:
so, some thoughts. I could see 3 general approaches. They effectively all try to achieve the same thing.

* by properties: The client/source would be identified by properties such as the sys_id, the comp_id, the mav_type and so on. This is essentially my current heuristic approach. 

* by separating set and status flags: Here the flags which would be carried in the SET messages and command would not be identical to the flags transmitted by the gimbal manager in its STATUS message. With some mild conditions and some mild cooperativity this would prevent that a client would sets the wrong flags (I will elaborate on this further below)

* by client id: Here the data fields would be extended by some who-am-I fields.  With some mild conditions and some mild cooperativity this would prevent that a client would sets the wrong flags (I will elaborate on this further below).

With mild cooperativity I mean that a client must be so friendly to identify itself correctly, but not the stronger level of cooperativity that it also _**behaves**_ correctly. Such a mild cooperativity is the situation we have e.g. with the component IDs and autopilot & mavtype heartbeat fields. If a MAVLink component hasn't set them appropriately it likely would screw up the network. E.g. a component with autopilot component ID but GIMBAL mavtype would likely confuse many implementations. However, there is no mechanism in MAVLink whatsoever to alert or prevent such a thing to happen.

I now realized that it is the cooperative behavior thing which I would be opposed to, but that a mild cooperativity I would find totally acceptable and meaningful, and that I in fact would fully agree with you that it wouldn't make sense to install such a mechanism, like it doesn't for the component identification.

With that distinction, and accepting mild cooperativity (which I do), the "only" problem left over is the fact that with the current flag system it is not so easy to prevent that a client mistakenly sets a wrong flag. This is because the request flags and the status flags are identical. Therefore, each client must painstakingly keep track of ALL flags in order to not mistakenly send outdated flags, which means nothing else then that it would try to set flags which it isn't the client for. Even if the client tries to painstakingly keep track, it nevertheless can (easily?) happen because there is a significant time gap between when request are send from the clients to the gimbal manager and when the gimbal manager reports the status. This maybe can be reduced, but not avoided. 

This is essentially the long version of 

> Besides for deconfliction, the actually client is also needed in order to rule out that e.g. the GCS client (mistakenly or by purpose) sets e.g. the COMPANION override flag in its SET messages.

> Yes, I did get that part.

The above approaches prevent such race conditions from happening, and all in one way or the other build on distinguishing request (aka participant raises hand) from status (aka chairman gives participant asking time).

With this type of mild condition, the gimbal manager then still can rule and govern and all that, but we wouldn't have that clumsiness for identify a client. 

It could well be that this is what you were saying all along, and I just didn't got it :).

---------------

There are several ways of how to put that into a protocol, by flags only, by a mixture of flags and fields, and so on. I am not really decided, but want to suggest as food for thought this:

in the SET messages/command we have flags

REQUEST_OVERRIDE
REQUEST_NUDGE
SET_RC_OVERRIDE
SET_RC_NUDGE
CLIENT_GCS
CLIENT_MISSION
CLIENT_COMPANION
CLIENT_1
CLIENT_2

in the gimbal manager STATUS messages we would have exactly the same flags as suggested in the above to report the status for 

MISSION_NUDGE
MISSION_OVERRIDE
GCS_NUDGE
GCS_OVERRIDE
COMPANION_NUDGE
COMPANION_OVERRIDE
RC_NUDGE
RC_OVERRIDE
CLIENT1_NUDGE
CLIENT1_OVERRIDE
CLIENT2_NUDGE
CLIENT2_OVERRIDE

The mild condition would be that a client only sets its client flag, which for the client is easy to do and I think is fairly acceptable. 

I'm not totally settled with the RC targeting thing. After the discussion with you I totally agree that it doesn't make sense to have a blown up mechanism to cover every single unlikely scenario, but we should have basic support. The issue with RC is that it is conceptually different to the other clients in that it cannot raise its hand itself, but that it are the other clients which raise its hand on its behalf. For instance, the GCS may set RC nudge, but also a situation may exist where a companion may set RC nudge. But there is no such thing as that RC would say by itself that it wants to nudge. That's why there are RC flags for each client to set, but there is no CLIENT_RC flag. But I'm not sure this is really covering the basics properly. You also mentioned the situation of dual-operator control. This would have to be better thought through. (I guess I would do it by attempting an implementation).

Cheers, Olli

julianoes, 2020-06-09:
> Therefore, each client must painstakingly keep track of ALL flags in order to not mistakenly send outdated flags, which means nothing else then that it would try to set flags which it isn't the client for. 

Ok. I don't understand why the client needs to keep track of the flags. Could you make an example to illustrate that?

olliw42, 2020-06-09:
assume current flags
assume a GCS sending SET_ATTITUDE with flag GCS_OVERRIDE, and assume a COMPANION sending SET_ATTITUDE with flag COMPANION_OVERRIDE (thus only companion should control the gimbal, the SET_ATTITUDE from the GCS should be ignored by the gimbal mananager)
to make it simple, assume the GCS forgets to keep the correct COMPANION flags

the GCS thus sends SET_ATTITUDE with GCS_OVERRIDE but all COMPANION flags set to zero, thus the gimbal manager will unset the COMPANION_OVERRIDE flag and accept the new attitude and send it to the gimbal
:
few ticks later the COMPANION sends SET_ATTITUDE with COMPANION_OVERRIDE, thus the gimbal manager will set the COMPANION_OVERRIDE flag and accept the new attitude and send it to the gimbal
:
few ticks later the GCS sends SET_ATTITUDE with GCS_OVERRIDE but all COMPANION flags set to zero, thus the gimbal manager will unset the COMPANION_OVERRIDE flag and accept the new attitude and send it to the gimbal
:

gimbal is confused as it receives conflicting info 

if you think it is too simple to assume that the GCS forgets to keep the correct COMPANION flags, you are right, but since it is not updated at infinite rate, situations where it is outdated and false flag changes are triggered and wrong attitudes send to the gimbal occur frequently. In my experiments I could observe all sorts of "funny" things happening, I could even see persistently oscillating flags, even though the gcs was sending permanently the same gcs flags and the companion the same companion flags

the problem obviously is that the GCS also sends COMPANION flags, and the COMPANION also GCS flags (aka one participant raises the hand of another participant)

with checking the client/source in the handling of incoming flags, the gimbal manager is able to accept only that flags in each SET_ATTITUDE which belong to the client/source. That is, in the SET_ATTITUDE from a GCS it will only accept the GCS related flags but ignore the COMPANION related flags, and vice versa.

with that, all issues go away and it all works perfectly fine, i.e. in the above scenario always only the attitude from the COMPANION will be send to the gimbal. Also all transitions in the flags are always handled fine. I.e. it works fine - by construction - in all situations, at every instant in time.

julianoes, 2020-06-10:
Ok thanks for explaining that. This was a misunderstand then. I never thought that a component has to keep track of the flags. All a component has to do is set its own flag. So, as in the above example the GCS sets GCS_OVERRIDE and the COMPANION sets COMPANION_OVERRIDE.

Then, the gimbal manager looks at all the incoming messages and determines which one is the one it attributes the highest priority. Of course, it has to assume some sort of timeout, e.g. that a SET_ATTITUDE message needs to arrive at least every second, otherwise a controller would be determined to be timed out and it can fall back to the next one.

With that scheme sending the messages becomes simple for all components and the chance of bugs/setting flags incorrectly is quite low, I would think.

olliw42, 2020-06-10:
I disagree. As I have tried to explain these situations do NOT occur because e.g. the GCS purposefully wants to play bad, they occur even if the GCS always tries to play good! They occur because of the time-discrete nature of the information flow of the status fields! And this is not just a theoretical discussion but I believe to have exactly this seen in my practical tests.

I am sure you will observe similar issues in practical situations.

The time pooling scheme you suggest I actually also had considered, but it won't work either since it does not prevent that a false participant raises the wrong hand. It just needs to happen that the last message the gimbal manager has received before sending his status is a SET_ATTITUDE from the GCS with the COMPANION flag set down ... (and don't think an AND would do it, the opposite requiring an OR also exists).

The only way I would see to overcome without change in the message is by sending the gimbal manager status message at ultra high rate, but I have ruled out this as an option.

There is no way out of the logical catch besides correcting the logic. Pl keep the chairman example in mind. :)

olliw42, 2020-06-10:
if it helps you better, consider this situation: You have two MAVLink components which each send heartbeats, but now assume that you DO not know the source ids. Which of the components has just armed? You can't tell, unless you know the source. That's why the MAVLink messages carry the source. You can create infinitely many situations there one runs into the same kind of issues as the gimbal manager when there is incomplete information flow. And the solution is always the same, to achieve complete information flow. Not fancy time-based concepts. ;)

julianoes, 2020-06-10:
Right, so I'm not saying not to also take the source into consideration, otherwise I can see how it doen't work at all. And my assumption is that the source is correct (as for heartbeats it needs to be), so we agree on that.

Now regarding time based concepts, I think, you need one of the two options:
1. A timeout mechanism: So after a timeout of e.g. 1-2 seconds has passed a component is no longer in control, and you can switch to the previous/next lower one.
2. Explicitly ending control using e.g. a NONE flag to signal that now control can be passed on.

olliw42, 2020-06-10:
> Right, so I'm not saying not to also take the source into consideration, otherwise I can see how it doen't work at all. And my assumption is that the source is correct (as for heartbeats it needs to be), so we agree on that.

perfect. Your assumption is what I called mild cooperativity. So, we totally agree on these points.

The necessary consequence is however that there must be the information in the SET messages which allows the gimbal manager to uniquely determine the source. The source system&component IDs in the message alone ONLY allow a heuristic identification. Thus, there must be something in addition in the messages.

I'm not yet sure we agree on this, and if so, on how to put this information into the SET messages.

> Now regarding time based concepts, I think, you need one of the two options:
>
>     1. A timeout mechanism: So after a timeout of e.g. 1-2 seconds has passed a component is no longer in control, and [...]

I agree that there must be a mechanism to not only raise the hands but also to lower it again. 

I also agree that a timeout is a technically working mechanism. I however advise against a time-out mechanism in order to ensure proper behavior (in normal operation conditions, i.e. no failures). The only use case I ever have seen is to not having to do something properly (again, not talking about catching error conditions).

In this case, since there have to be flags anyway, I don't see any advantage in a timeout. It in fact would be just an additional mechanism put up in addition to flags. In terms of "simple" or "not simple" I can't see how this possibly would make things simpler. 

Again, a timeout can be a very valid method to catch error conditions, such as then a client somehow dies. But this is clearly a very different purpose, and has nothing to do with clearing the client state. I would not assume that it is your common coding practice to use the error mechanism for clearing a state in normal operation. I don't do that at least.

You don't need a NONE. Neither OVERRIDE nor NUDGE raised means NONE. That is, the flags as suggested allow to do exactly your point 2, i.e., explicitly clear control. It may be that this is however not how you want it to have, but want to have three flags between which the client toggles. Considering simple vs not so simple and exhaustion of flags I guess I would not go with a none flag. But I actually wouldn't mind.

:)

julianoes, 2020-06-10:
Ok, thanks for the explanations. It's getting clearer to me and I think we're converging (or talking about the same thing)!

> In this case, since there have to be flags anyway, I don't see any advantage in a timeout.

I do:
1. Unfortunately in reality clients die all the time :). Or e.g. we kill them to see if they are the ones causing havoc with the gimbal.
2. RC or MAVLink is lost because of weak signals, again it's just a natural timeout.
3. Often, when e.g. implementing gimbal control on a companion computer you would send control messages when you want to move the gimbal. Once you're finished, you probably exit the program or at least don't bother to keep sending messages.
4. Messages (not commands) can be lost, so sending the message once with NONE (or whatever to signal none) is not enough. Instead you either have to send it several times (how many?) or until you see the change reflected in the status.
5. If everyone should always keep sending the message, then this means quite a bit of overhead for nothing really.
6. From what I've seen the normal way in MAVLink is to send a message if there is information to be sent, and just not send it if there is no valid information. I would agree that that's not always the right approach, it's just quite common.

> I would not assume that it is your common coding practice to use the error mechanism for clearing a state in normal operation.

As said above, it sort of is for many MAVLink things. For instance offboard control.

What I fear is that we'll have MAVLink nodes coming and going. Like for instance a tablet with controls would connect and a video operator would operate the camera for a bit. Then instead of pressing some button in the UI "hand control back to pilot" the workflow is likely to be to just exit the app, or worse, switch off the controller. Again, the done state might not be transmitted in time which mean that either an error timeout needs to trigger, or the pilot needs to have some sort of "emergency/test override" button to get back control.

What I'm trying to say is that in theory I agree with you that a positive stop signal would be the preferred way, however, in practice, I believe a simple timeout would work much better and more intuitive.

olliw42, 2020-06-10:
ok. I see your points and they are of course reasonable. Thx for telling.

You presented 1. and 2. as either or. But why not have them both, they don't really exclude themselves.

One could specify that a client should clear it's flags when it is done, but in addition specify that there should be the time out of few secs for the gimbal manager to clear the flags. 

In some sense you are very right, since in practice there would be a timeout anyways, maybe just not few secs but maybe a dozens of secs for failure protection, so it's somewhat pointless to argue over 2 secs or 10 secs.

I guess my point with (also) having 2. is that it just would feel very wrong to me if I would not actively lower the flags in my client implementations, but would rely on the gimbal manager to do this. Would feel like a bug to me (my hairs would stand up). So, I would want to do that, but since I also want to be compliant with the spec and not brew my own beer (even though home-brewed beer certainly tastes the best) I would want to see that in the spec. I could imagine that I'm not the only one with this kind of coding philosophy.

Maybe the have-it-both would be a resolution.

olliw42, 2020-06-11:
ähm, I just recalled: 
didn't you mention in the zoom chat that the timeout mechanism could be a problem for mission, as it wouldn't emit regularly?

julianoes, 2020-06-11:
That's a good suggestion. Why not encourage it but deal with the practical issues with a decent timeout. This also allows a (nice) implementation to pass control very quickly without a 1-10 second delay which is nicer! Good we agree on that.

Now do we agree that with this setup a client does not need to copy the status flags of others?

As an example:
1. GCS tablet gimbal control is being streamed with PILOT_CONTROL flag set at 20 Hz.
-> Gimbal manager gives control to GCS.
2. Then the companion takes over for a smart shot by sending SMARTSHOT_CONTROL flag set.
-> Gimbal manager gives control to companion.
3. Then the companion sends NONE flag after the shot is done (or it crashes there is a timeout):
-> Gimbal manager gives control back to GCS.

The companion does not need to acknowledge that there is a GCS also trying to control the gimbal, right? It just sends what it wants.

---

didn't you mention in the zoom chat that the timeout mechanism could be a problem for mission, as it wouldn't emit regularly?

Correct. So for commands we cannot use a timout. We have to keep it until another command is sent to reset the control. This is what I meant when I said "not giving back control after a command is a bug and needs to be fixed, one by one". For PX4 this means that whenever you exit mission mode, it would send a command to set gimbal mission control to NONE, like a destructor that cleans up whatever was around.
I'm aware that that's not perfect and might lead to issues but I do not have a better suggestion on how to approach that.

olliw42, 2020-06-11:
> That's a good suggestion. Why not encourage it but deal with the practical issues with a decent timeout. This also allows a (nice) implementation to pass control very quickly without a 1-10 second delay which is nicer! Good we agree on that.

great. :)

> > didn't you mention in the zoom chat that the timeout mechanism could be a problem for mission, as it wouldn't emit regularly?
> 
> Correct. So for commands we cannot use a timout. We have to keep it until another command is sent to reset the control.

two clarifications, pl

you don't mean "for **commands** we cannot use a timout" but mean "for **mission** we cannot use a timout", right? I guess we want to make any client being allowed to send any SET message and also the SET command, and not reserve the command for only the mission client.

I guess we want not only to "encourage" to reset the flags when done (aka set none), but want to specify that as the primary mode for clearing. But we of course still keep the 1-2 sec timeout in the gimbal manager's handling of the flags. This makes handling the flags unique across all clients, including mission, and probably will produce less confusion.

I actually think there can be simple solutions. With the MISSION I see two situations: (1) gimbal manager is on the component also doing the mission. In that case there can't be a loss of a message or such and the gimbal manager also can always know the end of a mission, thus there shouldn't be any problem at all. (2) gimbal manager is not on the component doing the mission. In that case the mission component would send SET messages/commands to the component hosting the gimbal manager, and it should not be difficult to code the mission component such to send what is needed for reliable operation. E.g. it simply could send at regular intervals for as long as the mission is not ended, in which case even the timeout would work, or send the final NONE 4 times, or whatever. So, I think, with the mechanism we just agreed on it should all be solvable with not much  hickups.

> Now do we agree that with this setup a client does not need to copy the status flags of others?

I can't comment on this without knowing the actual flag/source mechanism.

with things as currently in common.xml, I would not agree.

if we would have a mechanism which unambiguously would reveal the client/source of a SET message/command to the gimbal manager, and if the gimbal manager would be required to take the client/source into account such as to ignore the flags of all other clients, then the answer is obviously: yes

if it were such, it in fact would make life for the programmer of the clients much easier, since - by construction - they only would have to be concerned about their own flags

RC would need some more thought

julianoes, 2020-06-12:
> you don't mean "for commands we cannot use a timout" but mean "for mission we cannot use a timout", right?

I guess so, yes. However, generally commands in MAVLink don't need to be repeated so it would have to be explicitly stated if that's required.

> I can't comment on this without knowing the actual flag/source mechanism.

The actual flag/source mechanism as I see it right now would be something like:

```
NONE
MISSION_CONTROL
COMPANION_CONTROL
COMPANION_NUDGE
PILOT_CONTROL
PILOT_NUDGE
OPERATOR_CONTROL
OPERATOR_NUDGE
CUSTOM1_CONTROL
CUSTOM1_NUDGE
CUSTOM2_CONTROL
CUSTOM2_NUDGE
```

olliw42, 2020-06-12:
> I guess so, yes. However, generally commands in MAVLink don't need to be repeated so it would have to be explicitly stated if that's required.

I just would not like to have it artificially restricted. I note that MAVLink is not very consistent in this part, i.e., various commands exist which are send repeatedly.

> The actual flag/source mechanism as I see it right now would be something like:

this is just flags, but there is no source mechanism whatsoever in here yet. (the gimbal manager needs to know who has send flags in order to filter out which flags to only consider). So, what is your idea for the source mechanism?

NONE is probably zero flag

it's not directly obvious to me what distinguishes PILOT and OPERATOR, but you will clarify Iguess

:)

julianoes, 2020-06-12:
> various commands exist which are send repeatedly.

I'm only aware of the (old) gimbal commands so far. Which others are there?

> this is just flags, but there is no source mechanism whatsoever in here yet.

That's where we still don't agree. That's only if the clients send a flag that they are not allowed to send. If they behave no further heuristic checks are required. This is the point I was missing above.

> NONE is probably not an extra flag but means flag = 0

Right, so then you have to check if all other flags are not set, and if so it's NONE.

> it's not directly obvious to me what distinguishes PILOT and OPERATOR

Sorry, I was wondering if I should call it `CAMERA_OPERATOR` or `GIMBAL_OPERATOR`.

olliw42, 2020-06-12:
> That's where we still don't agree. That's only if the clients send a flag that they are not allowed to send. If they behave no further heuristic checks are required. This is the point I was ~missing~ not understanding above.

there is nothing to agree on. So, let's test it:

you are the gimbal manager

here on my side there are two clients

one of the clients is now sending you a SET_ATTITUDE message with this flags 

COMPANION_CONTROL & PILOT_NUDGE

now, dear gimbal manager, please make the right decision. Can you pl tell me which of the flag you need to ignore in handling the message?

olliw42, 2020-06-14:
you may find this example useful too:

same situation, you are the gimbal manager, here on my side there are two clients

you are in state COMPANION_CONTROL & PILOT_NUDGE

one of the clients is now sending you a SET_ATTITUDE message with this flags

flags = 0

now, dear gimbal manager, please make the right decision. Can you pl tell me which of the flag you need to clear in handling the message?

it's a variation of the same point. The examples should not only make you observe it's not working but also make you think about WHY it's not working. Once the why is clear also the possible solutions are clear. You may then reread the thread from post #1, will find the why described repeatedly, and also find several solutions be described.

:)

julianoes, 2020-06-15:
> COMPANION_CONTROL & PILOT_NUDGE

The way I'm thinking about this is that no component can ever set flags of two different intent/source. So if this one is sending COMPANION and PILOT at the same time that does not make any sense and is a bug, and might as well be rejected.

> one of the clients is now sending you a SET_ATTITUDE message with this flags
> flags = 0

I'm clearing whatever the source did before which now has flags 0, and fallback to whoever else is sending anything (with active flags).

olliw42, 2020-06-15:
> The way I'm thinking about this is that no component can ever set flags of two different intent/source. 

that's why I've added the 2nd example, to make you see that this is not the point ;)

> I'm clearing whatever the source did before which now has flags 0, and fallback to whoever else is sending anything (with active flags).

?????

I'm lost. And I'm a honestly a bit speechless now. 

You are not answering my question which flag you need to clear. So, just pl tell which of the two you need to clear, COMPANION_CONTROL or PILOT_NUDGE, and explain to me how you did arrive at that conclusion. Especially please explain me how you can infer which source is meant which is what you need to know in order to do what you just said yourself with "whatever the source did before". Just say either 

* the source was COMPANION and thus I need to clear COMPANION_CONTROL

or 

* the source was PILOT and thus I need to clear PILOT_NUDGE

you constantly argue as if you would know the source, but you don't. So, pl proof that you are right and just tell me the source.

olliw42, 2020-06-15:
to make it clear, I am asking you a simple question:

you are in state COMPANION_CONTROL & PILOT_NUDGE

you are receiving a SET_ATTITUDE message with flags = 0  

you have now only one out of exactly four options to react:

1. you don't do anything and stay in state COMPANION_CONTROL & PILOT_NUDGE

2. you clear COMPANION_CONTROL and assume the state PILOT_NUDGE

3. you clear PILOT_NUDGE and assume the state COMPANION_CONTROL

4. you clear both flags and assume the state 0

So, your answer is as simple as to tell me 1, 2, 3, or 4. As simple as that. But no longer answer is accepted.

julianoes, 2020-06-15:
> you constantly argue as if you would know the source, but you don't.

Of course I know the source.

> I'm clearing whatever the source did before which now has flags 0, and fallback to whoever else is sending anything (with active flags).

So for this case. I remember that COMPANION_CONTROL came from source companion and I remember that PILOT_NUDGE came from source autopilot.

So if one of these two sources sends me none, I know which one to clear based on my memory of what source did what. Does that make sense?

My point was that the autopilot does not need to include the COMPANION_CONTROL in its messages and neither does the companion need to include the PILOT_NUDGE. And from that I draw the conclusion that it is very unlikely that the autopilot will ever (by accident) send COMPANION_CONTROL. And because the autopilot is unlikely to make such a (silly) mistake, we don't need to use heuristics to filter out wrong messages.

So maybe to say it differently. I keep track of what source sets what intent but I don't make any assumptions about which source is allowed what intent. That way if a source stops again, I know what to clear.

olliw42, 2020-06-15:
you still have not told me how you determine the source. 

You now just say that you have remembered which one is the companion source, but you don't even try to explain how you possible have determined what the source is in this first place. You are constantly claiming that you have determined the source, but refuse to explain how you ever get or got to this information. At some point in time you must have had come to the insight of the source.

I actually start to strongly suspect that you confuse source with the source IDs in the message. We have ruled this out in the above as a non viable path, i.e., that's not an option.

If you want to uphold your line of reasoning, you need to explain at some point how you did ever got at the source information. So far you don't.

> My point was that the autopilot does not need to include the COMPANION_CONTROL in its messages and neither does the companion need to include the PILOT_NUDGE. And from that I draw the conclusion [...]

and since I was afraid that what is the irrelevant point will be confused with what is the relevant point I drafted the second example, which removed the irrelevant point. :)

I think the discussion is trapped, and I actually kind of can't believe it but also don't know any way forward anymore. I start to suspect one needs to wait for until you have a practical implementation which allows you to test in care the situation of two clients in practice. I just very strongly hope that the flags are not been written in stone before you reach that point, so that any later insight can still be coped with (and this insight will come sooner or later).

:)

julianoes, 2020-06-16:
> I actually start to strongly suspect that you confuse source with the source IDs in the message. We have ruled this out in the above as a non viable path, i.e., that's not an option.

Yes, that's what I'm assuming. Sorry that I was not clear about that.
I'm still thinking that this should generally work and we can ignore the corner cases. If that's absolutely incorrect, then I guess we need to still add the intent to the NONE so that you know who gave up control, e.g. PILOT_NONE if the pilot releases control.

So this would then mean that the autopilot does not need to send both flags: COMPANION_CONTROL & PILOT_NUDGE, it only sends PILOT_NUDGE because that's what it is concerned with.
In the same way the companion would send COMPANION_CONTROL.
Does that make sense?

olliw42, 2020-06-23:
hey Julian

In my opinion the old gimbal protocol suffered from two issues: poor messages AND a poor concept. Accordingly, in my opinion, a new protocol should provide better messages AND a new better concept. I genuinely think that one should learn from the past, and not repeat the mistakes of the past, if possible. This made me suggesting the gimbal manager concept. However, the above, also labeled "gimbal manager", has little to do with it, but is merely the old recipes and habits wrapped into new messages ("Alter Wein in neuen Schläuchen").

I don't see the big benefit of the "gimbal manager" stuff in it's current flavor. Auxiliary procedures are introduced for no need I can see, but with the disadvantage of auxiliary procedures. Stateness is introduced for no need I can see, but with the disadvantage of stateness. Various use scenarios are not addressed and it's unclear (to me) how they would fit in. It is said to be simpler but not easier to code or to explain. The discussion might have slightly improved things but I realize that the basic statement in the opening post still applies, that it sure works for what it is intended for but that the scope is way too narrow. However, this was true for the old protocol too, and in fact everything could and can be achieved with it too. And especially ALL of the above can equally well be achieved with the old MOUNT_CONTROL message and its mode flag by also just tracking the source id of the message. No better concept here, but old recipes reused. 

Since this is where you guys want to go, I would, for instance, recommend to remove all the gimbal manager flags. They don't serve any really useful purpose. All you need is a mode field in the SET functions so that each component=source can say O or N or 0. This would remove much clutter and make code and use easier, but achieve the very same. And probably other simplifications streamlining the messages towards this other concept are possible.

In contrast to the "gimbal manager", the gimbal device messages, with the two PRs https://github.com/mavlink/mavlink/pull/1399 and https://github.com/mavlink/mavlink/pull/1396, which I hope will be merged rather sooner than later, form a nice set of messages and are a very significant step forward, IMHO. They will be of lasting value, I believe.

I have to face it: Besides my efforts I was not very successful in my "mission" to bring in new terms of how to think about the topic and issue, and let go with the ever same solution approaches. I've underestimated the resilience of habits (also applies to me). I observe that this is also so for my suggestion of the COMPONENT_INFORMATION and the underlying vision. I am - in all honesty - shocked to see what is been made of it and have a hard time to not think that this is the greatest nonsense I have seen for a while. Here too, I failed, i.e., with each of my "missions" I've failed. I realized that it would have been better for the time budget of all sides if I just would not have suggested anything, and having recommended COMPONENT_INFORMATION I now in fact regret. I genuinely believe that things can become better, but not by non-action, so I made an effort. I only can apologize for whatever loss in time this has produced.

I wish to say that even though we may not fully agree on the outcome, it was a real pleasure to work with you on this. I say this without any irony but in all earnestness. But I think it's time for me to acknowledge my failure and let go.

Cheers, Olli

julianoes, 2020-06-23:
Not sure how to answer to this except thanks for all your input and work. And I believe we have better messages and a better concept and something clearly much better.

> The discussion might have slightly improved things but I realize that the basic statement in the opening post still applies, that it sure works for what it is intended for but that the scope is way [...]

I think by adding the intent we are a lot better than the current (narrow) scope and can cover most cases. (And yes there will always be more corner cases to think of but I don't think we need to aim to spec everything, plus with CUSTOM_INTENT_1/2 we can provide some flexibility.

> Since this is where you guys want to go, I would, for instance, recommend to remove all the gimbal manager flags. They don't serve any really useful purpose. All you need is a mode field in the SET functions so that each component=source can say O or N or 0.

Which flags would you remove and why? I'm confused here.

> ... and having recommended COMPONENT_INFORMATION I now in fact regret.

I disagree. I believe it will be useful. And since you initiated it some of the praise goes to you, if you want it or not ;).

Anyway, I'll try to put what we have discussed in this thread since I believe that it makes sense and addresses the issues that will inevitably come up as you rightfully pointed out.

olliw42, 2020-12-05:
discussion has kind of settled, and the issue went stale anyway, so I close it. :)

closed 2020-12-05 by olliw42