mavlink/mavlink #1404: 'mocha test' fails in pymavlink

TSC21, 2020-06-14:
This might be an issue to bring to pymavlink itself, but the same test is called in Travis in the Mavlink repo. The error is the following:

```
mocha test
/home/travis/build/mavlink/mavlink/pymavlink/generator/javascript/node_modules/mocha/lib/esm-utils.js:6
    return import(url.pathToFileURL(file));
           ^^^^^^
SyntaxError: Unexpected token import
    at createScript (vm.js:80:10)
    at Object.runInThisContext (vm.js:139:10)
    at Module._compile (module.js:617:28)
    at Object.Module._extensions..js (module.js:664:10)
    at Module.load (module.js:566:32)
    at tryModuleLoad (module.js:506:12)
    at Function.Module._load (module.js:498:3)
    at Module.require (module.js:597:17)
    at require (internal/module.js:11:18)
    at Object.<anonymous> (/home/travis/build/mavlink/mavlink/pymavlink/generator/javascript/node_modules/mocha/lib/cli/run-helpers.js:18:27)
npm ERR! Test failed.  See above for more details.
```

It might be a dependency problem with `babel` that needs to be solved: https://stackoverflow.com/a/37076911.

hamishwillee, 2020-06-16:
@peterbarker This something you know about?

stephendade, 2020-06-17:
I've just tried this with nodejs V12 and running ``npm test``. All works fine.

What version of nodejs are you running?

TSC21, 2020-06-17:
> I've just tried this with nodejs V12 and running `npm test`. All works fine.
> 
> What version of nodejs are you running?

This is happening on this repo CI (in Travis to be specific). I have not tested this locally.

hamishwillee, 2020-06-18:
Hi @stephendade 
Thanks! This is the generic test suite, run in Travis. This is the file [/scripts/test.sh](https://github.com/mavlink/mavlink/blob/master/scripts/test.sh) that creates libraries for the dialects and test code and runs it all. 

Travis runs this using Xenial (ie a cut down version of Ubuntu 16.04), which according to the docs supports [node 11](https://docs.travis-ci.com/user/reference/xenial/#javascript-and-nodejs-support).

Do you think you try run the test script this time and see if that works? If not, perhaps dropping to Node 11 and trying that?

Note, that mocha appears to depend on node 14. I very much doubt we have explicitly created such a dependency, so something else must have added this "under the hood".

TSC21, 2020-06-18:
Fix to CI in #1407. Now running a test pipeline to Python 2.7-3.8 and Node 12-14 in Ubuntu Focal 20.04. All passing.

closed 2020-06-18 by hamishwillee