mavlink/mavlink #1411: Issues implementing GENERATOR_STATUS

peterbarker, 2020-06-23:
I'm attempting to move a PR against ArduPilot to using `GENERATOR_STATUS` (https://github.com/ArduPilot/ardupilot/pull/14662)

I've come across these issues so far:
 - the type on the RPM field may not cover all generators; a gas-turbine-based generator may spin faster than 65,535 RPM
 - the enumeration values are missing `MAV_`, meaning some relative generic defines are floating around
 - the device I'm currently working with provides several fields that might be desired on the GCS:
   - device uptime in ms
   - time until next scheduled maintenance
 - several flags don't translate into the current flags-and-fault-flags
    - start inhibited
    - maintenance required
    - battery voltage is low

Any suggestions on how I might better integrate with this message would be appreciated.

auturgy, 2020-06-23:
It’s a WIP message, and I haven’t seen a reference implementation: so subject to change. Feel free to make a PR. @hamishwillee I might not be able to make the call tomorrow but would be good to get some guidance

hamishwillee, 2020-06-24:
@peterbarker Awesome to hear this is getting some testing. As James says, the right way to progress this is to create a PR that amends the message in the way you think best and invite the key stakeholders to comment.

How about you set up a Skype discussion with Leonard (@lthall) and @CarlEngel who came up with the original message and agree how you want to progress. Or I can set one up and invite you all in if you would prefer?

peterbarker, 2020-06-24:
On Tue, 23 Jun 2020, Hamish Willee wrote:

> I can set one up and invite you all in if you would prefer?

That'd be good, thanks!

CarlEngel, 2020-06-24:
I will just address some of my thinking in the issues. 

- The RPM field was more intended for the electrical generator. In turbine generators, the electrical generator would generally not be directly coupled to it. It would be geared down or have a secondary shaft running at a lower RPM. If the RPM of the turbine was needed then there are other messages that could be made. For example I was intending on using it with the `EFI_STATUS` message.
- The values should probably get the MAV_ added to them so this doesnt cause any problems. That was an oversight.
- For the last two, it was left as a WIP so that things could still be added. I just didnt want to start making the definition too long. The enum flags could easily get added.

hamishwillee, 2020-06-24:
@CarlEngel Great. Do you think you could join the Skype chat I set up with you? I added Peter to it and he wants to bring in some others.

hamishwillee, 2020-06-29:
@peterbarker Looks good to me. @CarlEngel any comment?

hamishwillee, 2020-07-08:
This has now been resolved

closed 2020-07-08 by hamishwillee