mavlink/mavlink #1415: Optimization of param_ext

lukegluke, 2020-06-28:
I understand that it's immensely impossible to release, but anyway I want to notice that if **char param_value[128]** was the last in mavlink_msg_param_ext_value (and ext_set) structures, it would allow to decrease size of packets in PARAM_EXT protocol, because packets can be (like in px4) trimmed efficient.
This is very useful for structures parameters (especially including strings) - I don't know if anybody use such things, but it is not a big deal to add struct parameters support over mavlink to px4.

```
    <message id="322" name="PARAM_EXT_VALUE">
      <description>Emit the value of a parameter. The inclusion of param_count and param_index in the message allows the recipient to keep track of received parameters and allows them to re-request missing parameters after a loss or timeout.</description>
      <field type="uint16_t" name="param_count">Total number of parameters</field>
      <field type="uint16_t" name="param_index">Index of this parameter</field>
      <field type="uint8_t" name="param_type" enum="MAV_PARAM_EXT_TYPE">Parameter type.</field>
      <field type="char[16]" name="param_id">Parameter id, terminated by NULL if the length is less than 16 human-readable chars and WITHOUT null termination (NULL) byte if the length is exactly 16 chars - applications have to provide 16+1 bytes storage if the ID is stored as string</field>
      <field type="char[128]" name="param_value">Parameter value</field>
    </message>
```

hamishwillee, 2020-06-29:
@lukegluke Correct. Yes, [PARAM_EXT_VALUE](https://mavlink.io/en/messages/common.html#PARAM_EXT_VALUE), [PARAM_EXT_SET](https://mavlink.io/en/messages/common.html#PARAM_EXT_SET) and [PARAM_EXT_ACK](https://mavlink.io/en/messages/common.html#PARAM_EXT_ACK]) should indeed all have had the char array last.

But as you say, this is now used extensively, and we're unlikely to get approval for a breaking change. 

I am therefore going to mark as closed, wontfix.

@dogmaphobic FYI. Something to keep in mind if it ever becomes feasible to update the param ext protocol.

closed 2020-06-29 by hamishwillee

dogmaphobic, 2020-06-29:
It makes sense, but as you said, it's too late for that now. The `PARAM_XXX` set was designed as a superset/replacement of the (then) existing `PARAM_` messages. Hence, the same position for the payload.