mavlink/mavlink #1422: Source installation does not generate mavlink-config.cmake

didrod, 2020-07-08:
Hi. I'm trying to prototype a new MAVLink dialect, and I want to make MAVROS to use my dialect.

MAVROS finds MAVLink by invoking [`find_package(mavlink)`](https://github.com/mavlink/mavros/blob/277164425e002adae33d89854e4f9209757da78c/libmavconn/cmake/Modules/MavrosMavlink.cmake#L3). This works if I installed mavlink library using a package manager since the package includes `mavlink-config.cmake`, however the source installation does not seem to have it.

## Steps to reproduce
``` bash
$ git clone --recursive https://github.com/mavlink/mavlink && cd mavlink
$ cmake -H. -B./build -DCMAKE_BUILD_TYPE=Release -DCMAKE_INSTALL_PREFIX=./install
$ cmake --build ./build -- -j12
$ cmake --build ./build --target install
$ find ./install -name '*.cmake' -print
```

## Expected result
`**/mavlink-config.cmake` should come out after invoking last command.

hamishwillee, 2020-07-09:
I have never tried this, so easy question first - did you follow https://mavlink.io/en/getting_started/ros.html >

@TSC21 FYI.

didrod, 2020-07-09:
Thanks. I didn't noticed the documentation was there.

closed 2020-07-09 by didrod

hamishwillee, 2020-07-09:
Easy to miss! Should probably be somewhere in ROS docs.