mavlink/mavlink #1440: How do I increase MAVLINK heartbeat timeoout

TomAndJerry342, 2020-08-01:
I have a Pixhawk 4 running PX4 and I have an LTE based telemetry link, the transmission time is too long for MAVLINK's heartbeat so it always 'loses' communication then regains it. This problem was fixed by turning on high latency mode but then it took way to long when I actually lost communication for Qgroundcontrol to timeout. Is there any way I can increase the maximum number of messages lost before it times out?

hamishwillee, 2020-08-03:
This is a PX4 question - so you should ask on the PX4 discussion boards.
That said, it will probably be somewhere in https://github.com/PX4/Firmware/blob/master/src/modules/mavlink/mavlink_main.cpp#L2148

hamishwillee, 2020-08-03:
Actually, I see I'm pointing at the wrong place - you're after PX4 timing out after the QGC late HEARTBEAT arrival. Same point though - this area is for general queries, and this is a query about use of code in a particular stack.

closed 2020-08-03 by hamishwillee