mavlink/mavlink #1445: ROI status message

dayjaby, 2020-08-07:
At the moment there exists no MAVLink message providing status information of the gimbal(s), especially whether or not they are locked to a ROI and where the ROI is located.

On PX4 side I'd like to have a way to process vehicle_roi messages and stream the status information of the ROI, mostly for unit testing whether DO_SET_ROI_* commands worked or not.

hamishwillee, 2020-08-10:
Actually there are, I believe: https://mavlink.io/en/services/gimbal_v2.html

@julianoes is the PX4 expert, and the design of these messages is ongoing.

dayjaby, 2020-08-10:
Actually there is no such message. As far as I understand, the gimbal device itself is agnostic about GPS coordinates and the gimbal manager handles the ROIs. Thus the gimbal manager should provide information about the current ROI in a status.

dayjaby, 2020-08-10:
Oh, Julian implemented this message already: https://github.com/mavlink/mavlink/pull/1418/files

Closing this and sorry for the confusion :)

closed 2020-08-10 by dayjaby