mavlink/mavlink #1448: VIDEO_STREAM_INFORMATION marked as WiP.

ben-crowhurst, 2020-08-09:
Hey, I was looking to implement VIDEO_STREAM_INFORMATION as a way of getting available video feeds.  I can see QGroundControl contains an implementation [here](https://github.com/mavlink/qgroundcontrol/blob/2e033034dc3944d329bb9fc3a52018d34d527d8e/src/Camera/QGCCameraControl.h#L26).  However, it is marked as WiP and has been for quite some time. 

Will this message ever get ratified and make it into the mainstream protocol?

hamishwillee, 2020-08-10:
@dogmaphobic [VIDEO_STREAM_INFORMATION](https://mavlink.io/en/messages/common.html#VIDEO_STREAM_INFORMATION) is now in production systems, so any reason not to remove the WIP?

Ditto for the others? I can do the work if you confirm this is food to go.

hamishwillee, 2020-10-14:
@ben-crowhurst Sorry for delay. There was no response from gus, but I've created PR to remove the WIP now: https://github.com/mavlink/mavlink/pull/1513

You can treat this as "good to go".

PR #1513 (2020-11-26): common: Remove WIP on video streaming
Fixes #1448 

This removes the `wip` tagging on `VIDEO_STREAM_STATUS`, [VIDEO_STREAM_INFORMATION](https://mavlink.io/en/messages/common.html#VIDEO_STREAM_INFORMATION), `MAV_CMD_VIDEO_START_STREAMING`,
`MAV_CMD_VIDEO_STOP_STREAMING` that was added with the PR in https://github.com/mavlink/mavlink/pull/1060

Generally WIP has been mis-used. Previously we've added it to common.xml to indicate "we're about to implement this". What should happen is that the message with wip gets added when there is a complete but untested implementation - essentially as a shake [... cut]
closed 2020-10-14 by hamishwillee