mavlink/mavlink #1459: Initiating Terrain upload from GCS

mnumanuyar, 2020-08-25:
As I understand currently terrain data is uploaded to vehicle from gcs upon mission upload and later as vehicle asks.  If  the vehicle failed to  receive needed information it interpolates from avaliable data.  Alternatively user can manually provide the data to vehicle as mentioned [here](https://ardupilot.org/plane/docs/common-terrain-following.html)

> If you are not flying with a GCS that can fetch and supply terrain data as the vehicles flies between waypoints, the you may need to load terrain data covering the flight route and/or area for RTL paths manually.

However reaching the SD card is not always easy. I would like a way to initiate terrain upload process from gcs. 

My suggestion is adding a meaning to TERRAIN_REQUEST message when it is send from gcs to vehicle.  Currently it is only  sent from vehicle to GCS  for requesting terrain information. If the definition of the message is changed so that, in addition to current behavior, when it is send from gcs, vehicle will check the available terrain info and start requesting and keeping track of the missing info.

On a side note, I just realised there is no target sysid/comp id parameters in related message definitions.  Does the current system works with mutiple vehicles ?

hamishwillee, 2020-08-26:
@DonLakeFlyer is the information only requested by vehicle on mission upload or is this done at the point of flight? Also is it done outside of missions?

What I'm getting at is that if it is only ever done in missions and when you're uploading a mission, then you're connected to the GCS and the idea is not needed. However if it can be needed when not connected to GCS then there is kind of a need for "offline terrain data" and it makes a lot of sense to push this from the GCS.

@auturgy Can you find out if ArduPilot supports this somehow "under the hood"? 

> On a side note, I just realised there is no target sysid/comp id parameters in related message definitions. how does the current system works with mutiple vehicles ?

Any system that gets the TERRAIN_REQUEST will send TERRAIN_DATA. So I guess that means if there are multiple connected GCS they will send the terrain data, and that every system will get the terrain data.

I think it will make little or no difference to the actual processing of information, but it will mean that a lot of un-necessary bandwidth is used up in the case where multiple GCS might respond. We would need addresses if we allow the GCS to start making push requests.

DonLakeFlyer, 2020-08-26:
> is the information only requested by vehicle on mission upload or is this done at the point of flight? Also is it done outside of missions?

It's done whenever it wants based on need. Whether that is missions only or not depends on firmware support. For example I think ArduPilot has terrain based RTL which I think can follow terrain without needing a mission. The vehicle can request any additional terrain data it needs at any time including mid-flight.

DonLakeFlyer, 2020-08-26:
> Any system that gets the TERRAIN_REQUEST will send TERRAIN_DATA.

That is true for any protocol which is pull from the vehicle side.

> We would need addresses if we allow the GCS to start making push requests.

I don't see how this has any relation to supporting push from the GCS->Vehicle direction. A push would be user initiated on a single gcs.

stephendade, 2020-08-27:
ArduPilot has a terrain website where users can download terrain tiles to put on their SD card: https://terrain.ardupilot.org/

Documentation at https://ardupilot.org/plane/docs/common-terrain-following.html#sources-of-terrain-data

>However reaching the SD card is not always easy. I would like a way to initiate terrain upload process from gcs.

Maybe MAVFTP on the GCS could be used to upload the tiles from the terrain generator to the SD card?