mavlink/mavlink #1464: MAVLink docs: incomplete for common

olliw42, 2020-08-31:
minimal.xml is now included in common.xml, but the respective messages are NOT also listed on the documents page
e.g, an important message like HEARTBEAT is now not listed anymore :)

hamishwillee, 2020-08-31:
https://mavlink.io/en/messages/minimal.html#HEARTBEAT

Common includes minimal.xml and this is listed. Are you suggesting that the dialect files should be auto-expanded out? I'm somewhat tempted by the idea.

olliw42, 2020-08-31:
> Are you suggesting that the dialect files should be auto-expanded out?

yes

I use the docs very often, and find it very unfortunate and less practical as it is right now.

hamishwillee, 2020-09-02:
Fair enough. I'll have a look at the docs generator in the next week or so. I think I can do a relatively easy hack.

olliw42, 2020-09-02:
great!

I realize I might have been a bit unclear. I guess I'm not actually suggesting to generally expand out the included xmls, i.e., also for dialects (dialects in the sense of the one google doc ). I suggest so only for standard and common (and other similar, if there ever are other similar). For dialects I guess I would handle it as you currently do it, i.e., just show the additions by that dialect.

hamishwillee, 2020-09-03:
@olliw42 Thanks. I misinterpreted you, but it doesn't matter because the dev call came to the same conclusion :-). I created a fix but waiting on review of the "approach".

olliw42, 2020-09-22:
this seems to have addressed to some reasonable extend, so I close.

closed 2020-09-22 by olliw42