mavlink/mavlink #1467: COMPONENT_INFORMATION remaining issues to work through

DonLakeFlyer, 2020-09-01:
This issue is meant to finish up discussion on the `COMPONENT_INFORMATION` message and resolve remaining issues. It only covers the fields of that specific message. It is not meant for discussion of specific metadata types associated with this message.

Here are the unresolved issues:
* What is the scope of `metadata_uid` as it relates to a gcs caching/reusing metadata?
  * A gcs needs to be able to cache metadata/json files to prevent having to download them every time.
  * What are the rules for when a `uid` match indicates that the cached file can be reused?
  * Can the `uid` generation be spec'ed such that the `uniqueness` is guaranteed to work across different firmware implementations?
  * Or is the `uid` only unique to files which come from the same firmware type? Assuming same firmware generates unique uids in a common way for all of it's variants?
  * Or does it need to be at a lower level of scoping than that?
* Should be the `uri` fields be reworked to allow room for longer uris?
  * They currently stand at 70 chars which isn't a lot of room to specify a unique uri.
  * Another possibility is to only specify the metadata uri. And then include some smaller bit of information which indicates translation metadata is available and in what format. And then the uri for that could be built by tacking on `.l10n` or something to the metadata uri. That could lead to more space for a larger metadata uri.
  * If no change to the structure then at a minimum they should be extended to take up as much room as possible to fill a single mavlink packet.
* File size is critical from a standpoint of storing these files on the board itself. 
  * Gzip only provides an ok level of compression
  * BZip would provide for smaller files after compression. Should we switch to that as the compression standard?

hamishwillee, 2020-09-02:
> File size is critical from a standpoint of storing these files on the board itself.

Or bson? I mention because Lorenz was suggesting it for sharing mission/parameter type data.

DonLakeFlyer, 2020-09-02:
> Or bson?

For metadata I think human readable is key. Also bson will only be smaller for files which have a lot of raw value numeric data.

hamishwillee, 2020-11-03:
@DonLakeFlyer Is this issue still needed? Bzip is still of interest AFAIK.

DonLakeFlyer, 2020-11-04:
All of these are still being worked on. But I think a better way to handle is I'll just post pulls with docs changes or whatever as solutions are proposed.

closed 2020-11-04 by DonLakeFlyer