mavlink/mavlink #1474: Mavlink release/kinetic/mavlink no longer compatible with Mavros 1.2.0

marcelino-pensa, 2020-09-10:
I am cloning mavlink through `https://github.com/mavlink/mavlink-gbp-release.git` in the branch `release/kinetic/mavlink`. It seems like the most recent rebase from 7hrs ago prevents mavros `1.2.0` from compiling. Is this an intentional change?

This is the error:

```
Errors     << mavros:make /root/rosws/logs/mavros/build.make.000.log
In file included from /root/rosws/src/pensa/third_party/mavros/mavros/src/lib/enum_to_string.cpp:19:0:
/root/rosws/src/pensa/third_party/mavros/mavros/include/mavros/utils.h:70:40: error: ‘std::__cxx11::string mavros::utils::to_string’ redeclared as different kind of symbol
 std::string to_string(mavlink::common::MAV_AUTOPILOT e);
                                        ^
/root/rosws/src/pensa/third_party/mavros/mavros/include/mavros/utils.h:68:13: note: previous declaration ‘std::__cxx11::string mavros::utils::to_string(mavlink::common::MAV_SENSOR_ORIENTATION)’
 std::string to_string(mavlink::common::MAV_SENSOR_ORIENTATION e);
             ^
/root/rosws/src/pensa/third_party/mavros/mavros/include/mavros/utils.h:70:23: error: ‘MAV_AUTOPILOT’ is not a member of ‘mavlink::common’
 std::string to_string(mavlink::common::MAV_AUTOPILOT e);
                       ^
/root/rosws/src/pensa/third_party/mavros/mavros/include/mavros/utils.h:70:23: note: suggested alternative:
In file included from /root/rosws/devel/include/mavlink/v2.0/common/common.hpp:2254:0,
                 from /root/rosws/src/pensa/third_party/mavros/libmavconn/include/mavconn/mavlink_dialect.h:26,
                 from /root/rosws/src/pensa/third_party/mavros/mavros_msgs/include/mavros_msgs/mavlink_convert.h:18,
                 from /root/rosws/src/pensa/third_party/mavros/mavros/include/mavros/utils.h:22,
                 from /root/rosws/src/pensa/third_party/mavros/mavros/src/lib/enum_to_string.cpp:19:
/root/rosws/devel/include/mavlink/v2.0/common/../minimal/minimal.hpp:34:12: note:   ‘mavlink::minimal::MAV_AUTOPILOT’
 enum class MAV_AUTOPILOT : uint8_t
            ^
In file included from /root/rosws/src/pensa/third_party/mavros/mavros/src/lib/enum_to_string.cpp:19:0:
/root/rosws/src/pensa/third_party/mavros/mavros/include/mavros/utils.h:71:40: error: ‘std::__cxx11::string mavros::utils::to_string’ redeclared as different kind of symbol
 std::string to_string(mavlink::common::MAV_TYPE e);
                                        ^
/root/rosws/src/pensa/third_party/mavros/mavros/include/mavros/utils.h:68:13: note: previous declaration ‘std::__cxx11::string mavros::utils::to_string(mavlink::common::MAV_SENSOR_ORIENTATION)’
 std::string to_string(mavlink::common::MAV_SENSOR_ORIENTATION e);
             ^
/root/rosws/src/pensa/third_party/mavros/mavros/include/mavros/utils.h:71:23: error: ‘MAV_TYPE’ is not a member of ‘mavlink::common’
 std::string to_string(mavlink::common::MAV_TYPE e);
                       ^
/root/rosws/src/pensa/third_party/mavros/mavros/include/mavros/utils.h:71:23: note: suggested alternative:
In file included from /root/rosws/devel/include/mavlink/v2.0/common/common.hpp:2254:0,
                 from /root/rosws/src/pensa/third_party/mavros/libmavconn/include/mavconn/mavlink_dialect.h:26,
                 from /root/rosws/src/pensa/third_party/mavros/mavros_msgs/include/mavros_msgs/mavlink_convert.h:18,
                 from /root/rosws/src/pensa/third_party/mavros/mavros/include/mavros/utils.h:22,
                 from /root/rosws/src/pensa/third_party/mavros/mavros/src/lib/enum_to_string.cpp:19:
/root/rosws/devel/include/mavlink/v2.0/common/../minimal/minimal.hpp:62:12: note:   ‘mavlink::minimal::MAV_TYPE’
 enum class MAV_TYPE : uint8_t
            ^
In file included from /root/rosws/src/pensa/third_party/mavros/mavros/src/lib/enum_to_string.cpp:19:0:
/root/rosws/src/pensa/third_party/mavros/mavros/include/mavros/utils.h:72:40: error: ‘std::__cxx11::string mavros::utils::to_string’ redeclared as different kind of symbol
 std::string to_string(mavlink::common::MAV_STATE e);
                                        ^
/root/rosws/src/pensa/third_party/mavros/mavros/include/mavros/utils.h:68:13: note: previous declaration ‘std::__cxx11::string mavros::utils::to_string(mavlink::common::MAV_SENSOR_ORIENTATION)’
 std::string to_string(mavlink::common::MAV_SENSOR_ORIENTATION e);
             ^
/root/rosws/src/pensa/third_party/mavros/mavros/include/mavros/utils.h:72:23: error: ‘MAV_STATE’ is not a member of ‘mavlink::common’
 std::string to_string(mavlink::common::MAV_STATE e);
                       ^
/root/rosws/src/pensa/third_party/mavros/mavros/include/mavros/utils.h:72:23: note: suggested alternative:
In file included from /root/rosws/devel/include/mavlink/v2.0/common/common.hpp:2254:0,
                 from /root/rosws/src/pensa/third_party/mavros/libmavconn/include/mavconn/mavlink_dialect.h:26,
                 from /root/rosws/src/pensa/third_party/mavros/mavros_msgs/include/mavros_msgs/mavlink_convert.h:18,
                 from /root/rosws/src/pensa/third_party/mavros/mavros/include/mavros/utils.h:22,
                 from /root/rosws/src/pensa/third_party/mavros/mavros/src/lib/enum_to_string.cpp:19:
/root/rosws/devel/include/mavlink/v2.0/common/../minimal/minimal.hpp:137:12: note:   ‘mavlink::minimal::MAV_STATE’
 enum class MAV_STATE : uint8_t
            ^
In file included from /root/rosws/src/pensa/third_party/mavros/mavros/src/lib/enum_to_string.cpp:19:0:
/root/rosws/src/pensa/third_party/mavros/mavros/include/mavros/utils.h:73:40: error: ‘std::__cxx11::string mavros::utils::to_string’ redeclared as different kind of symbol
 std::string to_string(mavlink::common::MAV_COMPONENT e);
                                        ^
/root/rosws/src/pensa/third_party/mavros/mavros/include/mavros/utils.h:68:13: note: previous declaration ‘std::__cxx11::string mavros::utils::to_string(mavlink::common::MAV_SENSOR_ORIENTATION)’
 std::string to_string(mavlink::common::MAV_SENSOR_ORIENTATION e);
             ^
[... 178 lines cut ...]
 enum class MAV_STATE : uint8_t
            ^
In file included from /root/rosws/src/pensa/third_party/mavros/mavros/src/lib/enum_sensor_orientation.cpp:17:0:
/root/rosws/src/pensa/third_party/mavros/mavros/include/mavros/utils.h:73:40: error: ‘std::__cxx11::string mavros::utils::to_string’ redeclared as different kind of symbol
 std::string to_string(mavlink::common::MAV_COMPONENT e);
                                        ^
/root/rosws/src/pensa/third_party/mavros/mavros/include/mavros/utils.h:68:13: note: previous declaration ‘std::__cxx11::string mavros::utils::to_string(mavlink::common::MAV_SENSOR_ORIENTATION)’
 std::string to_string(mavlink::common::MAV_SENSOR_ORIENTATION e);
             ^
/root/rosws/src/pensa/third_party/mavros/mavros/include/mavros/utils.h:73:23: error: ‘MAV_COMPONENT’ is not a member of ‘mavlink::common’
 std::string to_string(mavlink::common::MAV_COMPONENT e);
                       ^
/root/rosws/src/pensa/third_party/mavros/mavros/include/mavros/utils.h:73:23: note: suggested alternative:
In file included from /root/rosws/devel/include/mavlink/v2.0/common/common.hpp:2254:0,
                 from /root/rosws/src/pensa/third_party/mavros/libmavconn/include/mavconn/mavlink_dialect.h:26,
                 from /root/rosws/src/pensa/third_party/mavros/mavros_msgs/include/mavros_msgs/mavlink_convert.h:18,
                 from /root/rosws/src/pensa/third_party/mavros/mavros/include/mavros/utils.h:22,
                 from /root/rosws/src/pensa/third_party/mavros/mavros/src/lib/enum_sensor_orientation.cpp:17:
/root/rosws/devel/include/mavlink/v2.0/common/../minimal/minimal.hpp:156:12: note:   ‘mavlink::minimal::MAV_COMPONENT’
 enum class MAV_COMPONENT
            ^
In file included from /root/rosws/src/pensa/third_party/mavros/mavros/src/lib/enum_sensor_orientation.cpp:17:0:
/root/rosws/src/pensa/third_party/mavros/mavros/include/mavros/utils.h:114:18: error: ‘MAV_TYPE’ in namespace ‘mavlink::common’ does not name a type
 mavlink::common::MAV_TYPE mav_type_from_str(const std::string &mav_type);
```

marcelino-pensa, 2020-09-10:
I have just verified that `mavros` fails to compile in `1.3.0` and on `master`

marcelino-pensa, 2020-09-10:
Checking out to `release/kinetic/mavlink/2020.8.8-1` fixes my problem.

Before I close this issue, is `mavlink` in kinetic no longer supposed to work with `mavros`?

vooon, 2020-09-10:
From errors i see that headers didn't generate. Did you tried to clean your workspace and then build again? `cantin clean`.

marcelino-pensa, 2020-09-10:
Yeah, I did `catkin clean`. It was also failing in our CI environment, which builds from scratch every time...

vooon, 2020-09-11:
Found the problem: https://github.com/mavlink/mavlink/commit/a5e3138d15461f951b701582dd23c304b5a3bc01#diff-ceacc1e2390186650146a45be8fcb691

Because of that enums changes their namespace from `mvalink::common::` to `mavlink::minimal::`.
We have to update namespaces in mavros.

amilcarlucas, 2020-09-23:
The latest version works fine with mavros 1.4.0

amilcarlucas, 2020-09-28:
Re-test with mavros 1.4.0 and reopen this if it does not work.

closed 2020-09-28 by amilcarlucas