mavlink/mavlink #1477: question/bug: extension, should the data be aligned?

BazookaJoe1900, 2020-09-13:
I came to error of alignment on a  mavlink message that uses extension, here:
https://github.com/mavlink/c_library_v2/blob/09524c98db4eb389d3d50f7905a17805b216a9ea/common/mavlink_msg_distance_sensor.h#L16

the float is not aligned to 4 byte address.
I know that by-design and by-def, all mavlink messages fields should be aligned according to the fields type, but it seems not to be correct on extension.
is that ok? @LorenzMeier

hamishwillee, 2020-09-15:
Julian, can you comment on this?

julianoes, 2020-09-15:
This has come up in 2016, I think: https://github.com/ArduPilot/pymavlink/issues/5

BazookaJoe1900, 2020-09-15:
and what are your opinion? keep it like that? @julianoes @LorenzMeier

julianoes, 2020-09-16:
My opinion is to remove the packed attribute and try that out. I'd have to have a closer look of what that would mean though.

BazookaJoe1900, 2020-09-16:
I think that removing the packed is not good idea.
this is communication message and packed make sure that how its written on the h file, it will be packed into the message.

an option is to add 'reserved' or 'fillup' bytes between the default to the extension vars.
the problem is that its breaking compatibility.

I came to this issue because I 'accidentally' compiled the code from version 1.9 with the newest GCC. so it seems that its some kind of configuration that makes some warnings.
The more severe issue about the alignment can cause either hardfault if the MCU doesn't compatible to misaligned pointers. or it can be very non afficient to solve misaligned pointers by byte-by-byte copy,

julianoes, 2020-09-17:
>  this is communication message and packed make sure that how its written on the h file, it will be packed into the message.

I think it is copied before being written to the buffer but I have to check that again.

amilcarlucas, 2020-09-30:
Did you try regenerating with the latest pymavlink? I think some packaging stuff changed recently.

BazookaJoe1900, 2020-09-30:
@amilcarlucas . I think I did, but not sure now.
can you point me where the change has been done lately?
if there was such change it probably broke something because it changed the protocol... and changing the extension doesn't change the CRC hash

amilcarlucas, 2020-09-30:
https://github.com/ArduPilot/pymavlink/pull/412
It that what you want?

julianoes, 2020-10-05:
@BazookaJoe1900 that looks indeed like it could change it. Probably worth looking at the diff of generated headers with and without this change.

hamishwillee, 2020-10-07:
@BazookaJoe1900 Can you confirm whether the fix Amlicar points to in https://github.com/mavlink/mavlink/issues/1477#issuecomment-701408109 (went in May 3) is or is not a solution?

BazookaJoe1900, 2020-10-07:
I have tested the MAVPACKED behavior in 3 points:
1) before the change of  ArduPilot/pymavlink#412
all messages has MAVPACKED
2) with the change
only messages with extensions has MAVPACKED pragma (that what the pr wanted to do)
3) master version
same as in 2

as conclusion, messages with extensions might have non-aligned data.
some functions uses fields of mavlink as pointers, and from what I know some MCU doesn't allow non-aligned pointers, or might trigger exceptions, or will do coping byte-by-bytes. it depend on the MCU arch, flags, compilation.
I am not sure how ARM behaves on that cases, it might be less efficient.

julianoes, 2020-10-08:
@BazookaJoe1900 do you understand why it is packed for messages with extensions?

BazookaJoe1900, 2020-10-08:
Yes, it must be packet, to ensure that the extention fields will be right after the 'regulars'. Basically the regular don't need PACK because mavlink endure that they ate aligned.

julianoes, 2020-10-09:
Aha, I think I understand! Not sure what to do about it though.

BazookaJoe1900, 2020-10-09:
how I see it the original question answered, that data is not aligned.
we can't change it to be aligned, because it will break compatibility.
only from practice, no exceptions were reported, its seems that there is no critical issue here.
and I can't really tell if there are efficiency problems.

from my side, we can close the issue.
@julianoes ?

closed 2020-10-09 by julianoes