mavlink/mavlink #1481: AUTOPILOT_VERSION vendor_id, who can get, how to get?

olliw42, 2020-09-17:
The AUTOPILOT_VERSION has fields vendor_id and product_id. In https://github.com/mavlink/mavlink/pull/1469#discussion_r488288358 it was pointed out that these are the fields which allow to infer information on the hardware and should be used for that purpose.

I however could not find info on which vendor_id's are already existing, who is actually eligible to get an vendor_id assigned, and what the process would be to get a vendor_id for those who are eligible.

I'd like to get a vendor_id for the STorM32 gimbal and camera controller.

Context:

IMHO the current COMPONENT_INFORMATION is pretty useless for typical components or embedded gcses, but info on the hardware are quite useful even for typical components, and the AUTOPILOT_VERSION message is currently their best bet. 

The STorM32's  gimbal and camera mavlink components support the AUTOPILOT_VERSION message. Importantly, please note that even a non-flightcontroller component has to emit the AUTOPILOT_VERSION in order for e.g. MissionPlanner to behave properly for such components. Also note, that this message carries the capabilities field which conveys the MAV_PROTOCOL_CAPABILITY flags, some of which are also important to general components, in the case of the STorM32 e.g. the MAVFTP support flag. That is, even though the name of the message suggests that it is for an autopilot and not for other components, the MAVLink ecosystem, at least as it currently exits, essentially demands also other components to support this message. 

In view of these facts, asking for a vendor_id also for a non-autopilot component appears to be in the books.

auturgy, 2020-09-17:
These are just the USB ID's.  Some listed here - https://ardupilot.org/dev/docs/USB-IDs.html

olliw42, 2020-09-17:
ah, many thx for pointing that out.

yet, I would think, there must be kind of some process to get the vid recognized by mavlink,  I mean, it wouldn't really be an id if everyone would just use whatever they want to use ... and without such a process it would also be at odds with the implication in https://github.com/mavlink/mavlink/pull/1469#discussion_r488288358 that it allows to identify the vendor ...

so, should I understand this to say that one has to go through the (tedious/costly) process of acquiring a usb vid/pid?

hamishwillee, 2022-02-24:
@auturgy FMI do GCS or other consumers use vid/pid information for anything specific or is it just data points they might use for something?

My "guess" is that for autopilots these might be used for determining you have the correct hardware to work out compatible bootloader updates? If so I presume there should be a strategy and we should point to it so that GCS can get a list of the ids that they might need to work with (rather than trying to hunt these)?

OR to put it another way, can a random gimbal or camera allocate whatever values it likes?

FWIW [COMPONENT_INFORMATION_BASIC](https://mavlink.io/en/messages/development.html#COMPONENT_INFORMATION_BASIC) restores the information removed from the original component info message. So would also be good to note where PID/VID come from.

auturgy, 2022-02-24:
Pid/vid should be vendor specific. ArduPilot includes them in the bootloader as per vendor advice, or uses its own generic one. They are generally assigned by the USB-IF. https://www.usb.org/getting-vendor-id
I’m not sure what PX4/Auterion does.

Regards,

James

hamishwillee, 2022-02-24:
@bkueng Do you happen to know where the autopilot pid/vid are obtained from in the `AUTOPILOT_VERSION` on PX4?

olliw42, 2022-02-24:
since I originally raised the issue
FYI, I've also purchased a USB PID/VID for using it with STorM32, for this message, but I also have to frankly say that I didn't like it LOL. I understand using USB PID/VID for the purpose might be reasonable for professional stacks, but it's not so apt for non-professional projects.

> My "guess" is that for autopilots these might be used for determining you have the correct hardware to work out compatible bootloader updates? 

while this might be used that way, I think it can't or shouldn't in general. Just imagine you create an updated bootloader of  different version, or other small changes which however would have to be reflected. Should one then have to get a new USB PID/VID each time? I think there should be really an easier to control and handle, more flexible, number

since usage is however as it is, I think one way could be to not require it really to be a USB PID/VID, or more precisely to establish uniqueness through the fact of using USB PID/VID, but to have a mavlink controlled and maintained list of numbers, which since handled by mavlink would be ensured to be unique. These numbers could then be whatever kind, also the current USB PID/VID, but they would not have to be USB PID/VID, all they would be numbers assigned by mavlink.

As a byproduct, that way you would naturally also get a list of numbers, for GCSes or whatever.

just my 2 cents

auturgy, 2022-02-24:
What you’ve described is very similar to how board id is done for fw flashing etc: https://github.com/ArduPilot/ardupilot/blob/master/Tools/AP_Bootloader/board_types.txt and the px4 equivalent, which get periodically sync’d. Not part of mavlink though.
The pid/vid utility is questionable as you say. I’m not sure why we’re reporting it, apart from the fact it was included as a field in this message.

olliw42, 2022-02-24:
yes, I recall that mechanism, and yes, it is very similar if not identical in concept
the difference, and maybe a crucial difference, would be that it would be centrally managed by mavlink. E.g. it  would then not be a magic wizardry to work out what numbers are in use ... be it to help GCSes or be it to allow other systems to find a unique value. It also might help further reducing flight stack/component specific coding