mavlink/mavlink #1507: MAV_FRAME_GLOBAL_RELATIVE_ALT frame in MISSION_ITEM_INT message

ValentineStone, 2020-10-08:
### Suggestion: "warn" against using `MAV_FRAME_GLOBAL_RELATIVE_ALT` and other non-int frames with `MISSION_ITEM_INT` messages.

`common.xml` specifies that `MISSION_ITEM_INT`'s `x` and `y` params are to be `int32`:

[image]

But `frame` is not restricted from being `MAV_FRAME_GLOBAL_RELATIVE_ALT` in `MISSION_ITEM_INT`:

[image]

It just says 'The coordinate system of the waypoint', and there is nothing in the `MAV_FRAME` enum to specify clarify that `MAV_FRAME_GLOBAL_RELATIVE_ALT` can not be used with `MISSION_ITEM_INT`.
While it should be obvious to any sane person that a floats should not be sent over wire as ints (🤦) I think it would be a good idea to "restrict" `MISSION_ITEM_INT`'s `frame`, kind of like it's done in the `SET_POSITION_TARGET_GLOBAL_INT` for example:

[image]

This way people like me, who just saw `frame` `MAV_FRAME_GLOBAL_RELATIVE_ALT` with floating point `x` and `y` in QGroundControl's `.plan` files and through that it would be just flippin dandy to send them just like that inside a `MISSION_ITEM_INT ` could avoid the ensuing confusion))

hamishwillee, 2020-10-14:
@ValentineStone Thanks for the post. 

You can use `MAV_FRAME_GLOBAL_RELATIVE_ALT` with `MISSION_ITEM_INT` - although the autopilot receiving the message must supports the frame (I believe most do) which it does. Essentially the lats and longs get multiplied by 1E7 to convert them to integers as it says for x and y in your first image above. For reals you multiply by 1e4.

The SET_POSITION_TARGET_GLOBAL_INT example you give only restricts the frame because the message is "global" - ie you can't set local frames.

Am I missing something?

ValentineStone, 2020-10-14:
@hamishwillee Thanks, I see your point about `SET_POSITION_TARGET_GLOBAL_INT`.

As far as `MAV_FRAME_GLOBAL_RELATIVE_ALT` goes, i fail to see how you can use it with `MISSION_ITEM_INT`, since inside `MISSION_ITEM_INT` message `x` and `y` are both `int32_t`, and if I were to use `float` values, like say 69.69696969, -69.69696969 they would get truncated by the message packer to be 69, -69, which is obviously a different location. This makes `MAV_FRAME_GLOBAL_RELATIVE_ALT` impossible to use with `MISSION_ITEM_INT` does it not?

ValentineStone, 2020-10-15:
@hamishwillee I understand what you are saying, but wouldn't that just be `MAV_FRAME_GLOBAL_RELATIVE_ALT_INT` mav_frame instead, now that we scale to 1e7-1e4?

Also wouldn't this slight of hand, with converting back and forth during packing and parsing be just a bad practice in general?
Because 1 - now we must ensure that the packer and parser both respect this slight of hand, which seems to me unreasonable and error prone. And 2 - that feels like and unpredictable behavior in general, no? As it stands now, at least packer/parsers I've tested - pymavgen's javascript and typescript just use the standard `int` to `float` cast, which is a truncation of decimal part. That's the behavior I'd expect of it, whereas if i were to send a `MAV_FRAME_GLOBAL_RELATIVE_ALT` with a float, and got a 1e7-1e4 scaled values with the same `MAV_FRAME_GLOBAL_RELATIVE_ALT` from my packer, now that would have been unpredictable, because this new data is different from what i sent in an unconventional way. When I saw my values truncated I was like "Ok, i see what happened here - i probably messed up the types for this message", where as if i were to see the same mav_frame but different data I would probably be like "WTF just happened" and run off to dig around in the packer/parser source code.

To avoid confusion, I am not talking about the files, they can be stored as anything, and as long as it is not a binary file with an array of `MISSION_ITEM_INT` (which would bring us back to the packing/parsing debacle above) there is no problem. I solved my specific use case by testing for `float` typed mav_frames and using their `_INT` or whatever counterparts during the file -> mavlink mission in memory projection. I do know that using ints is much more accurate than floats for this))

My original request for a 'description' or 'warning' against the use of `float` mav_frames inside `int` typed `MISSION_ITEM_INT` only relates to direct to/from binary serialization by packer/parsers, not to the storage of missions in any file, as there is no standard for that in mavlink as of now.

hamishwillee, 2020-10-16:
@ValentineStone Correct, I am saying that in this context MAV_FRAME_GLOBAL_RELATIVE_ALT_INT  and MAV_FRAME_GLOBAL_RELATIVE_ALT would be the same thing.

Originally there was a message [MISSION_ITEM](https://mavlink.io/en/messages/common.html#MISSION_ITEM) that was used with MAV_FRAME_GLOBAL_RELATIVE_ALT so the float could be copied/sent direct. I suspect that MAV_FRAME_GLOBAL_RELATIVE_ALT_INT  should be used with the MISSION_ITEM_INT and the other frame should indeed be rejected. But since most systems understand the convention for the message, they do the scaling the same way irrespective of whether the frame is the int or non-int variant. 

We moved away from the float variant because the definition of a float was resulting in a significant loss of precision. i.e. it didn't work. 

So yes it is confusing, but the converting is pragmatic. It isn't going to change. What we need to do though is make sure that we have clear direction for users.

@julianoes Anything I have said here is wrong? And if it is correct, any thoughts on how we can reduce the problem for other users like @ValentineStone joining?

hamishwillee, 2020-10-16:
PS @julianoes It might simply make sense to do as @ValentineStone suggests and have a note that the frame MAV_FRAME_GLOBAL_RELATIVE_ALT_INT  should be used for int variant messages. The other frame might or might not work, but if they choose to use it then they take the risk of miss-using it.

ValentineStone, 2020-10-20:
@hamishwillee Thanks for the clarifications. I went the route of just not using float mav_frames in my codebase, but I see how for users who have to support old systems and gcs' this converting trick, although not standardized, can be a reliable way of dealing with the deprecated floats.

If this converting situation is prevalent in mavlink implementations I think it would be a good idea to add it to the docs, so that there would be no undefined behavior when using float typed `x` and `y` mav_frames in `_INT` messages

This seems like a good place for it: https://mavlink.io/en/services/mission.html#command_message_type
Like you have
> If MISSION_REQUEST is recieved the system should instead respond with MISSION_ITEM_INT items (as though it received MISSION_REQUEST_INT).

perhaps you can append something about float mav_frames and how they are to be interpreted and converted back and forth on send and receive.

hamishwillee, 2020-10-21:
Good idea. Please check https://github.com/mavlink/mavlink-devguide/pull/286

closed 2020-10-22 by hamishwillee