mavlink/mavlink #1523: Proposal for extending COMMAND_LONG to allow for arbitrary param types

olliw42, 2020-10-29:
One of the super annoying features of COMMAND_LONG is that all 7 parameters are float fields, but these fields are often missused to carry integers as demanded by need. That's why there is COMMAND_LONG_INT, in order to overcome the limitations of float-ening integers for at least some situations, which however is also just a half-cooked solution. 

What one really would want to have is a means to make every parameter of "arbitrary" type, specifically of type float, uint32_t or int32_t (smaller integers are then trivial). This here is a proposal of how that could be achieved.

In C the difference is if or not to type cast. Now it has to be like

`uint32_t val = payload.param1;`

while what one would want it to be is 

`uint32_t val = (uint32_t)payload.param1;`  (or maybe a memcpy)

So, what one would need is something like

`MAV_CMD_XXX_PARAM1_TYPE val = (MAV_CMD_XXX_PARAM1_TYPE)payload.param1;`

And similarly for the other way around for setting a param. And this is then actually already the proposal.

The proposal in detail:

* The type of a parameter could be specified in the xml by an attribute type="xxx" very much like it is done for fields in messages. E.g.

```
    <enum name="MAV_CMD">
      <entry value="12345" name="BLABLA" hasLocation="false" isDestination="false">
        <param index="1" label="huhu" type="float" units="oh" minValue="-180" maxValue="180">huhu</param>
        <param index="2" label="aha" type="uint32_t" units="ho" minValue="-180" maxValue="180">aha</param>
      </entry>
    </enum>
```

* The C code generator would then create respective defines. For the example e.g.

```
#define MAV_CMD_BLABLA_PARAM1_TYPE float
#define MAV_CMD_BLABLA_PARAM2_TYPE uint32_t
```

* For backwards compatibility for existing commands a type of float would be assumed per default if no type="" attribute is there. Since any code handling MAV_CMDs is specific for a specific cmd, old codes handling old cmds do not have to be changed, i.e., no backwards issue here. 

* The possible type attributes would be limited as compared to for messages, e.g. arrays or 64bit types would not be admitted.

I must admit that I'm focused here on C and don't really comprehend if and how that would extend to other languages (or architectures). I however think that the respective code generator always should be able to provide param setters and getters which do the job. To handle different architectures in C it might be e.g. that one rather would need macros like
```
#define MAV_CMD_BLABLA_PARAM1_TYPE float
#define MAV_CMD_BLABLA_PARAM1(x) (float)(x)
MAV_CMD_XXX_PARAM1_TYPE val = MAV_CMD_XXX_PARAM1(payload.param1);
```
in order to allow to handle things like different endianess. But as said, I think this is possible by the code generator, i.e., more a technical detail than a conceptual issue.

Cheers,
Olli

hamishwillee, 2020-11-04:
Hmmmm. Other suggestion that sometimes get made is to deprecate/replace the existing `COMMAND_` with `COMMAND_DOUBLE` that works with both types; albeit at the cost of larger message.

I think you mean `COMMAND_INT` not `COMMAND_LONG_INT`, which exists to send the large-ish floats used for latitude/longitude as scaled integers to avoid precision errors you get when they are sent unscaled in a float.

So in summary I think the idea is to add a data type to the XML that says what the type is. The type is then used to transparently and safely cast an integer (say) so that it can be sent in the float. If the rules can be implemented in other languages then sounds like a good approach - would give us option to do things like deprecate COMMAND_INT.  If nothing else, adding the expected type of the data in the xml could be used to prevent a command being used in the "wrong" type of message; or give warnings if a float value was sent in COMMAND_INT etc.

@amilcarlucas @LorenzMeier @peterbarker @julianoes Thoughts?

olliw42, 2020-11-04:
sorry, COMMAND_INT of course

double I think would not be wise since for widely-around platforms it would be quite a performance penalty

olliw42, 2020-11-09:
this proposal doesn't seem to inspire much :)

so let me give a specific example, which actually triggered thinking about it. The gimbal manager messages Julian introduced recently have a gimbal manager flag of size uint32_t. It must be larger than unit16_t since in that messages the lower 16 bits mirror the gimbal device flags; the gimbal manager parts are in the higher 16 bits. However, the gimbal can be controlled also via some CMDs, which requires that the gimbal manager flag is part of the CMD. Obviously, since a float can't hold a complete 32-bit integer the available flags for the gimbal manager part is in fact limited to just a 8 (or so) bits !! (I think it wasn't yet carefully investigated how many flags can be hold by the float, but I would guess it's 24, or 23). In short, the concept somewhat breaks here with the current CMD_LONG. What one would want is something along the proposal. So it's not just academic but this example is real.

I'd also like to emphasise again the analogy with the parameters, where the float param_value fields are handled in exactly the way which is proposed here for the CMD_LONG fields. In fact, code-wise one could do exactly (or similarly) what is proposed for them, namey to introduce a uinion for that purpose, see https://mavlink.io/en/services/parameter.html#mavgen-c-api. Might be better than what I proposed in the above. Anyway, the bottom line, any implementation which can handle params in MAVlink compatible ways should also be able to handle CMDs as proposed here.

:)

hamishwillee, 2020-11-11:
@olliw42 I think this is interesting. My thought was to raise it in the dev call. Unfortunately not all key people will be there, so may not progress much. Upshot, not ignoring!

julianoes, 2020-11-11:
Very interesting proposal @olliw42. I need to think more about it. In a perfect world I want type safety when using the APIs, so no casts that I have to implement manually and could get wrong.

olliw42, 2020-11-11:
I recite the example of the parameters :).
I don't disagree at all with your argument (one can't it's too obviously a good one). I though argue that it should be similar for both.

hamishwillee, 2020-11-11:
@olliw42 FYI summary of the dev call discussion was that we need more thought. Essentially we need to convince ourselves that:
- It is possible to implement this solution across all languages.
- The solution can be made transparent to users; i.e. we don't replace the complexity of having to know about scaling of floats with having to know about casting.
- That a system that does not know about what we're doing here does not get broken.
- That it is future proof-ish. I.e. we've at least thought about bigger data types etc.

My concern is the third case. To do this I think we might need another message type for this command and slowly move to deprecate the other types.

olliw42, 2020-11-12:
many thx for the feedback and the deep thoughts. Your points are of course all highly important.

I think however that the first 3 had been pretty much answered.

It seems to me that the factual analogy to the parameters isn't yet fully transmitting. That is, what is of concern here should be also of concern there, and vice versa. Ergo, what is not of concern there, shouldn't be of concern here.

1. It seems to work for params across all languages.

2. User-percieved complexity is a pretty elastic notion ... and there thus might not be two equal opinions. For params it appears to have been concluded to be worth it.
 
3. It highly depends on what you exactly mean by this. If you mean a system which is using a mavlink code which predates the change, then it won't get broken, as I've tried to explain in the above. A system which would use an "old" mavlink code would never digest a command which uses this new scheme.

4. I don't have an answer to that. It again very much depends on what exactly is meant by future proof and there might be not two equal opinions. By construction CMD_LONG is not future proof in this sense as it never would be able to hold larger data types without going wild. My argument goes the other way around. You don't lose anything by extending CMD_LONG but gain. A free lunch so to say. Introducing an all new message of course would be a viable alternative. I would however not like that it would be quite big and thus not suggest it as the new do it all general purpose message (which CMD_LONG kind of is, right). I mean, at the end of the day we could do a big CMD which can hold all and each message and use it instead of all messages with respective CMDs ... this would ultimately solve the routing problems at last ... but I don't think that large fixed-size universal container packets should be the new norm.

I guess the main concern I would have is that essentially all language generators would have to be adapted at once. At minimum they would have to be adapted such that it doesn't expose a new command using the extension if it doesn't handle it yet.

:)

olliw42, 2021-01-11:
I think it is still a reasonable suggestion and would like to keep it live. :)

hamishwillee, 2021-01-14:
> I guess the main concern I would have is that essentially all language generators would have to be adapted at once. At minimum they would have to be adapted such that it doesn't expose a new command using the extension if it doesn't handle it yet.

Yes. And who would do the work. 

@julianoes I believe we left this with you evaluating. @peterbarker do you have an opinion/advice?

olliw42, 2021-01-14:
> And who would do the work.

I understand the concern. 

First off, I think the key question to answer first is if or how valuable the proposed extension would be. IMHO, the question of workload, while important, should be only second, and the acceptance level should scale with the answer to the first question. I.e., "not valuable" should translate into "no effort acceptable" and "very valuable" should translate into "more effort acceptable" :).

I also believe that the situation is not nearly as bad as it may seem at first

1. It would affect only new commands but not old commands. That is, as long as an implementation isn't using such a new command it's irrelevant.
 
2. All languages provide a means to handle the command fields by hand, i.e., do some sort of retyping (typecasting). That is, whenever anyone wants to uses such a command it can be used. Don't forget that the handling of a command needs to be implemented anyways. All the unit stuff e.g. has to be done by hands, there is no automatism here.
 
If you think about it, all what a generator does is to provide some primitives to make it convenient, but it's not like that a generator would add some magic sauce to make it possible in the first place. For instance, in the python version one needs to go through such a manual convert-it-to-something-proper hassle anyways for strings/char arrays ... even worse maybe for parameter values, which are handled as float only too, but we all know they have a type... From the fact that things like that haven't yet been changed one only can conclude that the pressure to change is very low. I don't see why that should be different for command_long attributes.

4. Many of the generators are quite "old" anyways, and e.g. don't have newer xml attribute stuff added. That is, even though the list of generators may look long, we are actually talking just about few. 

5. If we don't do a stupid mistake, one essentially only needs to copy in the generators from how param is handled (unless one wants to make it better). That is, it's not like that the generator needs to be rewritten.

6. For C I would volunteer, which reduces the list count by one.

julianoes, 2021-02-03:
I had another read through this thread after it was brought up in today's MAVLink dev call.

Let me try to answer to above discussion points:

1. (and 3.) Backwards compatibility:
You're right that it would be backwards compatible as such if it is only used for new commands (as you suggest). This means however that language bindings which do not support this proposal can't support any new commands that are introduced and make use of the feature.

2. (and 5.) "Retyping"/library support:
I don't like the "retyping" part. It's not as straightforward as you suggest, casting alone does not work.
In C something like this is required: `int32_t val = *(int32_t*)(&payload.param1);` This could be done with a macro.
For Python you would need something like `struct.pack(">i", struct.pack(">f", payload.param1)[0])` but I haven't tested that.

For parameters there is in fact no support from the generated libraries to help with that, so it is implemented in user code and in my opinion quite ugly, something I would try to avoid rather than proliferate.
Also see:
https://github.com/PX4/PX4-Autopilot/blob/b2f0c149ecb5614b0bc878d0b5359df659e98051/src/modules/mavlink/mavlink_parameters.cpp#L439-L447
https://github.com/ArduPilot/pymavlink/blob/0e88456d42aff2323f307bafe44ad05680127bda/mavparm.py#L46-L47
https://github.com/ArduPilot/ardupilot/blob/de9382b0f4b481b4f597a14adfb7f4f59afeb717/libraries/AP_Param/AP_Param.cpp#L1764-L1779
https://github.com/mavlink/MAVSDK/blob/e869ba85909d39d69ad4b326e41c9378badd1417/src/core/mavlink_parameters.h#L254-L262

4./6. Generators:
I don't have a good grasp of what's used and/or maintained. I only ever use the C API myself.
As long as I don't have to do the work, I probably won't care too much :).

7. Impact:
My fear is that because this feature is only available for "new" commands it will not have much impact on the existing problems or lead to replacements of many existing commands in order to make use of this proposal. This would mean the proposal was worthwhile but also could make the spec quite messy because you end up with more _V2 variants and deprecated commands which generates other problems like limited adoption and a general mess.

My vote would probably be to come up with a better "command system" / RPC protocol for MAVLink v3 instead and continue with the status quo for v2.

olliw42, 2021-02-03:
excellent thoughts, it's hard to not agree with them, so I agree :)

I can't access that, since "v3" is lingering around since a while, but if it were so that that you guys seriously thinking about v3, then I would fully agree that it makes much sense to put all efforts in this than tidying around with v2. I would think it's actually high time for v3, with hopefully all lessons learned. :)

I guess should be closed.

THX for considering it !!!

closed 2021-02-03 by olliw42

hamishwillee, 2021-02-04:
@olliw42 Had a chat about this in the dev call. General agreement that this is a good idea, and that "in the end" this would be much better. However we see a messy state in the middle that is more confusing for adopters. On top of that the work should be co-ordinated for the generators - we can't just say "some are not important".

Upshot, this should be part of a larger set of breaking work - the notional "MAVLink 3". I am leaving this open until (at least) it is included in that scope (I'm looking for the right doc).

hamishwillee, 2021-02-04:
@olliw42 Arggg I posted this on a git issue that hadn't updated with Julian's comments ...

Julian's post is much better. The summary being "we think this would be disruptive to do as proposed" but we like it, see the need for a fix for this problem, and something should be done in MAVLink 3 timeframes. 

How serious are we about MAVLink3? I would say that we won't do MAVLink 3 unless we come up with a compelling set of reasons for a break. We will also work very hard to do things that avoid the break. I don't see that there is a set of such things that will convince the PX4 and ArduPilot communities to take the plunge yet, but I think this is one of the things that might lead us to that point.

olliw42, 2021-02-04:
well, all message & microservice things, like this one here, have technically nothing to do and are independent with a MAVlink v3, which would be on the header&footer bytes level. They only would be related through a "moment of change" 

the only reason for v3 I see is
1. parsing and routing for all messages without knowing the message (this to me would be "must be" compelling reason)
2. maybe different security concept, along the lines of the one proposal of the person who's name I forgot

However, for 1. there is a proposal for a "smooth" transition, which would avoid a v3.

there could be however quite a number of renovated messages & commands
1. identification, discovery
2. parameter handling
3. commands handshake
4.  disentangling commands for missions and as command
5. whatever I forgot now

However, these all are "just" xml things, and can/could be done any way and any time :)

=> looking at that I do not see any need for a v3.

> We will also work very hard to do things that avoid the break. 

=> if one concludes that something needs to be done one also needs to accept that something needs to be done

... and that it won't come without any pain ...

it's a bit of a circling event, one tries to avoid v3 because of the break but also to change xml's because of the break ... it's not obvious to me which path out you have in mind :)

hamishwillee, 2021-02-11:
At the moment there is no will to break anything, so I see no real way forward. But if there is a break, I hope to fix a lot of these things.

olliw42, 2021-02-11:
> At the moment there is no will to break anything

yeah, one can literally feel that :)

(and it's much understandable)(but comes with the cost to slow down general progress)

> But if there is a break, I hope to fix a lot of these things.

I just hope that in such a break all aspects are reminded and taken up, not like with Mavlink2. If at the time of a break this issue is recalled, it would have already served it's purpose :)

Many thx for bothering and considering anyways. !!