mavlink/mavlink #1535: Just FYI: STorM32's gimbal protocol

olliw42, 2020-12-05:
this is not an issue, just for your information.

there had been work on a new gimbal protocol v2 for quite some time, and through intense discussions fantastic progress was made, having resulted in the protocol as merged recently, and as it is also found in the mavlink docs now.

I am thankful for having had the chance to contribute in this discussion, and in fact it was a great pleasure and it also helped me tremendously to clear up my mind. Thanks Julian! and also Hamishwille! At some point however, when it came to the details such as flags and so on, different opinions emerged and I diverged. In honor of this great discussion I however feel like that I should share what I finally arrived at, not to reopen any discussion, but to share, in the best meaning of sharing ideas. The STorM32's gimbal protocol you can find defined in the [storm32.xml](https://github.com/olliw42/storm32bgc/blob/master/mavlink/storm32.xml), which also has links to more documentation. I use it 'in production' (if there is anything like this with the STorM32 LOL).

Thanks for all, cheers, Olli.

hamishwillee, 2020-12-08:
Thanks @olliw42 . Given that you and @julianoes put SO much work into this it is disappointing that we could not converge on an agreed standard! Or at least make Storm a dialect that includes the gimbal protocol. Thank you both for your hard work.

olliw42, 2020-12-09:
I really wouldn't call it disappointing. I think it reflects that the forces which you guys are constrained by and the systems&setups relevant to you are just so much very different to mine, which makes it in some way rather natural. I'm btw using two gimbal device messages of the standard.

:)

olliw42, 2021-02-04:
is now in external/dialect, so sufficiently FYI
closing

closed 2021-02-04 by olliw42