mavlink/mavlink #1550: Additional flight record messages

muramura, 2020-12-27:
I want to add a MAVLINK message in my flight log and notify GCS.
I believe that I can notify the items marked with * from the flight record section of the Japan Civil Aviation Bureau.
I will PR if this item is OK.

* Flight date
Pilot's Name
Flight Summary
* Unmanned Aircraft Registration Number
* Takeoff location
* Takeoff time
* Landing location
* Landing Time
* Flight time
* Total flight time
Items that affected the safety of the flight

[image]

auturgy, 2020-12-27:
I think we should consider how much of this needs to be exchanged between the autopilot and gcs, how much is already known from other messages, and how much can/should be directly input to the gcs without being exposed to mavlink. I understand the intent, and agree we should make it as easy as possible for compliance reporting, but need to get the design/details right.

hamishwillee, 2020-12-27:
@auturgy I read the intent to generate a log on GCS from what is coming out of the vehicle, in which case I totally agree - @muramura perhaps you might provide a reduced list of the information you think cannot already be inferred from other messages and which should be on the vehicle. 

It is possible that the intent is actually to inject some of the above information into an on-vehicle log, in which case we might need to think about this a bit more.

hamishwillee, 2021-06-23:
Closing. Please reopen with info requested in https://github.com/mavlink/mavlink/issues/1550#issuecomment-751526325 if you want to progress this.

closed 2021-06-23 by hamishwillee