mavlink/mavlink #1556: Documentation

amilcarlucas, 2021-01-06:
The documentation should be a part of this repository instead of a separated one. It would make it easier to keep the docs in sync with the documentation, and it would make it possible for ardupilot to host our own mavlink documentation.

hamishwillee, 2021-01-14:
The idea is that no one working with mavlink should have to host their own documentation because it is a standard. If however you mean that you want to create a mavlink doc that is somehow different, I'd like to understand the rationalle - what it might contain and whether that might be done within the one library. We already have too many silos.

More generally I'm not particularly opposed to moving the documentation under this repository other than the work required.

Anyone got other opinions on this?

auturgy, 2021-01-20:
I don't think we need to do this: rather than fork the docs, let's focus on continuing the effort to reduce the delta.
One thing that I think is worth discussing is documenting ardupilotmega.xml better - but that is a separate conversation.

closed 2021-01-20 by auturgy