mavlink/mavlink #1557: GPS_RTCM_DATA microservice is not documented

amilcarlucas, 2021-01-06:
The massage https://mavlink.io/en/messages/common.html#GPS_RTCM_DATA requires a specific statemachine implementation. That is missing from the documented microservices

hamishwillee, 2021-01-14:
@amilcarlucas Yes. Do you know who has done implementations who could help with this?

amilcarlucas, 2021-02-15:
Mavproxy and missionPlanner do thsi QGC probably as well.

hamishwillee, 2021-02-17:
@meee1 @DonLakeFlyer Do either of you know about the GPS_RTCM_DATA microservice sequence? I know from a quick internet service RTCM data is used to improve GPS positioning using DGPS. However it isn't clear what sends and what gets these messages, and where it gets the information from.

DonLakeFlyer, 2021-02-17:
@bkueng knows how this stuff works. He implemented it in QGC.

hamishwillee, 2021-02-18:
Thanks. Yes he does! Notes from our discussion:
1. This is used in RTK GPS systems to transport information from a base GPS that a rover GPS can use to improve its position estimate
2. The message is essentially a pass through from the base GPS connected to GCS, via mavlink to the autopilot, then reassembled and sent direct to the on-vehicle unit.
3. Relevant GCS code is here: https://github.com/mavlink/qgroundcontrol/blob/master/src/GPS/RTCM/RTCMMavlink.cc#L35:L60

I'll have a look at this, but in fact anyone interested can contribute if they need it earlier.

amilcarlucas, 2021-06-21:
This should not be left stale!!

amilcarlucas, 2022-04-19:
Again this should not go stale!!

julianoes, 2022-05-04:
There is also another message which is confusing:
https://mavlink.io/en/messages/common.html#GPS_INJECT_DATA
I would assume this one is the old and should be marked deprecated.

hamishwillee, 2022-05-05:
@julianoes I think you're right.
- On PX4 the [GPS_INJECT_DATA](https://mavlink.io/en/messages/common.html#GPS_INJECT_DATA) doesn't seem to be read anywhere or emitted, but it is the name of the uorb topic that writes to [GPS_RTCM_DATA](https://mavlink.io/en/messages/common.html#GPS_RTCM_DATA).
- It's largely a subset of GPS_RTCM_DATA - missing the flags, but with the ability to set target system/component.

@amilcarlucas What's the story with `GPS_INJECT_DATA` on ArduPilot - if it isn't used I think we should deprecate this now.

amilcarlucas, 2022-05-10:
The old message can be deprecated. It is not used in ArduPilot

hamishwillee, 2022-05-11:
Cool. Deprecated in https://github.com/mavlink/mavlink/pull/1842