mavlink/mavlink #1558: Proposal for more robust COMMAND_LONG response

olliw42, 2021-01-09:
This aspects of the standard procedure for handling COMMAND_LONG and COMMAND_INT commands bothers me for a while:

A command always results in some response, which is send back. As regards the response, the commands however can be distinguished into two kinds, namely

A) ack only: These commands do not have any information to return. The response consists of a COMMAND_ACK only. For instance, MAV_CMD_NAV_TAKEOFF triggers the take off action, but that's it.

B) requests information: These commands implicitly or explicitly imply that some information is returned. For instance, MAV_CMD_REQUEST_MESSAGE triggers the recipient to emit the requested message, but also e.g. MAV_CMD_GET_MESSAGE_INTERVAL and may other commands do have that behavior.

According to the standard, the response to type-B commands should consist of a COMMAND_ACK followed by the message carrying the respective information. 

But that's a problem, in several regards:

* It is not garanteed that the information is being delivered. The command -> command_ack cycle only ensures that the sender knows that the recipient has received and processed the command, but receiving a command_ack does not imply that the sender will also receive the requested information.

* Implementation is somewhat nasty, especially on the recipient side, but also the sender side, since the "state machine" is not just a simple receive & send but needs more involved techniques.

* Bandwidth.

I thus propose the following simplification:

**Proposal**: If a command is of type B and the result is MAV_RESULT_ACCEPTED, then only the respective information is send but no COMMAND_ACK.

Obviously, receiving the respective information is as much a positive acknowledgement as receiving the COMMAND_ACK with MAV_RESULT_ACCEPTED, that is no information is lost and the COMMAND_ACK is superfluous. In fact, the communication becomes more robust since only exactly one response message is the result of the command, and the sender code needs thus less fallbacks as regards to whether the communication was successful or not.

The proposed change would imply a behavior change and as such would break existing implementations. However, a transition could be easily orchestrated, e.g. by first asking emitters of type-B commands (e.g. GCSes) to change their behavior to the proposed behavior, and to ask digesters of type-B commands to follow later. In fact, it should not be difficult to make emitters of type-B commands to accept both behaviors, which would resolve the situation pretty quickly.

peterbarker, 2021-01-09:
Receipt of a message doesn't indicate the command was successful.  For example, some other mavlink client may have requested an instance of a message, and you're just seeing an incidental one.  That might mean you think your `SET_MESSAGE_INTERVAL` succeeded where it didn't.

In fact, I'd prefer to go the other way with these messages.  I'd like an opaque identifier to be included as an extension in `COMMAND_LONG` and `COMMAND_INT` and echoed back by the autopilot in the associated `CMD_ACK` messages.

olliw42, 2021-01-09:
I don't understand what you are saying.

btw, SET_MESSAGE_INTERVAL is type A in the above classification and the proposal would not affect it.

hamishwillee, 2021-01-14:
Hmm. This would reduce bandwidth, and I can see that a GCS could already just ignore the ack and choose to resend or not based on the result.

It just feels like the behaviour and implementation has changed from predictable in most cases to unpredictable. Tha tmakes me feel a bit uncomfortable. Not me that needs to be convinced though!

I will raise in the dev call. FYI @julianoes @LorenzMeier @auturgy

olliw42, 2021-01-14:
what do you my by that it changes behavior to more unpredictable?

hamishwillee, 2021-01-20:
Thanks @olliw42 

We discussed in the dev call. 
- This particular solution is not desirable because  every command would require bespoke handling
- The problem of possibly dropping the response messages is a direct consequence of the fact that MAVLink is deliberately not a reliable transport. The MAVLink solution to this problem is to resend in the application layer.
- We might also define a "microservice" for this - effectively just send command, set timeout on ack, wait for second response (of course we could also have had a dedicated request message, where the response was the emitted message, but we'd still have to handle those dedicated retries etc).

Upshot, thanks, but at this point no.

olliw42, 2021-01-21:
MANY thx for having considered this.

I admit I'm somewhat surprised since I don't understand the counter arguments but I accept that I don't have the complete picture which you do have. Again, many thx.

so I close :)

closed 2021-01-21 by olliw42