mavlink/mavlink #1563: Make static and inline prefixes optional

bjack205, 2021-01-21:
I'm working on interfacing Mavlink with Julia. I may eventually wrap the entire Mavlink C library as part of my research project, and it would be super convenient if Mavlink provided methods that weren't declared `static` or `lnline` so they can be used from a shared library.

For it's C interface, Julia requires a shared library, and I'm unable to call any of the functions in the Mavlink library since they're all prefixed with either `static`, `inline`, or both. 

Could you set up a macro so that these identifiers can be optionally excluded? Right now I have to set up a separate `.c` file that has wrapper functions to all the methods I'm calling from Julia, which is super inconvenient and error-prone, e.g.
```
uint16_t local_position_ned_pack(uint8_t system_id, uint8_t component_id, mavlink_message_t* msg,
                               uint32_t time_boot_ms, float x, float y, float z, float vx, float vy, float vz) {
    mavlink_msg_local_position_ned_pack(system_id, component_id, msg, time_boot_ms, 
        x, y, z, vx, vy, vz); 
}
```
Making the `inline` and `static` prefixes optional would be a much better, cleaner solution! That way I could have access to the entire C mavlink library.

hamishwillee, 2021-01-24:
@peterbarker @LorenzMeier Do you have any thoughts on this one?

peterbarker, 2021-01-24:
@bjack205 without comment on your request here (thought needed), to get you by you could consider generating those shim functions with a relatively straight-forward patch to pymavlink.

bjack205, 2021-01-24:
Thanks for the tip, I'll look into it. I'm guessing this is the function I should modify?
https://github.com/ArduPilot/pymavlink/blob/0e88456d42aff2323f307bafe44ad05680127bda/generator/mavgen_c.py

jasiachang, 2021-01-30:
static uint16_t local_position_ned_pack(uint8_t system_id, uint8_t component_id, mavlink_message_t* msg, uint32_t time_boot_ms, float x, float y, float z, float vx, float vy, float vz) 
{
    mavlink_msg_local_position_ned_pack(system_id, component_id, msg, time_boot_ms, 
        x, y, z, vx, vy, vz); 
}

hamishwillee, 2021-03-17:
I believe this is answered. We aren't planning on doing anything right now, but Peter has suggested an option above. Another option is to create a Julia mavlink Library.

closed 2021-03-17 by hamishwillee