mavlink/mavlink #1564: SET_ATTITUDE and SEND_ATTITIDE_TARGET can't loop.

malfarizit, 2021-01-26:
Hello I'm new here, i have a bit problem when using command `SET_ATTTITUDE_TARGET` and `SET_ATTITUDE`, in this case, I'm trying loop this message for example :
```
while 1:
     set_attitude(pitch_angle = 1 ,duration 1)
```
But can't loop, this message only execute one time, anyone explain why and how to fix this.

Thanks.

hamishwillee, 2021-01-27:
What programming language/library and what flight stack? When you say "the message only execute one time" do you mean it exits the loop after one time or crashes? 

Or are you saying that the function gets called multiple times, but the receiving end only gets the message once? And if so, how do you know?

malfarizit, 2021-01-27:
> What programming language/library and what flight stack? When you say "the message only execute one time" do you mean it exits the loop after one time or crashes?
>
> Or are you saying that the [...]

Hi @hamishwillee thanks for reply.

I'm using python and dronekit on pixhawk.

The indicator while I'm using `print` on loop, and then this print still looping but `attitude_target` only execute one time on left and back hold position.

hamishwillee, 2021-01-27:
Pixhawk is hardware - that doesn't tell me the flight stack. If you're using dronekit though it is a library question, not a mavlink question. You should ask on the dronekit gitter channel.

malfarizit, 2021-01-27:
Alright, sorry for asking.
Thanks shifu.

closed 2021-01-27 by malfarizit