mavlink/mavlink #1567: If I define a message with NO field, the generated source file would have mistakes in compile.

shaopengyuan, 2021-02-02:
I found that if I define a message without a field, the generated source file would have gramma mistake and thus can't be compiled.
For example, if I define a message as followes:
``` xml
<message name="ReqBombNum" id="240">

<description>Require Number of Bombs</description>

</message>
```
and then in the generated c source file, the pack function of reqbombnum message will be:
``` cpp
static inline uint16_t mavlink_msg_reqbombnum_pack(uint8_t system_id, uint8_t component_id, mavlink_message_t* msg,
                              )
```
then it will cause a compiling error because there is a "," before the last ")".

hamishwillee, 2021-02-02:
@shaopengyuan Technically this is a bug, but in practice this never comes up. Typically you might send an empty message to signal some action - and for actions we recommend using a command.

So assuming you wanted to get information back about a number of something you would define message for the information you want returned, and request it using [MAV_CMD_REQUEST_MESSAGE](https://mavlink.io/en/messages/common.html#MAV_CMD_REQUEST_MESSAGE). Note, in addition, you should not use the id 240 for a message unless you are using a closed system.

This could be cross posted to https://github.com/ArduPilot/pymavlink/issues if you have a use case where the above pattern would not work.

hamishwillee, 2021-02-03:
We talked briefly about this in the devcall. The opinion was that this is not really a bug and should be closed. We recommend using a command for this use case.

closed 2021-02-03 by hamishwillee