mavlink/mavlink #1572: Documentation fix: #124 GPS2_RAW, dilution unit wrongly listed as cm

rotorman, 2021-02-03:
Please check the documentation of GPS2_RAW (#124) at: [https://mavlink.io/en/messages/common.html#GPS2_RAW](https://mavlink.io/en/messages/common.html#GPS2_RAW)

It incorrectly lists _cm_ as dilution units for HDOP, VDOP (fields eph, epv). Dilution is unitless.
(On the other hand, first GPS message, [GPS_RAW_INT (#24)](https://mavlink.io/en/messages/common.html#GPS_RAW_INT), has correctly no unit listed).

hamishwillee, 2021-02-04:
So does [HIL_GPS](https://mavlink.io/en/messages/common.html#HIL_GPS).

I think you're right - @julianoes can you confirm?

julianoes, 2021-02-04:
I think @rotorman is right.

hamishwillee, 2021-02-11:
Thanks @rotorman. I have submitted PR for this in #1580

PR #1580 (2021-02-11): Removes units on messages with hdop, vdop
This removes the units from the HDOP, VDOP (fields are generally named eph, epv) on messages: `HIL_GPS`, `GPS2_RAW`, `GPS_INPUT`. Fixes #1572

This is correct technically; however some systems may have somehow "interpreted" the old units.

I did not update [HIGH_LATENCY2](https://mavlink.io/en/messages/common.html#HIGH_LATENCY2) eph, epv fields as these are not described as HDOP/VDOP fields - i.e. I think they really are distances.

[image]

@julianoes @auturgy FYI
closed 2021-02-11 by julianoes