mavlink/mavlink #1575: Unable to clear mission using mavlink After a number of try

jiapau, 2021-02-08:
I am testing the codes in SITL simulator. I used this command to clear missions in ardupilot before loading new missions.

the_connection.mav.mission_clear_all_send(1,1,0)

The mission types are all Mission.

After doing this a number of times. There will comes a problem where it will not be able to clear mission anymore.
It will comes with
MAV_MISSION_ERROR

I tried clearing
the_connection.mav.mission_clear_all_send(1,1,255)
It comes with error 3 which says the command is unsupported

Anyone knows why the Mission Clear all command will generate errors after a few try

I can only by pass this by writing new mission over.

hamishwillee, 2021-02-11:
1. ` mission_clear_all_send(1,1,255)` is invalid tells you that this autopilot does not support clearing mission items, goefences and rally points all at once.
1. [MAV_MISSION_ERROR](https://mavlink.io/en/messages/common.html#MAV_MISSION_ERROR) is documented as meaning "Generic error / not accepting mission commands at all right now.". If you are firing this off a number of times then perhaps you are getting the response while the system is trying to handle the first message you sent. Do you get this every time you send a message (then wait)?

You may find it best to ask this on the ArduPilot forum because they can tell you a bit more about what the actual code is doing. https://discuss.ardupilot.org/c/ground-control-software/mavlink/116

hamishwillee, 2021-02-11:
Note, this is basically what the sequence should look like: https://mavlink.io/en/services/mission.html#clear_mission

closed 2021-02-11 by hamishwillee