mavlink/mavlink #1584: Support for 2 airspeed sensors

Davidsastresas, 2021-02-14:
Related to this issue from QGC: https://github.com/mavlink/qgroundcontrol/issues/9435

Right now Mission Planner and Ardupilot are able to display this second airspeed sensor. They are using a named float mavlink message.

I think the setups with 2 airspeed sensors are quite common nowadays, so it would be great to add support for this in mavlink.

hamishwillee, 2021-02-17:
Mavlink appears to assume you just want to know the current airspeed, not how it is estimated.

Davidsastresas, 2021-02-17:
I saw a couple of Ardupilot users complaining of this fact. In Ardupilot right now the second airspeed sensor is sent by a named float message, and Mission planner has support for interpreting that second airspeed by the named float message.

So it seems the users were used to mission planner, and when they switched to QGC they missed this feature.

Sorry I wasn't able to attend the call, I just saw the dev call label.

hamishwillee, 2021-02-18:
Hi @Davidsastresas 

We did discuss it, but we couldn't resolve it without more information. The information is 
- "why is it needed". That isn't really answered by "ArduPilot users complained that it isn't there".
- "what information do you want" - indicated airspeed, calibrated airspeed, true airspeed, etc. 

You're also talking about a "second" airspeed sensor - but what is the first? If you're using something like VFR_HUD for the "first" then it might not be an airspeed sensor at all. It could be a fused value from multiple sensors, or the fallback of multiple sensors, or an airspeed _calculated_ from vehicle dynamics.

On some systems where there are two airspeed sensors the fallback happens automatically. In this case there is usually a status message that is used to log the changeover and for debugging.

Upshot, can you please more precisely characterise what it is that is required of an airspeed message?

Davidsastresas, 2021-02-18:
Well, I think it is a sensor, as a GPS, as an Imu, etc and we have support for sending several of those by mavlink. Nowadays it isn't strange to have a setup with 2 GPS or 2 airspeed sensors for redundancy. In fact most serious fixed wing platforms have dual airspeed sensors because it is a sensor prone to fail due to obstructions, etc. So I really think it makes sense to have information about both airspeed sensors, don't you think?

pompecukor, 2021-02-18:
> Hi @Davidsastresas
>
> We did discuss it, but we couldn't resolve it without more information. The information is
>
>     * "why is it needed". That isn't really answered by "ArduPilot users complained [...]

Hi,

Let me answer your question. Our Fixed wings and VTOL have dual airspeed sensors. It adds and extra level of safety to have both displayed. If you fly plane, then you know that one extremely important pref-light check to perform is the "blowjob" that is what we call someone blowing air in the pilot and the second operator verifying that the AS is not blocked and is indeed reacting to the air.
This why it was added to mission planner. 
You are right that if there is a failure (in flight) it (might) get reported. But that can already be too late.

You ask why it needed. So there you have it. Not a complaint. It is very important and the only reason it was not there for long is that using dual AS has not been so popular. At this day and age it is one of the most sensitive and vital parts of and airplane.

As for information, what we need is calibrate airspeed. The same that is readily available on the FC and is logged.

[image]

hamishwillee, 2021-02-25:
Thanks guys. OK, so I understand why it is needed now (i.e. "not for estimation and planning").

I still don't know which MAVLink message the data is coming in on. @meee1 Can you tell from above which message Airspeed data would be coming in from?

Davidsastresas, 2021-03-01:
I think that approach of a new AIRSPEED message with proper instance id would be the key, as we are doing now with batteries, etc

hamishwillee, 2021-03-01:
OK. Let's see what the dev call say about other fields we might need.

hamishwillee, 2021-03-10:
FYI, agreed we need new messages. Now we're just in discussion on the fields. That is happening in https://github.com/mavlink/mavlink/pull/1602

hamishwillee, 2021-04-22:
Hi @Davidsastresas 
After much ado, the `AIRSPEED` message has merged into the development.xml file as WIP (see https://github.com/mavlink/mavlink/pull/1602). 

What this means is that you can go back to ArduPilot and Mission planner and suggest that they start emitting this message for both the first and second sensor. As soon as they have a prototype and are sure they are OK with it the message can be moved across into common.xml for general use. Make sense?

Note that PX4 will also be providing an implementation of this at some point.

closed 2021-04-22 by hamishwillee