mavlink/mavlink #1588: Discussion: COMPONENT_INFORMATION extension

bkueng, 2021-02-16:
I'm opening this to discuss the extension of the [COMPONENT_INFORMATION](https://mavlink.io/en/services/component_information.html) API to external components, UAVCAN devices in particular.

At the minimum, it should be possible to get (static) version information and parameter metadata for those devices.

I suggest to extend the `COMP_METADATA_TYPE_VERSION` json file with an optional list of extra components:
```
{
  extra_components: [
  { "id": x, "name": "human readable name, like ESC0", "modelName": xy, "firmwareVersion": xy, "supportedCompMetadataTypes": [1]},
   ...
  ]
}
```
Each device would have its entry, and the autopilot generates the entries during bootup, and keeps it in RAM or writes to SD card (so no hotplugging).

It allows a great deal of flexibility for the autopilot handling & hosting of metadata. From minimal to extended:
- a common (parameter) metadata file is hosted on a (mavlink) server used for all devices. The autopilot just returns that URL.
  (same limitation here as for translation files: the autopilot does not know how to set the `uid`, as the file might change on the server).
- device-specific metadata is hosted on a (mavlink) server. The autopilot sets the URL depending on the connected device type.
- the autopilot knows and serves the metadata itself (could also be a companion)
- the autopilot downloads the metadata through UAVCAN, stores it on the SD card and sets a MAVFTP URL.

### Metadata matching
Metadata needs to be matched against data, for example parameters. I see these options for that:
1) use mavlink component ID's (`id` above would be the comp_id). This is currently how it's implemented in PX4 for UAVCANv0 parameters (component id == UAVCAN node id). Going that path, it essentially means the autopilot emulates MAVLink components, with minimal functionality:
   - sending out heartbeats (so that routing works)
   - handling parameters (currently implemented in PX4)
   - handling COMPONENT_INFORMATION requests
   - handling MAVFTP requests
   - potentially events in the future
2) specifically for parameters: add namespacing via string prefix (`id` above would be the prefix). However param names are limited in length, and multiple devices of the same type need to be considered
3) extend the parameter protocol with an additional ID (a sub component id)

I tend towards 1), but we need to ensure that there's no conflicting component id's.

Something else to consider: we could add fallback url's which are hardcoded in the GCS and used in case the provided URL does not work. The url could have the form: `mavlink.io/<comp_id (autopilot/camera/...)>/<autopilot (apm/px4/...)>/<COMP_METADATA_TYPE>.json`

@auturgy @DonLakeFlyer @hamishwillee

hamishwillee, 2021-02-17:
@bkueng What I think you are saying is:
1. GCS sends an "external thingy manager" component a request for version information (this external manager might be any component, but I assume mostly it will be a companion or the autopilot)
2. That component's version metadata includes a list of `extra_components` which it knows about, each with an id. 
3. In your option 1, is that set of ids are a mavlink component id. The "external thingy manager" acts as a proxy mavlink component for all those component ids - i.e. it emits heartbeats and requested messages when requested as those they were real mavlink components.

Is that right?

- Mapping nodes to [component ids](https://mavlink.io/en/messages/common.html#MAV_COMPONENT) might be problematic. In particular 0, 1 get a lot of use. I guess we could add a node id in the metadata and map the two.
- I presume we never need to differentiate across bus - ie if we start adding something other than UAVCAN too?
- We never need more than 250ish nodes?
- I believe PX4 only emits one heartbeat. It accepts messages intended for a number of components though. 
- Another option is that we drop the  `extra_components`. The "external thingy manager" could then just emit heartbeats and the extra components be discovered as normal.

I like the simplicity of having something pretend to be a mavlink node. 

@julianoes Is there anything we can learn architecturally on how to solve this from the gimbal manager?

hamishwillee, 2021-02-18:
Discussed in dev call. 

Emulating non-mavlink nodes as components:
1. Solution conceptually simple, and means that we could use mavlink "as per usual" to ask offboard components for component information, to set their parameters, etc.
1. Significant increase in message traffic if done properly as each node must emit its own heartbeat for routing etc. That's probably around 20 extra nodes on a typical system, which is quite a lot more traffic for something you don't always need.
1. Architecturally difficult for PX4 to emulate components (discussion around effort to get sequence numbers right).

Another proposed option is that autopilot (say) acts as proxy for requests for information. This is similar to Beat's proposal
- Request version info, and in that get information about attached attached non-mavlink components. The ids can be anything we like.
- To request detail about each external component make a new [MAV_CMD_REQUEST_MESSAGE](https://mavlink.io/en/messages/common.html#MAV_CMD_REQUEST_MESSAGE) to same component, specifying the external component in an unused param.
- This allows us to request whatever information we like about a non-mavlink component.

So basically the second solution does the "minimum" needed in terms of being able to _find out_ about attached components. 

If we want to write to parameters etc, that probably isn't something we consider in the context of component information.

closed 2021-06-23 by hamishwillee