mavlink/mavlink #1589: MAV_COMPONENT IDs for Smart Battery

sypaq-nexton, 2021-02-17:
Is there value in adding a component ID for smart batteries? E.g. 

```xml
    <enum name="MAV_COMPONENT">
      ...
      <entry value="180" name="MAV_COMP_ID_SMART_BATTERY">
        <description>Smart Battery #1.</description>
      </entry>
      <entry value="181" name="MAV_COMP_ID_SMART_BATTERY2">
        <description>Smart Battery #2.</description>
      </entry>
      ...
```

Happy to raise a PR if there's interest.

hamishwillee, 2021-02-17:
A good question. 

In short, I think we should define MAV_TYPE_BATTERY and it would make sense to define at least `MAV_COMP_ID_SMART_BATTERY` (and probably both). @julianoes can you sanity check my logic below.

----

If the smart battery connects direct to the autopilot via smbus or whatever then it is not a mavlink component. The mavlink component in this case is the autopilot so the MAV_TYPE and component ID are of the autopilot. Anything receiving the message knows the source from the system id and differentiates the batteries from the battery number field. Three are no messages going TO the battery, but if there were, the routing would be expected to be done via the autopilot based on a battery number field in said message.

If the smart battery is a mavlink node on the network it needs a system id, component id, and MAV_TYPE. 
- Typically a battery might infer this system id by listening for the autopilot id and matching it
  - (on multi-vehicle systems you would need to ensure that either the battery did not get heartbeats from other systems OR provide an interface to explicitly set the system ID). 
  - The MAV_TYPE would be hard coded to reflect the battery MAV_TYPE_BATTERY. This isn't such a big deal because we aren't addressing messages to the battery, but we might in future. However it also helps because the GCS can infer the type from heartbeat. 
  - The component id has to be unique in the system. It is useful for this to be pre-allocated with something like MAV_COMP_ID_SMART_BATTERY because that make a default setup plug-n-play conflict free. However if a system has two batteries that are both connected over mavlink then they will both in theory need their own component id so you will need some way to modify these.

Upshot, I think that defining these IDs probably makes sense, along with a MAV_TYPE. However they will only ever get used for batteries connected to PX4 over mavlink.

hamishwillee, 2021-02-18:
We had a chat about this. The consensus was "yes please", but the use of "SMART" is probably redundant (so do `MAV_COMP_ID_BATTERY`, and could we also have `MAV_TYPE_BATTERY` ?

sypaq-nexton, 2021-02-18:
Thanks @hamishwillee. I'll raise a PR now.

hamishwillee, 2021-03-17:
Closing as the PR was accepted. Thanks @sypaq-nexton - great to have this in. FYI I added info about these in the docs https://mavlink.io/en/services/battery.html#battery-components

closed 2021-03-17 by hamishwillee