mavlink/mavlink #1605: Reduced Bandwidth, Telemetry Optimized Messages

flybrianfly, 2021-03-06:
Are there plans to add reduced bandwidth messages optimized for telemetry? I'd be happy to help contribute code.

For example, I'd like to integrate mavlink with my [flight control system](https://github.com/bolderflight/). As a starting point, I'm only considering the transmission of  vehicle state data for real-time visualization in a GCS, specifically:
   * Attitude
   * GNSS position
   * Airspeed (or ground speed if not a fixed-wing)
   * Battery health (voltage and current)

This seems like a common and fairly minimal data set to transmit to a GCS. Currently, this requires sending the [attitude message](https://mavlink.io/en/messages/common.html#ATTITUDE), the [GNSS raw int message](https://mavlink.io/en/messages/common.html#GPS_RAW_INT), the [VFR HUD message](https://mavlink.io/en/messages/common.html#VFR_HUD), and the [Battery Status Message](https://mavlink.io/en/messages/common.html#BATTERY_STATUS).

The message sizes are:
   * Heartbeat: 9 bytes
   * Attitude: 28 bytes
   * GNSS: 52 bytes
   * HUD: 20 bytes
   * Battery: 49 bytes

Assuming that I send the heartbeat and battery data at 1 Hz, the attitude and HUD data at 25 Hz (not ideal, but kind of smooth to the human eye), and the GNSS at 5 Hz (update rate from the receiver), I'm at a required bandwidth of 12,144 bit/s. For a low-cost SiK radio running at 57,600 baud, that's optimistically 21% of my link budget and probably somewhat worse.

I understand that mavlink is meant to be a generic messaging system, but for the purposes of telemetry, I don't understand why the [attitude message](https://mavlink.io/en/messages/common.html#ATTITUDE) would include the body angular rates, how would that information be displayed on the GCS and how would an operator use it? Similarly, I would not include the DoP or uncertainty values for the GNSS data and groundspeed is redundant between the GNSS and HUD messages. These changes would reduce the bandwidth to about 8,944 bit/s or 74% of the original required bandwidth.

But there's further optimization available. I probably can't read attitude data on a GCS display to better than 0.1 degree anyway, so I could store the attitude data in 2 bytes for each channel instead of a 4 byte float. Similarly I can compress headings, speeds, and altitudes into 2 bytes each instead of 4. These changes would reduce the bandwidth to about 5,904 bit/s or 48% of the original requirement.

We aren't talking bit packing or any extreme compression schemes, just developing an optimized set of messages specifically for telemetry purposes. Is this a path worth pursuing? I'd be happy to help, but would like to make sure it's a direction the community at large would like to move in.

hamishwillee, 2021-03-17:
@flybrianfly Sorry for the delay in answering - I missed this one.

The short answer is no, there is no ongoing work in this area. The slightly longer answer is that you might find [HIGH_LATENCY2](https://mavlink.io/en/messages/common.html#HIGH_LATENCY2) useful. This is intended for systems with very slow and expensive bandwidth (i.e. satellite radio). It is relatively "big" but packages the essential information for a system to update it's status at very low rate (i.e. it is efficient).

The "why is it so" on particular messages is hard to answer (though you can look at various PRs in this system to get some idea of why things are as they are). Someone needed it at some point. Perhaps they were wrong and we've carried the burden of poor design ever since. FWIW you probably don't need to send the battery message at very high rate. It is possibly important to note that there are thousands upon thousands of vehicles flying with low cost SIK radios using MAVLink, so it is certainly do-able.

We are adverse to modifying messages in an incompatible way. We will however sometimes add new ones to replace flawed messages. 

I will raise this in the next devcall to see if anyone else would like to comment.

flybrianfly, 2021-03-17:
Hi @hamishwillee, thanks for the reply.

Yes, it seems like most of the messages would not need to be high rate. Indeed, I was recently testing mavlink with Mission Planner and it requested messages between 2 - 4 Hz. I suppose the only high rate, low-latency data would really be attitude, if the pilot is using a HUD for feedback, and airspeed. I've operated some UAS for research, which did not have a large range between approach speed and stall speed and required constant feedback on airspeed for the pilot.

To be clear, I wouldn't modify existing messages. Rather I think create a new message set for telemetry. The real issue would be getting QGC and Mission Planner to adopt the new messages.

hamishwillee, 2021-03-18:
@flybrianfly The dev call response was pretty much "no thank you". 

Essentially we have a set of message that have been proved to work well even with low cost radios over many years on many thousands of vehicles/GCS. The fact that some of the messages may waste some bandwidth is not proof of a problem that justifies the cost of making a breaking change for the GCS and flight stacks that use them (yes I understand you're talking about an addition, but the GCS would still need to cope with the possibility of both cases).

In other words, the broad compatibility of the message set is more valuable than saving a few bytes here and there.

Thanks for asking the question.

closed 2021-03-18 by hamishwillee