mavlink/mavlink #1608: Resolution of GPS / position values

chrislanzl, 2021-03-10:
It looks like currently the values of latitude/longitude are cut off in both GLOBAL_POSITION_INT and (the raw values in) GPS_RAW_INT.

Our gps module (RTK) supports resoultion of sub 10cm. So we lose like two digits after the comma.

I would suggest to support the higher resolution GPS by adding more digits to the respecting data fields, e.g. degE9.
Downside is of course that a 32 Bit integer won't be enough anymore.

Sorry, I'm quite new to mavlink. Probably this cannot be done because it owuld break compatibility mavlink1? 
Can a new message be added for RTK GPS

LorenzMeier, 2021-03-10:
We certainly can add a new message. If I did the math correctly the resolution of 1e7 is 2 cm. So yes, it might be visible for highest-performance RTK receivers, but for navigation purposes I would doubt that you can get a noticeable difference with current systems in a practical use case.

That said: We always should strive to support the latest and greatest and this feels like we should be using double precision floating point and draft a new message. A pull request is welcome!

hamishwillee, 2021-03-17:
My understanding is that we used a scaled integer int32 because a float, while occupying the same number of bytes, resulted in a precision error of several 10s of metres (vs that few cm for the scaled value). Working in unscaled values is much more natural, so provided the precision from double is sufficient for our current and predicted needs (and I am pretty sure it is) that would be "better".

hamishwillee, 2021-03-18:
This was discussed in the dev call. The general consensus was that the 2-ish cm resolution from the existing scaled integers is sufficient for current use cases - see above https://github.com/mavlink/mavlink/issues/1608#issuecomment-795491680

If you were able to demonstrate a real-world prototype showing the benefit of higher resolution messages we would of course be very pleased to accept them. I am closing this for now. Thanks for your suggestions!

closed 2021-03-18 by hamishwillee