mavlink/mavlink #1613: Generate JavaScript code using mavgenerate.py

nopnop2002, 2021-03-22:
I generated JavaScript code using mavgenerate.py.
But uint8_t [12] is generated as a string.

```
$ cat example.xml
<?xml version="1.0"?>
<mavlink>
        <include>common.xml</include>

        <version>1</version>
        <dialect>0</dialect>

        <messages>
                <message id="21100" name="EXAMPLE">
                        <description>Example</description>
                        <field type="uint32_t" name="data0">Example</field>
                        <field type="uint8_t[12]" name="data1">Example</field>
                        <field type="uint8_t[12]" name="data2" >Example</field>
                        <field type="uint16_t[12]" name="data3">Example</field>
                        <field type="float[12]" name="data4">Example</field>
                </message>

        </messages>
</mavlink>
```

```
mavlink20.messages.example = function(data0, data1, data2, data3, data4) {

    this.format = '<I12f12H12s12s';
    this.id = mavlink20.MAVLINK_MSG_ID_EXAMPLE;
    this.order_map = [0, 3, 4, 2, 1];
    this.crc_extra = 159;
    this.name = 'EXAMPLE';

    this.fieldnames = ['data0', 'data1', 'data2', 'data3', 'data4'];

    this.set(arguments);

}
```

- Incorrect
this.format = '<I12f12H12s12s';

- Correct
this.format = '<I12f12H12B12B';

nopnop2002, 2021-03-23:
I found reason.

In case of python:
The format is [s] only when field.type is an array of char.
https://github.com/ArduPilot/pymavlink/blob/0e88456d42aff2323f307bafe44ad05680127bda/generator/mavgen_python.py#L458

In case of JavaScript:
The format is [s] when field.type is an array of int8_t and uint8_t as well as char.
https://github.com/ArduPilot/pymavlink/blob/0e88456d42aff2323f307bafe44ad05680127bda/generator/mavgen_javascript_stable.py#L285

I don't know why.

hamishwillee, 2021-03-24:
If this is a problem (I don't know enough to assess) can you please [repost the issue here](https://github.com/ArduPilot/pymavlink/issues) and link here?

(basically this is a generator issue, so we put it alongside the generators).

nopnop2002, 2021-03-24:
@hamishwillee 

OK.
I'll report.

closed 2021-03-24 by nopnop2002