mavlink/mavlink #1615: field name "id" in statustext message creates naming conflict in JS

smkaeser, 2021-04-08:
In JS there is a conflict between the field name id and the message id that breaks creating statustext messages in JS. Please change the field name similar to #252.

```
mavlink20.messages.statustext = function(severity, text, id, chunk_seq) {
                                                                                          ^^ 
    this.format = '<B50sHB';
    this.id = mavlink20.MAVLINK_MSG_ID_STATUSTEXT;
          ^^
    this.order_map = [0, 1, 2, 3];
    this.crc_extra = 83;
    this.name = 'STATUSTEXT';

    this.fieldnames = ['severity', 'text', 'id', 'chunk_seq'];
                                                           ^^

    this.set(arguments);

}
mavlink20.messages.statustext.prototype = new mavlink20.message;
mavlink20.messages.statustext.prototype.pack = function(mav) {
    return mavlink20.message.prototype.pack.call(this, mav, this.crc_extra, 
               jspack.Pack(this.format, [ this.severity, this.text, this.id, this.chunk_seq]));
                                                                                            ^^^^^
}
```

smkaeser, 2022-06-03:
The same problem exists for messages with a field-name "name". E.g. the messages NAMED_VALUE_INT and NAMED_VALUE_FLOAT. This seems to be a more global issue that should be solved independent from the individual field-names of each message.

hamishwillee, 2022-06-08:
@peterbarker Would appreciate your thoughts on this - or could you tell me who might help for the Javascript generator?

There was a "fix" to xsd way back in https://github.com/mavlink/mavlink/pull/252 to ensure the name attribute (used in enum, entry, message, field) could not be a javascript keyword. This does not appear in the [current schema](https://github.com/ArduPilot/pymavlink/blob/master/generator/mavschema.xsd#L12-L17) which only checks it's a word character with no whitespace.

We could add that back, but of course immediately break anything that is misnamed - like NAMED_VALUE_INT. ALso it wouldn't fix the "int".

I think the problem probably needs to be fixed at the generator end, but I'd like to at least confirm that is the right approach, and move this error if it is.