mavlink/mavlink #1617: MAVLink signature check in C library cannot verify signature generated by mavlink_sign_packet

JXS2012, 2021-04-14:
The mavlink_signature_check function in mavlink_helpers.h is not able to verify signature generated by mavlink_sign_packet function. The issue seems to come from the mavlink_message_t definition.

The following line in mavlink_signature_check assumes that the message ID field took 3 bytes:
mavlink_sha256_update(&ctx, p, MAVLINK_CORE_HEADER_LEN+1+msg->len);
However, mavlink_message_t stores the message ID in uint32 and took 4 bytes. This misalignment lead to failure in signature verification.

The compiler used is Visual Studio 2019, compiled on Windows.

davidbuzz, 2021-05-25:
type 'mavlink_message_t' contains a 'msgid' field that according to the spec is "three bytes", ie 24 bits, not 32. 
spec:
https://mavlink.io/en/guide/message_signing.html#frame-format
C generator output treats it as a 24-bit "bitfield" inside a struct.
 https://github.com/mavlink/c_library_v2/blob/1b8cc8e008cec07367d190411bc8b6e0a4bda484/mavlink_types.h#L117

Where is the problem?  

In short, the data IS "three bytes" not 4.   If your code is taking mavlink_message_t.msgid and copying it into a standard uint32, then its at that point that it becomes 4 bytes, one of which is irrelevant/ignored.

i recommend closing this ssue, as i see nothing wrong here, just  @JXS2012 's misunderstanding of packed bitfields in structs.

JXS2012, 2021-05-26:
When I test this library compiled with VS2019, the mavlink_message_t is storing the msgid using 4 bytes in memory instead of 3 bytes, even though it is declared as a 24 bitfield in the struct. I don't think bitfield is going to pack the memory here since mavlink_message_t doesn't contain any other value in bitfield form.

JXS2012, 2021-05-26:
#include <iostream>
struct S {
    unsigned int b1 : 24;
    unsigned int b2;
};
int main()
{
    std::cout << sizeof(S) << '\n'; // prints 8 bytes for me instead of 7
}

davidbuzz, 2021-05-26:
Bitfields ordering is known to be compiler and platform implementation depenant. GCC on Linux happens to be a compiler that produces spec-compliant bit packing by default. It is known that others don't.   For example, i found that mingw has a compiler flag that makes it compliant , but its not by default.

davidbuzz, 2021-05-26:
if ( 'x86_64-w64-mingw32-g++' in cfg.env.CXX  ) or ( 'i686-w64-mingw32-g++' in cfg.env.CXX ): 
            env.CXXFLAGS += [ 
            '-mno-ms-bitfields' # without this mingw targetting windows builds uses a packing format not compatible with the normal gcc one. 
            ];

JXS2012, 2021-05-26:
I'd suggest update the mavlink_signature_check so that it is no longer compiler dependent. The following line

mavlink_sha256_update(&ctx, p, MAVLINK_CORE_HEADER_LEN+1+msg->len);

can be split into two parts to verify the header and payload separately:

mavlink_sha256_update(&ctx, p, MAVLINK_NUM_HEADER_BYTES);
mavlink_sha256_update(&ctx, _MAV_PAYLOA(msg), msg->len);

hamishwillee, 2021-06-02:
@davidbuzz Does ^^^ https://github.com/mavlink/mavlink/issues/1617#issuecomment-849002424 seem reasonable?

I'm not competent to judge the proposal, but if it allows the check to be compiler independent that would be a good thing.

davidbuzz, 2021-06-02:
I'd say that if a PR was opened with this change, and it passed all the the CI tests under linux, then accepting a change like this that makes it easier to use with VS2019 would be reasonable.

hamishwillee, 2021-06-02:
Great. @JXS2012 Could you please create a PR for your proposed change - I'm guessing to https://github.com/ArduPilot/pymavlink/blob/master/generator/C/include_v2.0/mavlink_helpers.h#L136 ?

JXS2012, 2021-06-02:
What is the qualification steps needed from my side? Is there a set of tests that I should run? It is not very clear on this page: https://mavlink.io/en/contributing/contributing.html

davidbuzz, 2021-06-02:
> What is the qualification steps needed from my side? Is there a set of tests that I should run? It is not very clear on this page: https://mavlink.io/en/contributing/contributing.html

All the repositories in question have "autotests", also known as "CI" or Continuous Integration", these tests are magically run by github's server/s whenever a PR is opened.   
EG:
https://github.com/mavlink/mavlink/actions
https://github.com/ardupilot/mavlink/actions
https://github.com/ardupilot/pymavlink/actions

look into the respective folder of each repo to know more:
.github/workflows/

Open a PR with code that u think fixes the issue, and it *will* get tested.   :-)

hamishwillee, 2021-06-03:
FYI, I've added a PR to capture in docs that you don't have to run specific tests yourself - but your PR has to pass CI https://github.com/mavlink/mavlink-devguide/pull/363

JXS2012, 2021-06-03:
Here is a link to the signature check pull request: https://github.com/ArduPilot/pymavlink/pull/549