mavlink/mavlink #1619: MAVLink signature check in C library does not check signing_streams's link_id when creating new stream

dongyy1148, 2021-04-14:
The mavlink_signature_check function in mavlink_helpers.h does not check signing_streams's link_id when creating new stream.

For example, if I create two channels including two mavlink_signing_t* s1 and s2. s1 and s2 has different linkID, since they are using different links. However, I give them the same secret key, so they will have same 6 bytes signature. If I send a message generated by s1 to s2, it will get into mavlink_signature_check function and pass the 6 bytes signature check. Then s2 will accept the message and create a new stream like:
```
		signing_streams->stream[i].sysid = msg->sysid;
		signing_streams->stream[i].compid = msg->compid;
		signing_streams->stream[i].link_id = link_id;
		signing_streams->num_signing_streams++;
```

And it will not care whether s1 and s2 mavlink_signing_t has different link ID. Is this behavior expected? Based on your description on your website: 
_The purpose of the link_id field in the MAVLink 2 signing structure is to prevent cross-channel replay attacks. Without the link_id an attacker could record a packet (such as a disarm request) on one channel, then play it back on a different channel._

I don't think s2 should be able to accept the message from s1, since they have different link IDs.

The compiler used is Visual Studio 2019, compiled on Windows.

hamishwillee, 2021-04-22:
@julianoes Can you look at this or find someone else technically capable of doing it? Or is this something that should be posted in ArduPilot/pymavlink because it is a tools issue.

I had a look at [mavlink_signature_check](https://github.com/mavlink/c_library_v2/blob/master/mavlink_helpers.h#L120). Wasn't sure.

julianoes, 2021-04-29:
I'm sorry. I don't have capacity to look into signing right now.

hamishwillee, 2021-06-02:
@dongyy1148 As this affects the guts of the C library, could you please post it in the project that manages the mavgen toolchain? I.e. post this in https://github.com/ArduPilot/pymavlink/issues and then add a link back to this issue?

dongyy1148, 2021-09-01:
> @dongyy1148 As this affects the guts of the C library, could you please post it in the project that manages the mavgen toolchain? I.e. post this in https://github.com/ArduPilot/pymavlink/issues and then add a link back to this issue?

Sure