mavlink/mavlink #1625: Size of compiled MAVLink function calls

damurashov, 2021-04-29:
Hi!

I've encoutered a problem which is that calling mavlink functions (I use `c_mavlink_v2`, release date is 2018 or so) compiles into huge chunks of code. E.g. `mavlink_msg_sys_status_encode` alone weighs 7Kb. 

My project is running out of memory (embedded), and removing its other modules is not an option.

Could you highlight any workarounds, give me some hints?

Thank you!

damurashov, 2021-04-29:
By the way, I believe there is a chance that the culprit is use of `#define`s. 

```c++
// ...
#define _mav_put_uint32_t(buf, wire_offset, b) byte_swap_4(&buf[wire_offset], (const char *)&b)
#define _mav_put_int32_t(buf, wire_offset, b)  byte_swap_4(&buf[wire_offset], (const char *)&b)
#define _mav_put_uint64_t(buf, wire_offset, b) byte_swap_8(&buf[wire_offset], (const char *)&b)
// ...
```

Do you think I could replace it globally somehow?

hamishwillee, 2021-05-05:
@damurashov Yes. Two options. Firstly you could look at https://github.com/mavlink/mavlink/issues/1094 - that's the last time this came up. There are some assumptions in there that may or may not suit your case:

> Actually I had a big buffer to store the message for mavlink_msg_to_send_buffer().
After reducing that and reducing MAVLINK_COMM_NUM_BUFFERS and MAVLINK_MAX_PAYLOAD_LEN it can fit in SRAM of pro mini. Only 50% used (1kB).

> I did was a hack for my very particular situation. I do not recommend to use my modifications, because I assume to have only one comm link and a maximum payload of 70 bytes which might not be true for other applications.

The other alternative you might consider is using https://github.com/olliw42/fastmavlink - **this is not supported or validated by MAVLink project**. But OllieW who wrote it says that it is much smaller and more efficient - so may well be worth a look. I'd be interesting in finding out what your profiling indicates.