mavlink/mavlink #1641: Addition enum for HIL_SENSOR.fields_updated and HIGHRES_IMU.fields_updated

python36, 2021-05-21:
I suggest adding enumerations for `HIL_SENSOR.fields_updated` and  `HIGHRES_IMU.fields_updated`.

hamishwillee, 2021-05-26:
Hmmm, so 
1 - HIL_SENSOR_UPDATED_XACC
4 - HIL_SENSOR_UPDATED_YACC
8 - HIL_SENSOR_UPDATED_ZACC
...
?- HIL_SENSOR_UPDATED_TEMP

It's definitely not "wrong", but I'm not sure how useful it would be either. @julianoes got an opinion on this?

python36, 2021-05-26:
This is much clearer. For example now like this:
```
hil_sensor_send(
    time_usec=0, xacc=0, yacc=0, zacc=1, xgyro=0, ygyro=0, zgyro=0, xmag=1,
    ymag=0, zmag=0, abs_pressure=0, diff_pressure=0, pressure_alt=0, temperature=1,
    fields_updated=4164)
```
when there will be enums:
```
hil_sensor_send(
    time_usec=0, xacc=0, yacc=0, zacc=1, xgyro=0, ygyro=0, zgyro=0, xmag=1,
    ymag=0, zmag=0, abs_pressure=0, diff_pressure=0, pressure_alt=0, temperature=1,
    fields_updated=(HIL_SENSOR_UPDATED_ZACC + HIL_SENSOR_UPDATED_XMAG + HIL_SENSOR_UPDATED_TEMPERATURE))
```

hamishwillee, 2021-05-27:
Good example. I'm not opposed. Can you please create a PR for these?