mavlink/mavlink #1654: c_library_v2: MAVLINK_CHECK_MESSAGE_LENGTH doesn't work with Empty-Byte Payload Truncation

sypaq-nexton, 2021-07-09:
Adding this issue here, as I can't add it directly to the https://github.com/mavlink/c_library_v2 repo.

When using the `MAVLINK_CHECK_MESSAGE_LENGTH` `#define`, messages can be unexpectedly dropped. It appears to be because MAVLink2 Empty-Byte Payload Truncation (as detailed [here](http://mavlink.io/en/mavgen_c/#additional-checks)) causes messages to have shortened lengths, which fails the message length check.

I believe the fix is to only check the minimum length for MAVLink1 messages, e.g. change the code at [mavlink_helpers.h#L705](https://github.com/mavlink/c_library_v2/blob/master/mavlink_helpers.h#L705) to something like:

```cpp
#ifdef MAVLINK_CHECK_MESSAGE_LENGTH
        if ((rxmsg->len > mavlink_max_message_length(rxmsg)) ||
            ((status->flags & MAVLINK_STATUS_FLAG_IN_MAVLINK1) && 
             (rxmsg->len < mavlink_min_message_length(rxmsg))))
        {
            _mav_parse_error(status);
            status->parse_state = MAVLINK_PARSE_STATE_IDLE;
            break;
        }
#endif
```

A similar check is done just prior to [mavlink_helpers.h#L678](https://github.com/mavlink/c_library_v2/blob/master/mavlink_helpers.h#L678), which is the only other place `MAVLINK_CHECK_MESSAGE_LENGTH` is used.

hamishwillee, 2021-07-14:
Howdy!

1. Thanks for cross posting this to https://github.com/ArduPilot/pymavlink/pull/569 - that's where the tools are maintained. 
2. I think you might be correct about the problem though. The `rxmsg->len` that comes from the message is the actual payload length (truncated). The `mavlink_min_message_length` compares against a minimal length that appears to be based on the size of the memory required for a full payload.
3. I don't think there is much point changing the code in mavlink_helpers.h#L705 to be conditional on mavlink 1 because this is following the statement `case MAVLINK_PARSE_STATE_GOT_MSGID2:` which you shouldn't get to in mavlink 1 (which only has one byte for ids).

FYI @peterbarker

sypaq-nexton, 2021-07-14:
Thanks @hamishwillee. You're right -- we can just remove the length check completely in the `MAVLINK_PARSE_STATE_GOT_MSGID2` case. I've tested it locally, and it all seems to work. PR has been updated.