mavlink/mavlink #1655: How to unify the origin of the NED coordinate system for all UAVs in a multi-unmanned system?

swarm5, 2021-07-13:
In multi-unmanned systems, it is very necessary to unify the origin of the local coordinate system for all UAVs.
The origin of the local coordinate system is the home point, and I think we can achieve the unification of the local coordinate system of all UAVs by setting the home point of all UAVs to a value of the same latitude and longitude height in this way.
However, I have verified using mavros, gazebo, and in fact this solution does not work. I can check the telemetry information of UAVs in QGC, and the result shows that the home point has been changed successfully, but the position information of local_ned is not changed, which means the origin of local_ned is the takeoff point of UAVs instead of home point?

hamishwillee, 2021-07-14:
Probably - @TSC21 do you know what PX4 (and others) use to set the origin of the local ned frame? I.e. is it arming point or takeoff point or something similar? Can it be changed? Does it have a relationship to whatever MAVLink thinks is home?

@swarm5 The home point is usually the point at which the vehicle arms - at least when using a global coordinate frame/GPS. MAVLink doesn't specifically define this, and it is not mandatory that it can be changed using https://mavlink.io/en/messages/common.html#MAV_CMD_DO_SET_HOME. Further, I don't think that MAVLink states the origin for a local frame, but I suspect it will be the arming point. I did not know that there was a "Home" point in the local frame (or if it existed that it was different than the origin of the local frame). How are you checking the Home location?

What flight stack are you using?

hamishwillee, 2021-07-14:
PS @TSC21 if you were using a swarm in local frame, how would you solve this problem?

TSC21, 2021-07-14:
> PS @TSC21 if you were using a swarm in local frame, how would you solve this problem?

One can use https://mavlink.io/en/messages/common.html#SET_GPS_GLOBAL_ORIGIN.

hamishwillee, 2021-07-15:
@TSC21 How would that work for a LOCAL frame as per this question? 

Note, I am assuming this is an indoor swarm, or similar.

TSC21, 2021-07-15:
> @TSC21 How would that work for a LOCAL frame as per this question?
> 
> Note, I am assuming this is an indoor swarm, or similar.

For the local frame would work as long as the reprojection is computed from the global to the local coordinates in PX4. Without global position though, there's no concept of common local frame, unless you force an offset on the estimation of each vehicle (which can be done out of PX4, in the companion/mission computer). But this is not defined by Mavlink or PX4.

hamishwillee, 2021-07-21:
Thanks @TSC21 . Closing this as "answered".

closed 2021-07-21 by hamishwillee