mavlink/mavlink #1656: Is COMPONENT_INFORMATION still WIP ?

patrickelectric, 2021-07-15:
There is no issue to work over it or any deadline to have it ready, what is the current status ?

hamishwillee, 2021-07-15:
Not really. It is in released products and appears to be working well. However it hasn't been "signed off" and WIP tagging removed. I'll confirm we're all happy that it is stable in the next dev call.

hamishwillee, 2021-07-15:
@bkueng Anything look like it requires modification? Did the translation stuff get resolved?

bkueng, 2021-07-16:
Translation details were still open afaik. @DonLakeFlyer wanted to look into this.

DonLakeFlyer, 2021-07-20:
I didn't think there was missing spec on translation just that it's NYI in QGC. If so, I don't remember what is missing from the spec.

hamishwillee, 2021-07-21:
@DonLakeFlyer What does NYI mean?

My understanding, FWIW, is that all we have defined for translation is that the general metadata has sections for each supported metadata, and that each such section contains the URI where we can find translation for the metadata - e.g. see `translationUri` in https://github.com/mavlink/mavlink/blob/master/component_information/general.schema.json#L57

I don't think we've defined what the metadata file looks like. I would have guessed that for parameters say it would also be json, where the parameter name is the key, and we have nested translation fields for shortDesc and longDesc. Do you already have a defined format in which you'd take such metadata?

DonLakeFlyer, 2021-07-21:
I thought we talked about going with the QT .ts format since it is supported by Crowdin. Also the QGC tools can generate .ts files from json.

hamishwillee, 2021-07-21:
> I thought we talked about going with the QT .ts format since it is supported by Crowdin. Also the QGC tools can generate .ts files from json.

@DonLakeFlyer I don't remember that. 
- Can Qt do that conversion at runtime from json? 
- Not every system runs with Qt, and json is much more generic. 

Assuming json we would need to define the schema.

DonLakeFlyer, 2021-07-22:
> * Can Qt do that conversion at runtime from json?

No. You have to implement support for it yourself. The file format is just convenient. The Qt format is pretty simple xml file schema which Crowdin already understands. Using has little to do with actually using Qt to process it in this case.

hamishwillee, 2021-07-28:
@bkueng You're perhaps best person to evaluate the format: https://support.crowdin.com/file-formats/ts/

I don't quite get the detail of how this would work - specifically how we'd map the component information file into a set of translation files.

bkueng, 2021-07-28:
AFAIK there's one step still missing: Qt [expects .qm files at runtime](https://doc.qt.io/qt-5/qtranslator.html#load), which are generated from .ts files using a tool [lrelease](https://doc.qt.io/qt-5/linguist-manager.html) at QGC build time.

I also thought the file can contain multiple translations. Seems not to be the case though.

DonLakeFlyer, 2021-07-31:
QGC will need to implement it's own QTranslator override which reads .ts files dynamically at runtime and install it into the app. All doable.

> I also thought the file can contain multiple translations. Seems not to be the case though.

Hmm, that is a problem. My suggestion would be to create an xml container spec around the .ts xml spec. Then you use crowdin .ts translation support and in the firmware build process somewhere you combine the individual .ts language files into the uber-ts file which can support multiple languages in a single xml file.

hamishwillee, 2021-08-04:
Feels odd to have XML for translation when proposing JSON for everything else. I guess there is no reason NOT to use it, but I would like to see what this actually looks like in TS for some strings - specifically how the context is provided. The original example looks like this:
```
<context>
  <name>Page</name>
  <message>
    <source>Text for translation</source>
    <comment>commenting</comment>
    <translation type="obsolete">translated string</translation>
  </message>
  <message>
     <source>Text for translation #2</source>
     <extracomment>some text</extracomment>
    <translation>some translation</translation>
  </message>
</context>
```
To provide a translation of the general information file we need to map strings for these keys: vendorName, modelName, firmwareVersion, hardwareVersion - what would that look like? I guess "Page" is general_information.json (or whatevever) but then the mapping appears to be based on just the source text, not the specific keys. Feels a bit "square peg for round hole".

hamishwillee, 2021-08-12:
So FWIW I think COMPONENT_INFORMATION, as in the message should no longer be WIP - it is in a public release. However we do still need to address translation above ^^^. @bkueng I assume you are driving this (?) and that at some point it will reach your list of priorities?

bkueng, 2021-08-12:
My assumption is that Don does, as it's on the GCS side and he has a clear idea for implementation.

hamishwillee, 2022-02-24:
I would still _prefer_ json to .ts for translations. 
- We're not longer pulling into Qt directly, but via a parser, so there is no particular benefit in the .ts format. 
- That means recipient needs just one parser. 
- Crowdin supports json, but we would have to define our workflow for this and we would still need a container format
- We don't want to mandate a format that (other) translation services might not understand. Json would be pretty generic. But I guess maybe .ts (Qt, not typscript) is pretty common.

hamishwillee, 2022-02-24:
Assuming we agree a format(s), what are the next steps?

My understanding is that once the format is agreed:
- @bkueng :
  - Will create a parser to generate source files from our parameter definitions that crowdin can use for translation
  - Will create parser to aggregate translated files into one file, and host somewhere
- @DonLakeFlyer :
  - Update fetch code in component information implementation to get translations and parse them.
  - Will create a parser to go through the second file that Beat is creating and update strings appropriately.

Is that about right?

hamishwillee, 2022-03-02:
Note, I've renamed the focus of this to establishing  what has to happen for translation. The discussion of WIP is at #1806

bkueng, 2022-03-02:
Waiting for Don's answer, but basically I can do whatever is needed.
We also need to think about the CRC: as translations on a server can change at any time, and the Firmware does not have the file at build time, I'm not sure how that was intended to be handled.

hamishwillee, 2022-03-03:
@bkueng 

> We also need to think about the CRC: as translations on a server can change at any time, and the Firmware does not have the file at build time, I'm not sure how that was intended to be handled.

This came up in the dev call as a blocker for "fitness for purpose". I will comment in the https://github.com/mavlink/mavlink/pull/1806

DonLakeFlyer, 2022-03-03:
The reason I suggested .ts was that I assumed Crowdin has better support for it. That is what QGC uses for all of it's other strings and it seems to work really well. Not sure how well a generic xml translation format works in Crowdin. I think it might be as good with respect to automatically moving translations across from version to version. The reason being it won't know about the additional context information which is included in .ts files in addition to the string itself. Maybe Crowdin doesn't actually do anything with that though. I always assumed it did. Don't know.

Also json is just miles easier to parse than xml from a coding perspective. This is the reason why QGC went with json for all of it's internal file formats.

DonLakeFlyer, 2022-03-03:
> Is that about right?

What you've laid out is correct.

hamishwillee, 2022-03-09:
Thanks @DonLakeFlyer @bkueng 
1. I've just done a straw poll of the 10 most popular translation alternatives to crowdin and they all support .ts files to at least some extent. 
2. Crowdin does "some things" with the context https://support.crowdin.com/file-formats/ts/ . The notes only cover the small number of things in their example message so some other context might not be captured - hard to say.

Upshot, I'm not opposed, though as you say, you'll have more difficulty parsing them. @auturgy unless you have a strong alternative reason for some other format, I think we should proceed with .ts files.

@bkueng Is there enough information in https://doc.qt.io/qt-5/linguist-ts-file-format.html and https://support.crowdin.com/file-formats/ts/#file-sample to prototype generating a source translation file for parameters? We can then try that out on crowdin and get Don to review it.

auturgy, 2022-03-09:
@meee1 your thoughts?

hamishwillee, 2022-03-09:
> We also need to think about the CRC: as translations on a server can change at any time, and the Firmware does not have the file at build time, I'm not sure how that was intended to be handled.

Yes, it should be possible to update translations used by GCS after firmware compile time/whenever they are updated. 
Broadly:
1. The vehicle should be able to tell the GCS _where_ it gets translations and/or updates
2. The vehicle should not have to _host_ translations (they are too big, and in any case, then we'd have to work out how to notify of updates)
3. Because the vehicle should not need to know about when translations have been updated it should not be the thing that updates the GCS of change.
4. Translation CRC should be moved out of the metadata files as it can't be hard coded without becoming unusable.
5. Translation file URL OK where they are, though perhaps they could just as easily be inside the metadata file to which they belong.

Since the vehicle can't provide updates on translations, and it is IMO too much to ask of a server that it provide push notifications, the implication is that the GCS must either update translations to some schedule (irrespective of the file has changed or not) or poll for changes in some way.

1. The simplest way to implement this would be to have a fixed URL for the translations, which is replaced with a new file whenever the translations are updated, and have the GCS simply reload the file to replace all its translation according to some schedule. This might be every week, or every time it restarts, if the language is changed etc. Maybe even something the user can trigger and/or disable in settings. Only downside is that you'll be doing an unnecessary download, possibly many times.
2. We might further be able to detect file changes in some way and update only when needed:
   - There is no easy way to reliably do this in HTTP (`WANT-DIGEST` is not broadly supported by servers so you can't just request it)
   - There are some _unreliable_ ways - e.g. you can get file size doing a HEAD request on the file and check the `Content-Length` header. This might offer an option to extend the time for downloads.
   - We might do something like specify that the server put a CRC for the file in an adjacent file (to the translation). Then the GCS could read this first, and only update the file if the directory changed. A bit risky as people forget to update support files. 
   - Other ideas?
 
@DonLakeFlyer @meee1 Any leanings here on what would make most sense for the GCS - just read the latest and swamp the existing? If so, what schedule makes sense.

DonLakeFlyer, 2022-03-09:
> * We might do something like specify that the server put a CRC for the file in an adjacent file (to the translation). Then the GCS could read this first, and only update the file if the directory changed. A bit risky as people forget to update support files.

This would seem like the only reasonable option. If there is no way to determine a newer file exists just over HTTP.

DonLakeFlyer, 2022-03-09:
Hmm: https://stackoverflow.com/questions/29314287/python-requests-download-only-if-newer

hamishwillee, 2022-03-10:
@DonLakeFlyer Great catch - yes, `Last-Modified` will probably work as a hint. Proposal above updated with this in mind https://github.com/mavlink/mavlink/issues/1656#issuecomment-1062533008

@meee1 @bkueng OK by you?

bkueng, 2022-03-10:
Yes fine with me. If the GCS doesn't check every startup and we compress the data, I think just downloading the whole file would work too.

hamishwillee, 2022-03-10:
Cool. I think it would work too 
- unless you think we should mandate compression of the data we should still check for change before downloading the data.
- Should we mandate HTTPS (not HTTP) to be in line with reasonable security?

@DonLakeFlyer You're the maintainer - is the approach suggested OK by you? If so we can move forward to modify the schema.

hamishwillee, 2022-03-16:
OK, so we discussed in the dev all. This translation download approach makes sense. We do think HTTPS should generally be mandated for any download from the Internet from a ground station.

As to the next steps in https://github.com/mavlink/mavlink/issues/1656#issuecomment-1049471042
@bkueng I guess you could start by generating TS format output from our source while waiting on @DonLakeFlyer to define the container format. Or you could define your own container format and go "all the way" - if it doesn't quite work for Don then the modifications for the container would likely be small. Seem reasonable @bkueng ?

bkueng, 2022-03-17:
Yes I'll look into it. My preference is to use an existing format.

bkueng, 2022-03-23:
I started looking into this, .ts files can only contain a single language. So I tested with a simple [key-value json](https://support.crowdin.com/file-formats/json/) file, with the following steps:
- convert parameters.json metadata to key-value json with https://github.com/mavlink/mavlink/blob/component_metadata_translation/translation_test/parameters_metadata_to_translation.py
  - this is the output: https://github.com/mavlink/mavlink/blob/component_metadata_translation/translation_test/to_be_translated.json
- uploading the output to crowdin
- do the translations
- download, resulting in a zip file with separate json files per language: https://github.com/mavlink/mavlink/blob/component_metadata_translation/translation_test/translations.zip
- use a simple [script](https://github.com/mavlink/mavlink/blob/component_metadata_translation/translation_test/combine_translations.py) to combine the languages
  - combined json output: https://github.com/mavlink/mavlink/blob/component_metadata_translation/translation_test/translated.json

The key-values have this form:
```
  "param/MPC_XY_CRUISE/shortDesc": "Default horizontal velocity in mission",
  "param/MPC_XY_CRUISE/longDesc": "Horizontal velocity used when flying autonomously in e.g. Missions, RTL, Goto.",
```
Another option is to use the string as key:
```
  "Default horizontal velocity in mission": "Default horizontal velocity in mission",
  "Horizontal velocity used when flying autonomously in e.g. Missions, RTL, Goto.": "Horizontal velocity used when flying autonomously in e.g. Missions, RTL, Goto.",
```
With the 2. option the lookup is simplified as the to-be-translated string can be used directly, but it's less robust to string changes and contains no context (the same string might translate differently depending on context).
I don't really have a preference here. We could also add both.

DonLakeFlyer, 2022-03-23:
Here is an example Fact MetaData json file: https://github.com/mavlink/qgroundcontrol/blob/master/src/Settings/App.SettingsGroup.json

DonLakeFlyer, 2022-03-23:
I can't quite remember but is the .ts format for COMPONENT_INFORMATION json metadata supposed to be generic? Or specific to a COMPONENT_INFORMATION type json format?

DonLakeFlyer, 2022-03-23:
For example I would like to see next steps in component information type to be able to get rid of all the json metadata QGC now needs to maintain on its own for mavlink command usage in missions. That would also includes lots of strings.

bkueng, 2022-03-23:
The main prohibiting reason against .ts is that there's only 1 language in a file. There's different ways around this:
- download a zipped/tar file containing all .ts files
- use a generic url with something like `%locale%` as placeholder for the language
- add a summary json file which then points to the individual .ts files on the server (including a last-modified timestamp)

I don't mind using either of these options though.

> My own opinion would be that it should be generic such that future metadata types don't need their own parsers. Which would means the json format needs some more info so that it can be parser. See QGC extra json stuff here: https://github.com/mavlink/qgroundcontrol/blob/master/translations/README.md as an example.

Yes that makes sense.

DonLakeFlyer, 2022-03-23:
I guess it all depends on whether you want to limit download sizes. If yes, then locale, if no, zipped all langs. I can't remember if QGC has cross platform multi-file archive unzip support. That could be an issue

hamishwillee, 2022-03-23:
I have no strong opinion about the above. Some thoughts:
- Not having to write a parser multiple times makes sense. I am not sure it would be possible in all cases. But we already have some examples where we do things like create UIs from definitions, so presumably some of that can be abstracted.
- Using `QGC Fact MetaData json` is something to consider if it works for both PX4 and ArduPilot, doesn't contain too much "cruft" and can reasonably be extended to work for other cases and by other ground stations.
- 
  > I would like to see next steps in component information type to be able to get rid of all the json metadata QGC now needs to maintain on its own for mavlink command usage in missions.
  
  This would require some level of analysis. We've talked around this idea several times and the biggest problem seems to be how you keep the information maintained. Is there an example file?

bkueng, 2022-03-25:
Here's the generic .ts version: https://github.com/mavlink/mavlink/tree/component_metadata_translation/translation_test_ts
- Parameters metadata contain an additional section which describes the fields to be translated:
  ```
  "translation": {
	  "items": {
		  "parameters": {
			  "list": {
				  "key": "name",
				  "translate": [ "shortDesc", "longDesc" ],
				  "translate-global": ["category", "group"],
				  "items": {
					  "bitmask": {
						  "list": {
							  "key": "index",
							  "translate": [ "description" ]
						  }
					  },
					  "values": {
						  "list": {
							  "key": "value",
							  "translate": [ "description" ]
						  }
					  }
				  }
			  }
		  }
	  }
  },
  ```
- [parameters_metadata_to_ts.py](https://github.com/mavlink/mavlink/blob/component_metadata_translation/translation_test_ts/parameters_metadata_to_ts.py) generates a .ts file from that with lookup context in the form of `/parameters/ASPD_SCALE_3/shortDesc`
- the GCS can then do the same to lookup the translated fields

As every parameter contains a category that's repeated in many cases, I added a `_globals` list to deduplicate these and therefore having to translate it only once.

Then on the server we can have a summary json in the form of:
```
{
	"de_DE": {
		"url": "translations_de_DE.json.xz",
		"last-modified": "timestamp"
	},
	"fr_FR": {
		"url": "translations_fr_FR.json.xz",
		"last-modified": "timestamp"
	}
}
```

hamishwillee, 2022-03-30:
PS We will need to describe the translation approach "generically" for the docs using parameters as an example. I don't think that will be too hard - the only complication is deciding how the name key will work.

bkueng, 2022-03-30:
Cool.

> My concern here is that it is a bit more of a pain for person uploading and maintaining the translations to create this file - and if they screw it up then you won't be downloading anything. If you guys think this is not a realistic concern then that's cool by me.

We can provide the script to update the summary after updating a file.
I think the simplest is to host translations on github and then let CI update the summary once a new crowdin update comes in.
I also think the extra file is a bit of an additional burden, but is easy to setup if we point to an existing implementation. It also provides the most flexibility, should we want to add more translation-specific metadata at some point. But as I said, other options are fine with me as well.

> As an aside, I know it does not matter, but the inconsistent indentation in "translation": { in https://github.com/mavlink/mavlink/blob/component_metadata_translation/translation_test_ts/parameters.json is odd.

I'll clean these things up once I do the PR (it starts to make sense to have a style-check in CI).

> PS We will need to describe the translation approach "generically" for the docs using parameters as an example. I don't think that will be too hard - the only complication is deciding how the name key will work.

I already added a schema for all translation parts: https://github.com/mavlink/mavlink/blob/component_metadata_translation/component_information/translation.schema.json. This can be accompanied by documentation, and checked when adding new metadata files with translations.