mavlink/mavlink #1657: Fence Altitude and Entry / Exit Times

flybrianfly, 2021-07-16:
For manned aircraft flying under IFR, blocks of airspace are sometimes assigned with entry or exit times. Also, airspace for manned aircraft have altitude definitions, enabling aircraft to fly above or under controlled airspace.

As part of a NASA SBIR grant, I'm currently developing a companion computer to autonomously plan flight paths given inclusion and exclusion zones defined by MAV_CMD_NAV_FENCE. I'd really like to extend MAV_CMD_NAV_FENCE to capture altitude extents of the fences, using parameter 7 (currently reserved).

Looking ahead to potential UAS in the NAS integration, it seems likely that some early solutions will be assigning blocks of airspace to a UAS with either an entry or exit time limit, similar to current IFR procedures for manned aircraft. I'd also like to extend MAV_CMD_NAV_FENCE to optionally include time restrictions. Currently, parameter 3 and 4 are available across all of the MAV_CMD_NAV_FENCE types.

I'd be happy to contribute to the development, but would like to check with the community about whether these suggestions would be welcome first.

Best,
Brian

hamishwillee, 2021-07-16:
Hi Brian,
Thanks for taking the time to discuss. 
- There are other things going on in fences at the moment, including things like being able to set the fence action via MAVLink and being able to set it per-fence. This might fit in with that kind of work. 
- This seems pretty reasonable for circular fences, but not sure whether reasonable for polygon fences (I guess nothing to stop the vertex height being different except complicating the fence calculation).

We have a dev call every 2nd week - https://mavlink.io/en/about/support.html#schedule. Would you be interested in coming along to discuss? Either way, let's see what others say.

@auturgy @LorenzMeier @julianoes FYI

auturgy, 2021-07-16:
Is this linked to Icarous?

flybrianfly, 2021-07-16:
> There are other things going on in fences at the moment, including things like being able to set the fence action via MAVLink and being able to set it per-fence. This might fit in with that kind of work.

Yes, seems reasonable to include with other fence work.

> This seems pretty reasonable for circular fences, but not sure whether reasonable for polygon fences (I guess nothing to stop the vertex height being different except complicating the fence calculation).

I was envisioning a few different cases of using altitude:
- If altitude is not defined for the fence, use the current treatment and assume the inclusion or exclusion fence is infinitely tall.
- If one altitude is defined, treat the fence as extending from the surface up to that altitude.
- If two altitudes are defined (i.e. two circular fence centers or two polygon vertices at the same position, but different altitudes), treat the fence as a slab between those altitudes.

For polygons with sloping altitude requirements, I would imagine that different autopilot or path planning implementations might treat them differently. So far for our flight testing, we have been creating axis aligned bounding boxes around our fences to make the autonomous path planning component more straightforward. This approach would flatten sloping altitude requirements. Alternatively, some autopilot implementations may chose to compute the fence altitude slope and determine whether the vehicle is inside the fence based on its lateral position and the fence height at that point.

> We have a dev call every 2nd week - https://mavlink.io/en/about/support.html#schedule. Would you be interested in coming along to discuss? Either way, let's see what others say.

I would be, but I'm located in the Mountain Time Zone of the US, so I might need to wait for a meeting with better US / Europe time overlap.

> Is this linked to Icarous?

No, it's more aligned with UAS in the NAS and the nascent Urban Air Mobility (UAM / AAM) industry (i.e. how to autonomously plan approach paths in urban environments. We see a lot of potential use in UAS, especially as a "direct to" method. For example, a drone flying a surveying mission over a farm detects that it's running low on battery and needs to fly directly to land. The farmer defined fences around buildings, towers, and trees on the farm, so the drone can autonomously plan a path from it's current location to the desired landing location while avoiding obstacles. Altitude limits are useful because some obstacles (like a barn) the drone could likely fly over, but others (like a tower or a group of tall trees) it would need to plan to fly around.

hamishwillee, 2021-07-21:
Hi @flybrianfly 

We discussed this in the dev call last night. Summary of discussion is below. This should be read as "yes we are interested" and "here are some ideas to bash about", not "you must do this":
1. Yes, allowing specification of the geofence to extend into 3rd dimension is something we would welcome.
2. Use cases of interest is not just ceiling altitude but also floor. I.e. we'd like to be able to specify a flight corridor inclusion zone/exclusion zone.
3. We only briefly discussed the geometry of the ceiling/floor of geofence. Some thoughts
   - As discussed, a single value for altitude allows only a flat ceiling in AMSL. @auturgy noted that if frames that support terrain altitude (AGL) this would actually follow terrain geometry.
   - Not sure whether we want to defined angled ceiling on polygon, for example by setting the vertices at different altitudes.
   - There was however interest in perhaps allowing a cone shaped exclusion or inclusion (this not supported currently).
4. Do not feel constrained to using the existing command parameter(s) for this. 
5. Each fence should be able to define its own altitude(s)
6. It should be possible to define a default altitude if no fence altitude is specified. Note that the spec does not currently define any altitude, so a flight stack is free to (and does in different cases) define infinitely tall geofences AND/OR default ceiling altitudes in parameters (i.e. they exist, but the fence protocol does not know about them). 

We recently added to the spec as "WIP" the concept of per fence actions. So right now if you create a fence definition with multiple circular and polygon fences they all get the one action - RTL, warning, or whatever. The new concept is that there is a MAV_CMD for setting the fence action which allows you to specify the action as one of those options OR the system default. IF this command is used in a fence definition then it sets the action for the next fence geometry (only). IF used as a command it sets the system default fence action.

The same concept might be used here. There is a command that sets the ceiling and floor altitude (and possibly also an angle projecting from the floor to the ceiling to allow cone-like exclusion). If it appears in a fence definition it applies to the next fence geometry. If used in a command it sets the system default. If the system does not support the message the fence definition is thrown out.

Is this enough feedback for you to continue your work? Can I close this?
PS Happy to try arrange a discussion in your timezone to work though details once you have a chance to think about this a bit more. 
Can I cl

flybrianfly, 2021-07-22:
> There was however interest in perhaps allowing a cone shaped exclusion or inclusion (this not supported currently).

I'm curious what the use case for this would be.

> Is this enough feedback for you to continue your work? Can I close this?

Yes, thank you this is great. I'll issue a PR when I make some code progress.

closed 2021-07-22 by flybrianfly

hamishwillee, 2021-07-28:
>> There was however interest in perhaps allowing a cone shaped exclusion or inclusion (this not supported currently).

> I'm curious what the use case for this would be.

My "understanding" was that it might be a way of defining the area below a landing path that you really don't want someone to enter. I use the term "understanding" because I might have that wrong - sorry - the discussion can be quite free ranging and I don't necessarily get everything. I see it as somewhat orthogonal in any case.