mavlink/mavlink #1660: How to send MAG_CAL_ACK and stop MAG_CAL_REPORT message?

bjubes, 2021-07-26:
The documentation says that [MAG_CAL_REPORT](https://mavlink.io/en/messages/common.html#MAG_CAL_REPORT) will be sent repeatedly until MAG_CAL_ACK is sent, but MAG_CAL_ACK is not mentioned anywhere else. How do I send MAG_CAL_ACK and stop the report from being sent?

hamishwillee, 2021-07-28:
@bjubes I believe this MAG_CAL_REPORT doc is incorrect and reflects a legacy implementation. That is based on this appearing along side some other docs that mention missing or ArduPilot only APIs. 

@julianoes Can you confirm how the calibration microservice works or point me to someone who can explain this (presumably this is understood from the MAVSDK implementation).

- From what I can tell all calibrations are kicked off by MAV_CMD_PREFLIGHT_CALIBRATION. I assume that returns the normal ACK if the command is accepted and calibration proceeds?
- How does the the autopilot notify status for the _different_ calibrations? 
- How does it notify when done? 
- Is MAG_CAL_REPORT used, and if so does it need to explicitly be stopped?

julianoes, 2021-07-28:
@hamishwillee and @bjubes you're asking all the right questions. Unfortunately, the answers are not very good, at least for PX4.

> From what I can tell all calibrations are kicked off by MAV_CMD_PREFLIGHT_CALIBRATION. I assume that returns the normal ACK if the command is accepted and calibration proceeds?

Correct, ACK first, and then it starts.

> How does the the autopilot notify status for the different calibrations?

PX4 abuses [statustext messages](https://github.com/PX4/PX4-Autopilot/blob/master/src/modules/commander/calibration_messages.h) for this.

> How does it notify when done?

It sends [a special statustext message](https://github.com/PX4/PX4-Autopilot/blob/607be59fd5b47160326052dea8eea7edc7619d66/src/modules/commander/calibration_messages.h#L53).

> Is MAG_CAL_REPORT used, and if so does it need to explicitly be stopped?

It doesn't look like it's implemented for PX4.

For ArduPilot, it looks like there is more of a calibration protocol specified, e.g.:
https://github.com/ArduPilot/mavlink/blob/5034afb62a2f8aef2b11717cbb0428a4115f4c60/message_definitions/v1.0/ardupilotmega.xml#L214-L223
I'm not sure how it works in detail though.

hamishwillee, 2021-07-28:
Thanks @julianoes - I suspect some of those statustexts will change to events in the next versions.

I'll see if I can get more information out of ArduPilot.

hamishwillee, 2021-07-28:
FWIW the ArduPilot dialect includes a number of commands and messages.
My GUESS would be that they use 
- `MAV_CMD_DO_START_MAG_CAL` to start a calibration. 
- As it is running they emit `MAG_CAL_PROGRESS` 
- At the end they emit `MAG_CAL_REPORT` until it is cancelled - using `MAV_CMD_DO_ACCEPT_MAG_CAL`
- `MAV_CMD_DO_CANCEL_MAG_CAL` can stop the whole thing
- Don't know what `MAV_CMD_FIXED_MAG_CAL_FIELD` and `MAV_CMD_FIXED_MAG_CAL` are needed for. Probably setting part of the calibration that is "fixed" to allow fast calibration.

Ardupilot Dialect
- [MAV_CMD_DO_START_MAG_CAL](https://mavlink.io/en/messages/ardupilotmega.html#MAV_CMD_DO_START_MAG_CAL)
- [MAV_CMD_DO_ACCEPT_MAG_CAL](https://mavlink.io/en/messages/ardupilotmega.html#MAV_CMD_DO_ACCEPT_MAG_CAL)
- [MAV_CMD_DO_CANCEL_MAG_CAL](https://mavlink.io/en/messages/ardupilotmega.html#MAV_CMD_DO_CANCEL_MAG_CAL)
- [MAV_CMD_FIXED_MAG_CAL](https://mavlink.io/en/messages/ardupilotmega.html#MAV_CMD_FIXED_MAG_CAL)
- [MAV_CMD_FIXED_MAG_CAL_FIELD](https://mavlink.io/en/messages/ardupilotmega.html#MAV_CMD_FIXED_MAG_CAL_FIELD)
- [MAG_CAL_PROGRESS](https://mavlink.io/en/messages/ardupilotmega.html#MAG_CAL_PROGRESS)
Common:
- [MAG_CAL_REPORT](https://mavlink.io/en/messages/common.html#MAG_CAL_REPORT)

julianoes, 2021-07-28:
> Thanks @julianoes - I suspect some of those statustexts will change to events in the next versions.

I guess that would be the idea, yes.

bjubes, 2021-07-28:
I am using ArduPilot. MAV_CMD_DO_CANCEL_MAG_CAL was the message I was looking for. Sending it after receiving MAG_CAL_REPORT prevents the MAV from sending any more.
Thanks!

closed 2021-07-28 by bjubes