mavlink/mavlink #1665: Mavlink checksum calculation

vtol-ronit, 2021-08-07:
Hi,
I'm writing code to communicate with qgcs. I received heartbeat message from qgcs. I generated mavlink folder by using ardupilot.xml,

Now I want to calculate checksum for received message. But I'm not able to understand how should I start for the process.

I try to follow the below link to understand the flow but I think things have changed because the macro he used for CRC_EXTRA is one dimensional and the macro I've is 2 dimensional array. I'm not able to understand how to access those elements.

#https://github.com/mavlink/mavlink/issues/30

Can someone help me how should I proceed to get the checksum of received mavlink message?

@ LorenzMeier

hamishwillee, 2021-08-08:
I don't understand your question, but the best way to look at this is the examine the C library source code. Basically see  https://github.com/mavlink/c_library_v2/blob/master/checksum.h and https://github.com/mavlink/c_library_v2/blob/master/mavlink_helpers.h#L621

vtol-ronit, 2021-09-11:
I was doing a silly mistake and it lead me to a confusion. I would like to summarize the whole checksum calculation here - 

Once you've received a mavlink message,
1. initialize a checksum variable with 0xFFFF 
2. call the **mavlink_get_msg_entry** function and pass the message_id (6th byte of message of received message for mavlink v1) of received message
3.  Then call **crc_accumulate_buffer** function. In argument pass **checksum** variable address and received message but exclude message_header and last 2 byte which is for checksum.
4. Then call **crc_accumulate** function and pass crc_extra and address of checksum variable as argument.
5. You'll get 16 bit checksum.

closed 2021-09-11 by vtol-ronit