mavlink/mavlink #1666: Path Planning Microservice TRAJECTORY_REPRESENTATION_WAYPOINTS

flybrianfly, 2021-08-09:
In the path planning microservice [TRAJECTORY_REPRESENTATION_WAYPOINTS message](http://mavlink.io/en/messages/common.html#TRAJECTORY_REPRESENTATION_WAYPOINTS), there is an array of five uint16_t for MAV_CMD. How should the MAV_CMD be encoded? I'm familiar with MAV_CMD from the mission protocol microservice and the command protocol microservice, but I'm not seeing how the array of five uint16_t should be used to encode the MAV_CMD for the path planning microservice.

For mission protocol and command protocol, the MAV_CMD is a single uint16_t, but I can see the benefit of sending additional parameters, since the [TRAJECTORY_REPRESENTATION_WAYPOINTS message](http://mavlink.io/en/messages/common.html#TRAJECTORY_REPRESENTATION_WAYPOINTS) lacks the param1 - param4 that [MISSION_ITEM_INT](http://mavlink.io/en/messages/common.html#MISSION_ITEM_INT) messages have; however, the extra parameters in those messages are floats, not uint16_t, so I'm not sure how to encode the MAV_CMD piece.

flybrianfly, 2021-08-09:
To add on to this slightly, what if my path planner would like to command the UAV to [loiter to altitude](http://mavlink.io/en/messages/common.html#MAV_CMD_NAV_LOITER_TO_ALT) or [change speed](http://mavlink.io/en/messages/common.html#MAV_CMD_DO_CHANGE_SPEED), how would that be sent via the path planning microservice?

flybrianfly, 2021-08-11:
Makes sense, thanks!

flybrianfly, 2021-10-04:
Apologies for re-opening this, but on a closer look:
1. I don't see how a companion computer would send PX4 a loiter to altitude command. From this documentation (https://docs.px4.io/master/en/computer_vision/path_planning_interface.html#px4-waypoint-interface), the only setpoint information commanded is vehicle position and velocity.

I'm considering a use case where the difference between the vehicle's current altitude and commanded altitude would require a descent rate exceeding the vehicle's capabilities. I suppose the options are to: (1) have the companion computer include a loiter command, or (2) have the companion computer send the command with no knowledge of the aircraft performance, the autopilot would then need to either insert a loiter or reject the guidance from the companion computer. I think it would be beneficial for the companion computer to be able to send the full-suite of MISSION ITEMS.

2. I'd like to have an API between UAS components that doesn't leave anything open to interpretation; it's the only way that we will improve interoperability between manufacturers as an industry.

hamishwillee, 2021-10-06:
@flybrianfly My interpretation is that the **companion** never ever sends _commands_ - it sends **setpoints** as discussed in https://docs.px4.io/master/en/computer_vision/path_planning_interface.html#companion-waypoint-interface

So to summarise, 
- the `TRAJECTORY_REPRESENTATION_WAYPOINTS` is sent out by the autopilot with the current _setpoint_ it expects to go to, the current _waypoint_ and the next _waypoint_.
- The companion sends back `TRAJECTORY_REPRESENTATION_WAYPOINTS` with a _setpoint_. 
  - If the planner does not know how to plan for the current command or understand the current intent (e.g. mission, or landing or whatever) then it just mirrors back the setpoint it got from the vehicle. 
  - If the planner does know about the command/context then it does planning according to its own algorithm and sends back an appropriate setpoint for where it would prefer the vehicle to move.
  
 The vehicle will then attempt to reach the setpoint. If it can't achieve the setpoint it just won't. The vehicle will however sent the information it has about its intent to the companion, which will then send a follow new setpoint. 

Upshot is that the interface doesn't know or care about whether the vehicle can meet the requirements. Probably a planner could do better if it has this information but we don't expose more than the vehicle type through MAVLink. I'm also not sure that it would be through this interface - there is a separation of concerns.

Thoughts?

flybrianfly, 2021-10-06:
If the path planner knows the maximum ascent and descent angles, then it can ensure that the vehicle will be able to follow the vertical path guidance. This has been useful, especially for fixed wing vehicles, when trying to fly autonomous approaches to landing when the path planner was engaged with the aircraft too high and too close to be able to follow the glideslope. It can add loitering ascents / descents as necessary.

Additionally, if the airspeed and maximum bank angle are known, the path planner can take into account the vehicle turn radius when planning routes around obstacles.

hamishwillee, 2021-10-07:
> The current MISSION ITEMS cover any maneuver that a path planner may want to command and enable the path planner to autonomously handle more complex mission scenarios than the current setpoint approach.

I so completely disagree :-0. 
- This approach is actually the "old" way of managing dynamic missions - you have a mission, you spot that you want to change it for whatever reason, so you upload a new mission that has new points. It's a bit crap because the autopilot always finishes the last point it has buffered before doing the new waypoint, but it works and you could do this now if you really needed this.
- The setpoint interface provides the finest-possible-grained control over the vehicle - you are saying **exactly** where and how you want the vehicle to move. Sending a mission item is far more limited - you only have what the mission interface allows.
- I understand that it is "easy" - rather than plotting a precise path you simply command a new target. But it isn't as flexible.

hamishwillee, 2021-10-13:
@flybrianfly IMO commanding a loiter is something you would have to plan out yourself. The underlying mission is going to sit and wait until you do something that meets the requirement of going to the next waypoint. 

The landing stuff though might be a valid concern. Interestingly in offboard mode on PX4 these _are_ [taken care of since you can specify the type of a setpoint](https://docs.px4.io/master/en/flight_modes/offboard.html#fixed-wing) (this is a completely different model).

Let me see if I can find out more information.

flybrianfly, 2021-11-15:
I'm circling back to this and Offboard mode seems really interesting. @hamishwillee, do you have any insight into when Offboard mode should be used vs path planning interface? They seem to be accomplishing the same end-goal and offboard mode seems to offer more flexibility. Understand that some of this may be historical, but as the developer of a newer flight stack, trying to figure out which services should be supported. Thanks!

hamishwillee, 2021-11-17:
@flybrianfly Perhaps :-) YOu might get a better answer in the PX4 #computer-vision channel though.

The offboard mode has been around forever and is supported on many flight stacks as the way for a companion computer to precisely control vehicle movement - typically by providing position or velocity setpoints. It allows very precise control over movement by a companion computer.

However for the purpose of trajectory planning the problem is that it has zero understanding of what the vehicle is trying to do - you're handing over full control to the companion computer. That's mostly OK if the companion computer knows how to execute everything in a mission itself and you're just using your vehicle as a dumb platform. No one does this "in general" because you'd have to write something to convert missions to setpoints on your companion - possible and pretty cool, but not trivial. Further, some things might not be possible to implement in a companion computer - for example you might have a mission with a command to control a gripper - but if that command isn't supported on the flight stack in offboard mode, you can't use it.

By contrast, the trajectory interface provides an _intent_ to the companion computer. That means if you're using trajectory planning in a mission you're getting the full benefit of the flight stack to execute your mission, with the companion computer providing setpoints as needed. The main use case being obstacle avoidance, where you only change the setpoints back to the vehicle if you see an obstacle in the planned path. You still get all the mission and existing integration with the ground station, etc.

I hope that helps! The approach was developed initially for PX4 obstacle avoidance, which is why I suggest asking on the PX4 slack channel.