mavlink/mavlink #1667: "‘to_char_array’ was not declared in this scope" when running gtest suite generated with mavgen

WBurzynski, 2021-08-10:
Hello,
 I generated mavlink c++11 libraries using mavgenerate gui. I wanted to run generated gtest suite, but in many test there is error `‘to_char_array’ was not declared in this scope` and i could not find definition in any of generated files. Do i have to provide my own implementation of this function or is it issue with generator?

hamishwillee, 2021-08-11:
@julianoes Have you run into this problem?

julianoes, 2021-08-11:
I don't know. All I found with a quick search was this: https://github.com/ArduPilot/pymavlink/blob/118bcc4bfc2998c3ae825c6654f039cdf644db2d/generator/CPP11/test/posix/mtest.cpp#L24

WBurzynski, 2021-08-11:
I replaced `to_char_array()` with, `mavlink::set_string_z()` in tests, and it works just fine, but i think this should be handled in a first place. I actually was suprised to see any test, there was no mention in documentation at mavlink.io

hamishwillee, 2021-08-11:
@peterbarker This something you have run into before? I.e. is it a generator bug?

vooon, 2021-08-12:
@WBurzynski how did you run it?

WBurzynski, 2021-08-12:
@vooon i created simple cmake project:
```cmake
cmake_minimum_required(VERSION 3.11)

project(mavlink-cpp11_tests)

# setup conan dependencies
include(cmake/conan.cmake)
run_conan()

enable_testing()
include(GoogleTest)

add_executable(mavlink-cpp11 main.cpp)
target_include_directories(mavlink-cpp11 PUBLIC include)
target_compile_features(mavlink-cpp11 PRIVATE cxx_std_17)
target_link_libraries(mavlink-cpp11 CONAN_PKG::gtest)

# let gtest find and execute tests
gtest_discover_tests(mavlink-cpp11)
```
and cpp file:
```cpp
#include "mavlink-cpp11/common/gtestsuite.hpp"
```
and builded it.

vooon, 2021-08-12:
You should use that makefile:

https://github.com/ArduPilot/pymavlink/blob/4dac715e4e394714d0676fa6675ff89d2871f380/generator/CPP11/test/posix/Makefile

julianoes, 2021-08-30:
@WBurzynski did this resolve your issue?

WBurzynski, 2021-08-31:
@julianoes unfortunatelly thiis makefile did not work, because it could not find googletest in set location. But because i did find solution i mentioned above and run tests, i did not investigate problem.

julianoes, 2021-08-31:
If anyone has a link to instructions how to build and run these tests that would be appreciated. (I have never used or touched them as far as I remember.)

hamishwillee, 2021-09-01:
@peterbarker Any ideas? I haven't been able to get the CI tests to run locally.