mavlink/mavlink #1678: Integrated Camera/Gimbal Systems Documentation

dlwalter, 2021-08-20:
With the Gimbal V2 Protocol getting mainlined we should look at updating the documentation for vendors implementing an "all-in-one" standalone integrated camera/gimbal video system to ease integration into PX4/QGroundControl based systems.  Examples of these COTS gimbals would be Nextvision's Colibri and Nighthawk, and Trillium HD40 with more coming soon.  

The current Gimbal V2 documentation (https://mavlink.io/en/services/gimbal_v2.html) comes from the perspective of pointing a gimbal. The Camera Protocol Documentation comes from the perspective of a camera.  However, an all-in-one implementation demands communication using both, and then a few more messages from the rest of the mavlink message set.

We suggest a _section of documentation is needed to explain how to manage an integrated smart gimbal from the QGC interface_ (or other base station).  To get started, we suggest outlining the intended user actions, and then the mavlink messages that support those actions.

Simply having this mapping table of actions to messages would have helped us digest and implement these new protocols much faster into new products..

A new section could cover the specific use case of a fully integrated camera/gimbal.

**A standalone gimbaled video system has the following features:**
- Video Streaming and Recording from 1 or more cameras.
- Still Image Capture from 1 or more cameras.
- 3 Axis stabilization creating a camera view that is level to the horizon
- Accepts Pan/Tilt controls to slew the camera view around while stabilized in the Navigation frame.

It may also include:
- Retract into a safe position during landing
- Switching between visible, IR, or dual views
- Zoom using mechanical, digital, or lens changes
- Slew to a global position (given global position of UAS)
- Track a point or area on the ground
- Nudge the camera while tracking
- Stream global position of camera view center for map display along with slant range
- Stream global position of tracked point/area
- Allow stabilization turned off for specific axes

	
**If we frame the problem from the viewpoint of the user we can assume the following intentions/actions:**

- Use a joystick or other interface device to pan and tilt a stabilized gimbal to control a camera while keeping the horizon level.  When released the joystick will point at the same attitude relative to north/horizon despite aircraft movements.
- Zoom in and out a gimbaled camera or switch FOV from narrow to wide
- Start/Stop Recording Video
- Capture single/multiple still images
- Select between IR and visible video on a gimbaled camera or view IR and visible side by side
- Change the Thermal palette (LUT) if supported.
- Retract the Gimbal while landing
- Select a point on a map and have the gimbaled camera payload look at that point.
- View on a map the point where the gimbaled camera is looking along with Slant Range
- View on a map an overlay displaying the camera image bounds on the ground
- Switch the gimballed camera to track (using computer vision techniques) an object or area under a cursor
- While Tracking, Nudge the gimbal view to pan/tilt while keeping the track in view.
- Using the touchscreen, command the camera to track the object on the image where the finger was pressed.
- Change gimbal setting to turn off yaw stabilization and lock yaw to the aircraft

The following are the relevant Mavlink enums, messages, and commands used to implement these and other features.

**Enums:**
STORAGE_STATUS
STORAGE_TYPE
CAMERA_CAP_FLAGS
CAMERA_TRACKING_STATUS_FLAGS
CAMERA_TRACKING_MODE
CAMERA_TRACKING_TARGET_DATA
CAMERA_ZOOM_TYPE
CAMERA_MODE
GIMBAL_MANAGER_CAP_FLAGS
GIMBAL_MANAGER_FLAGS
GIMBAL_DEVICE_FLAGS
GIMBAL_DEVICE_CAP_FLAGS
GIMBAL_DEVICE_ERROR_FLAGS
VIDEO_STREAM_STATUS_FLAGS
VIDEO_STREAM_TYPE

**Commands:**
MAV_CMD_DO_SET_ROI_LOCATION
MAV_CMD_DO_SET_ROI_WPNNEXT_OFFSET
MAV_CMD_DO_SET_ROI_NONE
MAV_CMD_DO_SET_ROI_SYSID
MAV_CMD_DO_CONTROL_VIDEO
MAV_CMD_DO_GIMBAL_MANAGER_PITCHYAW
MAV_CMD_DO_GIMBAL_MANAGER_CONFIGURE
MAV_CMD_DO_TRIGGER_CONTROL
MAV_CMD_SET_CAMERA_MODE
MAV_CMD_SET_CAMERA_ZOOM
MAV_CMD_IMAGE_START_CAPTURE
MAV_CMD_IMAGE_STOP_CAPTURE
MAV_CMD_CAMERA_TRACK_POINT
MAV_CMD_CAMERA_TRACK_RECTANGLE
MAV_CMD_CAMERA_STOP_TRACKING
MAV_CMD_RESET_CAMERA_SETTINGS
MAV_CMD_VIDEO_START_CAPTURE
MAV_CMD_VIDEO_STOP_CAPTURE
MAV_CMD_VIDEO_START_STREAMING
MAV_CMD_VIDEO_STOP_STREAMING
MAV_CMD_REQUEST_STORAGE_INFORMATION
MAV_CMD_REQUEST_STORAGE_FORMAT
MAV_CMD_REQUEST_CAMERA_SETTINGS
MAV_CMD_REQUEST_CAMERA_INFORMATION
MAV_CMD_REQUEST_CAMERA_CAPTURE_STATUS
MAV_CMD_REQUEST_CAMERA_IMAGE_CAPTURE
MAV_CMD_REQUEST_VIDEO_STREAM_INFORMATION
MAV_CMD_REQUEST_VIDEO_STREAM_STATUS

**Messages:**
CAMERA_TRIGGER
CAMERA_INFORMATION
CAMERA_SETTINGS
CAMERA_CAPTURE_STATUS
CAMERA_FOV_STATUS
CAMERA_TRACKING_IMAGE_STATUS
CAMERA_TRACKING_GEO_STATUS
GIMBAL_MANAGER_INFORMATION
GIMBAL_MANAGER_STATUS
GIMBAL_MANAGER_SET_ATTITUDE
GIMBAL_MANAGER_SET_PITCHYAW
GIMBAL_MANAGER_SET_MANUAL_CONTROL
GIMBAL_DEVICE_INFORMATION
GIMBAL_DEVICE_SET_ATTITUDE
GIMBAL_DEVICE_ATTITUDE_STATUS
AUTOPILOT_STATE_FOR_GIMBAL_DEVICE
VIDEO_STREAM_INFORMATION
VIDEO_STREAM_STATUS

If there are addition user actions and features that I have missed, please suggest those as well.

hamishwillee, 2021-08-22:
@dlwalter Thanks for posting this.  Sounds very reasonable. I know most (if not all) of these cases were discussed in the design - that should be captured in the docs.

@olliw42 Anything here you think is missed or answers you can add to this? 

@julianoes is on holiday. I "think" he is back next week (7 days), so this won't be looked at in detail before then, and there is likely to be some delay while he ramps up again.

olliw42, 2021-08-23:
> @olliw42 Anything here you think is missed or answers you can add to this?

there would be lots to say, but I feel like I did that already, and my ideas didn't transmit well back then, so I would not expect them to transmit better now. It's a complex topic and can be handled with very different philosophies (I for myself am happy with my concept of "super mode", it works well for me). 

:)

it is a very good and helpful compilation of what integrated systems should allow nowadays

dlwalter, 2021-08-24:
@hamishwillee copy on @julianoes being on holiday.  I can start a PR and create a document skeleton to start throwing ideas down.

In addition, there are a couple features that may not be supported with the current spec and I'd also like to know if the community has any ideas on these - 

**Controlling a Thermal Video Stream**
- Toggling between Thermal/Visible/Overlay/SplitView - what message/command to control this as well as a CAMERA_CAP_FLAGS to indicate capability
- Changing the Thermal Camera LUT/Palette - what message/command to control white hot/black hot, artificial color etc.

**Streaming CAMERA_FOV_STATUS to GCS to overlay on map**
- This calculation requires global position of the aircraft, which is not included in AUTOPILOT_STATE_FOR_GIMBAL_DEVICE.  GLOBAL_POSITION_INT would be required and possibly terrain data.
- Do we indicate capability of CAMERA_FOV_STATUS with another Camera capability flag?  Or just detect the message output if it is present.
- Slant Range currently would have to be calculated on the ground station, is there room in a message for this to be added? 

**Triggering Camera Tracking**
- MAV_CMD_CAMERA_TRACK_POINT/RECTANGLE needs a video timestamp or frame count to account for video latency to the ground station so the onboard computer knows what was intended to be tracked.

hamishwillee, 2021-08-25:
@dlwalter 

> "Toggling between Thermal/Visible/Overlay/SplitView - what message/command to control this as well as a CAMERA_CAP_FLAGS to indicate capability"

You're talking about a camera that supports streaming dual modes - thermal and visible?

So basically the camera API does basic stuff using messages; everything else is done using the [camera definition file](https://mavlink.io/en/services/camera_def.html). Essentially this allows you to set up a UI for your camera and map to named parameters that a GCS can change to modify settings. I imagine thermal settings would have to be set this way. 

It does seem possible that you can check if a stream you are being sent already is thermal: https://mavlink.io/en/messages/common.html#VIDEO_STREAM_STATUS_FLAGS_THERMAL

@dogmaphobic 
- Can you confirm how this works for toggling the type of thermal stream you want?
- What if you want to display a switcher for toggling the thermal display mode in a GCS at the top level rather than the custom configuration UI for your camera?

MatejFranceskin, 2021-08-25:
So far we implemented switching between EO/IR in practice in two different ways using what mavlink already provides:
1. We actually have 2 separate cameras, each one with its own camera definition XML file, sparate video stream and parameters. There is no special handling for thermal camera as it has its own set of parameters like palette etc.
2. Some cameras appear as one camera to GCS as they cannot produce EO & IR video streams in parallel. In this the switch between EO & IR is done via parameter in camera definition file. Available parameters like IR palette appears in GUI only when IR camera is selected.

Maybe we could improve handling for case 2 by standardizing EO/IR camera switch parameter? It would have to be more generic, like "select camera stream" for cameras offering multiple video streams with different sets of parameters.

olliw42, 2021-08-28:
> Adding new standard parameter like CAM_STREAM is the only improvement that I can think of now.

YES! PLEASE :)

https://github.com/mavlink/mavlink/issues/1306

(wanted to bring this up again anyhow, good occasion to do it LOL)

hamishwillee, 2021-08-29:
OK! Thanks guys. I'll add the set (including whatever makes sense to me from #1306 on Weds for discussion in the dev call). We can then discuss around that.

julianoes, 2021-08-30:
This all sounds reasonable to me. And further documentation of how that can be implemented would be great. In parallel, I should write an example of a camera/gimbal based on MAVSDK.

One point, I would like to comment on though:

> The current Gimbal V2 documentation (https://mavlink.io/en/services/gimbal_v2.html) comes from the perspective of pointing a gimbal.

Yes, but we always had in mind how to deal with an integrated/smart camera/gimbal "combo". That's why the gimbal manager is not a mavlink component but it's a thing a mavlink component can implement. So in this case, the camera implements the gimbal manager (and uses all its messages) but one mavlink node/component is enough.

hamishwillee, 2021-09-01:
@olliw42 So the set you are interested in
```
CAM_VIDEV
CAM_PHOTOEV
CAM_VIDISO
CAM_PHOTOISO
CAM_VIDSHUTSPD
CAM_PHOTOSHUTSPD
```
note, used "Photo" in full to be consisten with below.

And Gus further suggested [this set](https://github.com/mavlink/mavlink/issues/1306#issuecomment-587976197) as examples that are somewhat standard.
```
CAM_VIDFMT          Video encoding format
CAM_VIDFPS          Video frame rate
CAM_PHOTORATIO      Photo Aspect Ratio
CAM_PHOTOFMT        Photo image (saving) format
CAM_PHOTOSIZE       Photo size
CAM_PHOTOQUAL       Photo image (saving) quality (for compressed modes)
```
And of course `CAM_STREAM`.

@julianoes Do you know know of other ones we should make standard? Good to get as much common stuff in as we can.

Further, would be great to see that example and docs, you know you have to do it all right? :-) 
Minimally, I think it would be useful to have an interim section in the docs that copies in what you think is relevant from the above. Should include the statement you made above of how this might be implemented. 
IN fact, you could start with that :-)

julianoes, 2021-09-02:
One question: so if something is "common"/"standard" does it mean it has to be implemented? Or is it just a suggestion? I would prefer it to be a suggestion, and only have `CAM_MODE` be a default.

olliw42, 2021-09-02:
small correction, the list I would want to suggest is (the two in the first post of https://github.com/mavlink/mavlink/issues/1306 were missing):

```
CAM_VIDFPS
CAM_VIDFOV
CAM_VIDEV
CAM_PHOTOEV
CAM_VIDISO
CAM_PHOTOISO
CAM_VIDSHUTSPD
CAM_PHOTOSHUTSPD
```

hamishwillee, 2021-09-08:
Thanks @olliw42 - I will think on the list again tomorrow.

>"common"/"standard" does it mean it has to be implemented? Or is it just a suggestion? I would prefer it to be a suggestion, and only have CAM_MODE be a default.

@julianoes What it means is that if you have a camera that supports the functionality described by one of these keys, then you should use these keys to refer to them rather than some arbitrary key. If you do, then a GCS can pull out those values into a particular UI if it so wishes. If you were irritating and chose to use a different key then the settings UI would still be available just as it is now - but a GCS would have no way to control where/how the settings were rendered.

You don't have to support the functionality and you don't have to have the same settings or whatever. 

Hope that makes sense.

hamishwillee, 2021-09-23:
Cam stream added by https://github.com/mavlink/mavlink-devguide/pull/395

MiCH28K, 2026-02-13:
Hello! I am new to Mavlink and right now I'm trying to implement Mavlink protocol controls for my integrated gimbal/camera. I'm currently having trouble implementing the Mavlink protocol for controlling a camera and a gimbal manager on a single device. I have a number of questions I can't find answers to:
1) Should my device send a single heartbeat (for the camera/gimbal only) or two separate heartbeats with different MAV_TYPE and component IDs? In the latter case, can I use the same UDP socket? Will this cause problems with GroundControlStation and PIX?
2) If I send only one heartbeat with MAV_TYPE_CAMERA and MAV_COMP_ID_CAMERA, should GroundControlStation or PIX send COMPONENT_INFORMATION_BASIC to my camera, or will it only send CAMERA_INFORMATION? If the latter, how can it know if the gimbal manager is present on the camera if the CAMERA_INFORMATION field doesn't have the MAV_PROTOCOL_CAPABILITY field?
3) The camera should regularly send CAMERA_INFORMATION, and the gimbal should send GIMBAL_MANAGER_STATUS. How can this be implemented on a single device if Mavlink messages are exchanged via UDP? What component ID and MAV_TYPE should be chosen?

I'm very sorry if I'm in the wrong issue with those questions, would be glad to receive any advice.

hamishwillee, 2026-02-17:
@dlwalter I assume you have forgotten about this. Would be great to get some documentation on the topic? Or @julianoes would you be able to do this?

If not, I might add a short section just with this kind of thing ^^^ added.