mavlink/mavlink #1680: Inconsistency in MAVFtp URI definition

olliw42, 2021-08-22:
https://mavlink.io/en/services/camera.html#camera_identification says that the mavftp uri is `mavlinkftp://` while https://mavlink.io/en/services/ftp.html#mavlink-ftp-url-scheme says it is `mftp://[;comp=<id>]<path>`, which is also used in https://mavlink.io/en/services/component_information.html#file-locationsurls.

While technically it could be different for different microservices, I suspect it should be identical. Pl double-check and correct as needed. :)

hamishwillee, 2021-08-22:
Thanks @olliw42 . They should be the same, but in fact the camera API does indeed use `mavlinkftp`. 

@julianoes Any way we can make this consistent or have we gone too far :-(. 
We might perhaps make it possible to get camera information via component information API and move to using just one approach in future?

olliw42, 2021-08-23:
is mavlinkftp actually used by any implementation? 

I tried with QGC and it doesn't and I also couldn't see any code in QGC (I might be mistaken though)

if some implementation indeed uses already mavlinkftp I think one easily could allow for mftp in addition and mark it as the recommended one in the docs. 
(it should be really a piece of cake to update a code to also hadle mftp in addition to mavlinkftp).

> We might perhaps make it possible to get camera information via component information API and move to using just one approach in future?

this sounds compelling but given the deficiencies of component information in its current form (and I suspect also future form) I seriously hope it won't... (my 2 cents)(recall that component information initially was drafted similarly to camera information but most bits there then removed, so it isn't a 1to1 substitute)

hamishwillee, 2021-08-25:
I was wrong, it was fixed up to mftp - see https://github.com/mavlink/qgroundcontrol/pull/9225 and https://github.com/mavlink/qgroundcontrol/blob/master/src/Vehicle/FTPManager.cc#L22
I'll tidy the docs.

Regarding having the camera information file delivered via `COMPONENT_INFORMATION` that would certainly require some more thought. 

You didn't like `COMPONENT_INFORMATION` because the message now lacks basic information right?
I see your point. The camera info message is designed so you can implement a basic camera using just mavlink messages/command, but for advanced features you need to implement FTP. You are arguing that there is a class of basic information you might want that you shouldn't have to implement all the extra MAVFTP stuff to get.

I think I agree. What we should have done is had two messages - one for basic component info and one for metadata. We started that way, but it was the basic info that got dumped when we needed the room for bigger URLs.

Anyway, I think if this was changed it would just be an alternative way to get the camera file, not a replacement for all the other info.

olliw42, 2021-08-25:
> I was wrong, it was fixed up to mftp - see [mavlink/qgroundcontrol#9225](https://github.com/mavlink/qgroundcontrol/pull/9225) and https://github.com/mavlink/qgroundcontrol/blob/master/src/Vehicle/FTPManager.cc#L22
> Docs fixed in [mavlink/mavlink-devguide@44cca2d](https://github.com/mavlink/mavlink-devguide/commit/44cca2d1813ddc1d2b5b8f28075384cdd721ba67)

hold on

this is ONLY for component_information ... QGC does not have an implementation to use mavftp for camera (as much as I see) ... so you CANNOT conclude from this that it was changed also to mftp for camera too (which was the topic here) !

all you can conclude from this is that for component_information it is mftp and that QGC has no implementation for camera so that the camera spec can easily be changed without affecting QGC

so, at the end of the day, you still have to (1) decide if you want to change it for camera and (2) check if there are other implementations which would be affected

olliw42, 2021-08-25:
> You didn't like `COMPONENT_INFORMATION` because the message now lacks basic information right?

not only because of this, not only because of this ... that's one point, yes, but just one ... 

there are more fundamental issues with merging camera_info into component_info ... it's not a 1-to-1 substitute also conceptually and functionally wise
(you could make the same argument for gimbal_info, there it becomes obvious how miserable of an idea this logic would be)

but the whole component_information discussion is so - don't know how to say it - "strange", that I find it difficult to engage publicly

hamishwillee, 2021-08-25:
Curses. You're right. 

I am pretty sure that QGC does not support mftp download of camera definition files (wandered through the code and just found [http here](https://github.com/mavlink/qgroundcontrol/blob/master/src/Camera/QGCCameraControl.cc#L1957)). Also seems that MAVSDK supports using HTTP. So it may be that as you guess, nothing does.

@julianoes Do you know if anything supports mftp for camera information file download? I have changed the docs to say `mftp` required - and I'll change them back if you suggest there is an implementation in the wild that uses mavlinkftp for the URL

julianoes, 2021-08-30:
> @julianoes Do you know if anything supports mftp for camera information file download? I have changed the docs to say mftp required - and I'll change them back if you suggest there is an implementation in the wild that uses mavlinkftp for the URL

I don't know of an implementation using mavlink FTP. The ones I'm aware of ues HTTP, pretty sure.

So we can change it to `mftp`, however, I would actually prever mavlinkftp or mavftp because MFTP is already "taken": https://acronyms.thefreedictionary.com/MFTP :joy:.

olliw42, 2021-08-30:
mavftp would be my choice, not too long and not too short 😜

hamishwillee, 2021-08-30:
@julianoes So changing camera to something else when FTP not supported by anything is easy. We now have component stuff in the wild, so do you think it can be changed? If you do, and you feel strongly this should be done, can you do it?

If not, we probably should stick with mftp and live with it. I'm not too concerned about the acronym being overloaded - that link already has 9 definitions, all of which make sense in their context.

julianoes, 2021-08-31:
It's implemented, let's leave it as `mftp`:
https://github.com/PX4/PX4-Autopilot/blob/f9d8c613b048f58eb3110e9af13cb3a89c4c866f/src/modules/mavlink/streams/COMPONENT_INFORMATION.hpp#L67

hamishwillee, 2021-09-01:
Cool. I am closing this then. The mavlink docs say this too.

closed 2021-09-01 by hamishwillee

olliw42, 2021-09-01:
great, thx for clearing this up

I've seen you also already changed the docs for camera, fantastic

:+1: